vadimgrn/usbip-win2 #5: Devices are not attached

ttyz, 2022-02-05:
I have issues attaching devices. Despite a message saying

```
usbip: debug: usbip.c:151:[run_command] running command: attach
usbip: debug: usbip_vhci.c:61:[usbip_vhci_driver_open] device path: \\?\usbipwin#vhci#1&19724a8b&b#{d35f7840-6a0c-11d2-b841-00c04fad5171}
succesfully attached to port 1
```

The device does not actually appear in the device manager (or disappears after a very short time). I have tried with multiple different devices and always get this result. With usbip-win v.0.3.5 the device attaches without issues, but does not function correctly, so I decided to try this version as it seems to be maintained.

Software I use:

Client:
usbip-win 0.7.5.1
Windows 10 build 19041 (qemu VM)

Server:
usbipd 2.0
Linux 5.16

Things I still want to try:

1. Collect logs from the driver using the instructions in the README.md
2. Build the latest version of the driver from `master` or `develop`

This would probably take me some time as I am not as familiar with Windows development and tools involved, But I will update the issue as soon as I have more logs to show

If you have any other suggestions to try, would be happy to do so

vadimgrn, 2022-02-05:
Driver log is required. Do not use develop branch, it can be unstable.

ttyz, 2022-02-05:
I have some progress (but unfortunately no logs)

With lhe latest build I can successfully attach at least one device (a Dell usb mouse) and it works as expected. As soon as I try to attach anything else (tried a Logitech Webcam, DualSense controller and a TP-Link Bluetooth adapter) I get a BSOD with error code `SYSTEM_SERVICE_EXCEPTION` or `SYSTEM_THREAD_EXCEPTION_NOT_HANDLED` and the driver log file is completely empty. I will continue experimenting with different devices and my methodology, in the hopes of getting some logs out.

I do run Windows in a qemu VM, but I would consider switching to VirtualBox only as a last resort option. I need PCI passthrough for this system and last time I tried it did not work for me in VirtualBox.

vadimgrn, 2022-02-05:
There are several known places where BSOD occurs often. I'm working on that, but this is laborious task.

vadimgrn, 2022-03-14:
Try new release, pls

ttyz, 2022-03-20:
The behaviour is the same. I managed to collect a trace for the webcam: https://gist.github.com/ttyz/6e665e82a26414ea3f671974b5b97c3d

I will try to experiment with the devices that give me BSODs and attach the findings here as well.

docterling, 2022-03-25:
I met the problem too,when I detach the webcam,it display "port 1 is succesfully detached" and the traceview msg:

> 16858599	vhci	3796	4628	0	16411	03\25\2022-12:02:47:312	VDEV_VHCI: irql !0x00000000(Low)
16858600	vhci	3796	4628	0	16412	03\25\2022-12:02:47:312	VDEV_VHCI: enter irql 0x00000000(Low), USBIP_VHCI_UNPLUG_HARDWARE(0X2A4004)
16858601	vhci	3796	4628	0	16413	03\25\2022-12:02:47:312	Unplugging device FFFF870CA2567D60 on port 2
16858602	vhci	3796	4628	0	16414	03\25\2022-12:02:47:312	VDEV_VHCI: leave STATUS_SUCCESS
16858603	vhci	3796	4628	0	16415	03\25\2022-12:02:47:312	VDEV_VHCI: irql !0x00000000(Low)
16858604	vhci	3796	4628	0	16416	03\25\2022-12:02:47:312	VDEV_VHCI: irql !0x00000000(Low)
16858605	vhci	8808	7636	1	16417	03\25\2022-12:02:47:312	Enter irql 0x00000000(Low), write buffer 34470, irp a191e240
16858606	vhci	8808	7636	1	16418	03\25\2022-12:02:47:312	Leave 0xc000009d(STATUS_DEVICE_NOT_CONNECTED), transferred 0, irp a191e240
16858607	vhci	8808	2104	0	16419	03\25\2022-12:02:47:312	VDEV_VHCI: irql !0x00000000(Low)
16858608	vhci	8808	2104	0	16420	03\25\2022-12:02:47:312	Device is already unplugged, port 2
16858609	vhci	8808	2104	0	16421	03\25\2022-12:02:47:312	a3db9a50
16858610	vhci	8808	2104	0	16422	03\25\2022-12:02:47:312	a3db9a50
16858611	vhci	4	388	2	16423	03\25\2022-12:02:47:313	VDEV_VHCI: irql !0x00000000(Low)

but in the device manager the device was still there,so every time I must restart the PC

docterling, 2022-03-26:
in the bushound the usbip-win ude driver,hc and hub and device were tierd in tree view,like:
[image]

while the usbip-win2 driver，hc and hub and device look like:
[image]

yeap,no tree hierarchic,does it matter?
but the result is that usbip-win2 works find---usb speaker can play sound and usbip-win ude driver can't

vadimgrn, 2022-08-11:
Outdated, several releases have been issued

closed 2022-08-11 by vadimgrn