vadimgrn/usbip-win2 #7: usbipd

colemickens, 2022-02-19:
Are you aware of usbipd? I just used it last night to throw my keyboard and mouse over to my laptop. It works well from what I can tell. 

https://github.com/dorssel/usbipd-win

In my case, I forward the local 3240 to my laptop on port 13240 and then use `--tcp-port 13240` with my client commands.

realivanjx, 2022-05-19:
isnt this project has both client and server for windows and the other project only the server?

colemickens, 2022-05-19:
The README for `usbipd-win` shows uses binaries from the project to both share (bind) and receive (attach) USB devices.

realivanjx, 2022-05-19:
see https://github.com/dorssel/usbipd-win/issues/65

closed 2022-05-19 by colemickens

realivanjx, 2022-05-19:
btw which usbip client are you using?