vadimgrn/usbip-win2 #8: [question] is visual studio 2022 supported?

realivanjx, 2022-05-18:
im wondering if visual studio 2022 with the preview sdk and wdk are supported. or do i need to use visual studio 2019? thanks.

vadimgrn, 2022-05-18:
I'm waiting for release of WDK for VS-2022. I don't want to install preview versions of SDK/WDK.

vadimgrn, 2022-05-27:
VS2022 can be used to build "develop" branch.

docterling, 2022-05-31:
1、
windows 10 professional 2004(for this wsk version,I reinstall my windows to this version)
Microsoft Visual Studio Enterprise 2022 (64 bit) - Current - version 17.2.2
sdk version 10.0.22621.0
wdk version 10.0.22621.0
2、
Install driver/usbip_test.pfx (password: usbip)
"Trusted Root Certification Authority" in "Local Computer" (not current user) and
"Trusted Publishers" in "Local Computer" (not current user)
Set certificate driver signing for usbip_vhci project
bcdedit.exe /set TESTSIGNING ON and reboot
3、
in cmd with administrator priviledge
devcon.exe install usbip_root.inf USBIPWIN\root
after that in device manager,it looks like somethings wrong
[image]

and the property details:
Windows cannot load the device driver for this hardware. The driver may be damaged or missing. (Code 39)
{driver input point not found}
%hs device driver cannot find the input point %hs in the driver %hs.
any suggestions?
and the previous develop version(not wsk) ok for me

docterling, 2022-06-01:
yes,I also try that to
[image]
but the problem the same

docterling, 2022-06-02:
I set this,and it works now!
[image]

vadimgrn, 2022-06-02:
Thank you, I was going to revise these settings of the project.

docterling, 2022-06-02:
1.this one is when I change the sound volume over the task bar
[BSOD1.log](https://github.com/vadimgrn/usbip-win2/files/8822133/BSOD1.log)
2.this one is do attach-detach-attach,and do the log cmds
[BSOD2.log](https://github.com/vadimgrn/usbip-win2/files/8822150/BSOD2.log)

vadimgrn, 2022-06-08:
I've fixed several bugs that cause BSOD. I'm testing vhci driver using Driver Verifier and there are no issues so far. Could you please try your devices with new build?

docterling, 2022-06-09:
1. I do attach-change vol-detach，when attach windows will play the new usb device arriving sound,but The voice sounds trembling，and then I change the sound volume through task bar,the ui seem stuck and no sound,after that I do detach
[attach-change vol-detach.txt](https://github.com/vadimgrn/usbip-win2/files/8866613/attach-change.vol-detach.txt)
2. in 1 after detach I've found the server was crashed and I restart it,and I do attach and change sound volume again,it BSOD
[BSOD3.txt](https://github.com/vadimgrn/usbip-win2/files/8866638/BSOD3.txt)
3. just do attach-detach-attach
[BSOD4-attach-detach-attach.txt](https://github.com/vadimgrn/usbip-win2/files/8866650/BSOD4-attach-detach-attach.txt)

vadimgrn, 2022-06-09:
Please, make logs exactly as described in "Obtaining USB/IP logs on Windows". Your logs are hard to understand due to another formatting.

vadimgrn, 2022-06-09:
BSOD4-attach-detach-attach.txt caused by NT_ASSERT, I've commented it.
BSOD3.txt is very similar to behavior of the driver before recent fixes, make sure that you are building and testing the latest develop branch.

vadimgrn, 2022-06-09:
I've fixed bug that I found in BSOD5-change.vol.txt, thank you!
copy_isoc_data:src_offset(0) != src_len(192)

docterling, 2022-06-10:
I've tried,when attach it will failed on sending and then auto unplug,here is the etl log file
[LogSession_20220610_093143.zip](https://github.com/vadimgrn/usbip-win2/files/8875146/LogSession_20220610_093143.zip)
I am used to using the traceview.exe,You can navigate to the following line
vhci	wsk.events.cpp	634	WskDisconnectEvent	_**06\10\2022-09:34:03:423**_	vpdo 49ccb420, Flags 0x9

vadimgrn, 2022-06-10:
I can't and won't view binary file.

docterling, 2022-06-10:
I've just tried again
the strange thing is the device name in device manager,now it's
[image]
it used to be "AUCTOPUS I-65"
but the sound is ok now : )

vadimgrn, 2022-06-10:
Do you notice a difference between this WSK build and old version that uses usbip_xfer.exe ?

docterling, 2022-06-13:
xfer first connect to server and get the device-configration descriptor and then create vpdo
wsk first create vpdo and then connect
but when plugin xfer is not use the already got descrptor info
after IoInvalidateDeviceRelations the upper driver will URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE
I know the sting "AUCTOPUS I-65" is frome iProduct string in device descriptor,but I don't know the string "USB Audio Device" from where?

vadimgrn, 2022-08-11:
Fixed

closed 2022-08-11 by vadimgrn