vadimgrn/usbip-win2 #11: BSOD crash while trying to connect to AiM racing dash

unrivaledrides, 2022-07-10:
Hi, I've been using your usbip for a week or so, and it shows great promise from what else I've seen and tried. My application is a bit different than most, as I'm trying to ship USB data from a race car to tuning computers in a nearby trailer. My USB devices are an AiM dash, AiM logging camera (though video is stored on an SD card, not via USB), and an ECUMaster Power Management Unit. 
Server side: I've installed an industrial router (Teltonika 955) in the car that runs Ubuntu (kernel 5.4.147), and the USB devices mentioned above connect to the USB port on the router (via 3-1 non-powered splitter/hub). usbip list -l on the server shows:
 ---------------------------------
   Teltonika RUT9 series 2022
 ---------------------------------
root@Teltonika-RUT955:~# usbip list -l
 - busid 1-1.1.3 (11cc:2002)
   unknown vendor : unknown product (11cc:2002)

 - busid 1-1.3 (04b4:0003)
   Cypress Semiconductor Corp. : unknown product (04b4:0003)

 - busid 1-1.1.2 (11cc:0110)
   unknown vendor : unknown product (11cc:0110)

 - busid 1-1.4 (2c7c:0125)
   Quectel Wireless Solutions Co., Ltd. : EC25 LTE modem (2c7c:0125)

 - busid usb1 (1d6b:0002)
   Linux Foundation : 2.0 root hub (1d6b:0002)

 - busid 1-1.1 (05e3:0610)
   Genesys Logic, Inc. : Hub (05e3:0610)

 - busid 1-1.1.1 (0483:5740)
   STMicroelectronics : Virtual COM Port (0483:5740)

 - busid 1-1 (05e3:0610)
   Genesys Logic, Inc. : Hub (05e3:0610)

For the client side, I'm using an ordinary laptop running Windows 11. I use AiM and ECUMaster applications to access the USB data coming via usbip. All the devices work, at least for a while (15s to 30minutes), but all end up as a BSOD. Sometimes two devices will work at a time, though I've never had all three working at once. I've spent nearly all of the past week gathering software in order to capture the memory dump and trace from such a crash, and I finally did that. Windows is so much harder than Linux. Sigh.

In this case, the error was a hang during attach to device 1-1.1.2. 30s or so. When that happens CTL-C does nothing, so typically I go to the server and kill the bind, and when I did that this time, it resulted in a BSOD. 

C:\Users\jp>usbip list -r rut955
Exportable USB devices
======================
 - rut955
    1-1.1.1: STMicroelectronics : Virtual COM Port (0483:5740)
           : /sys/devices/platform/ahb/1b000000.usb/usb1/1-1/1-1.1/1-1.1.1
           : (Defined at Interface level) (00/00/00)
           :  0 - Communications/Abstract (modem)/AT-commands (v.25ter) (02/02/01)
           :  1 - (Defined at Interface level) (00/00/00)

    1-1.1.3: unknown vendor : unknown product (11cc:2002)
           : /sys/devices/platform/ahb/1b000000.usb/usb1/1-1/1-1.1/1-1.1.3
           : (Defined at Interface level) (00/00/00)

    1-1.1.2: unknown vendor : unknown product (11cc:0110)
           : /sys/devices/platform/ahb/1b000000.usb/usb1/1-1/1-1.1/1-1.1.2
           : (Defined at Interface level) (00/00/00)

I think this is all the data you request, but please let me know if it isn't:  
[unrivaledrides-bsod.zip](https://github.com/vadimgrn/usbip-win2/files/9079877/unrivaledrides-bsod.zip)

vadimgrn, 2022-07-11:
Hi, thank you for the feedback. I need you to get backlog of the BSOD as well as driver's log as described here [Debugging BSOD](https://github.com/vadimgrn/usbip-win2/tree/develop#debugging-bsod).

vadimgrn, 2022-07-11:
> In this case, the error was a hang during attach to device 1-1.1.2. 30s or so. When that happens CTL-C does nothing, so typically I go to the server and kill the bind, and when I did that this time, it resulted in a BSOD.
Hang during attach means that the driver is connecting to remote host which is down or unreachable. I'll try to reproduce this issue, but it would be better to get a full backtrace of BSoD.

vadimgrn, 2022-07-11:
To log gathering is simple. Use these tools from WDK [tracelog.zip](https://github.com/vadimgrn/usbip-win2/files/9076085/tracelog.zip) and this fresh debug build [usbip-win2-0.9.0.1-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/9078862/usbip-win2-0.9.0.1-Debug.zip).

vadimgrn, 2022-07-11:
> I go to the server and kill the bind, and when I did that this time, it resulted in a BSOD.
Is that means that you issues "usbip unbind -b 1-1.1.2" or literally kills usbipd?

vadimgrn, 2022-07-11:
I've managed to reproduce this BSOD, but it's very interesting to get backtrace of BSOD when devices are working.

unrivaledrides, 2022-07-11:
I'll work on getting a better trace now. To answer your question, yes, I unbound it from the server side: "usbip unbind -b 1-1.1.2". And that's great news that you've reproduced the BSOD.

As mentioned, my Windows skills are poor, but I'll get started now.

vadimgrn, 2022-07-11:
I'm developer on Linux only since 2009. This driver is a hobby :)

unrivaledrides, 2022-07-11:
I see you are a beginner on Linux, haha! I'm glad you've taken up this hobby. It can really help my little race team.

Here's another log of device 1-1.1.1 working correctly. I hope this is acceptable, and I'll keep working on it.

[usbip-vhci-pmu.zip](https://github.com/vadimgrn/usbip-win2/files/9084976/usbip-vhci-pmu.zip)

unrivaledrides, 2022-07-11:
Here's the log of the hang (or "freeze" as the youngsters say) on attach attempt to 1-1.1.2 (though not very interesting):

[usbip-vhci-aimDash.zip](https://github.com/vadimgrn/usbip-win2/files/9085254/usbip-vhci-aimDash.zip)

unrivaledrides, 2022-07-12:
Working on it now. I believe I'm doing exactly that, but obviously something is amiss.

unrivaledrides, 2022-07-12:
ok, let's see if this is more useful to you. It's a crash right at "attach" to 1-1.1.3 (the AiM camera):

[BetterBSOD.zip](https://github.com/vadimgrn/usbip-win2/files/9093652/BetterBSOD.zip)

unrivaledrides, 2022-07-12:
I'm still working on this. Can you tell me what the file extension for symbols are? pdb?

vadimgrn, 2022-07-12:
*.pdb are installed into the usbip-win2 folder by debug build

unrivaledrides, 2022-07-12:
I think I got it now. I see a file/function name, line number, and symbol name.

[maybe_this_is_it.zip](https://github.com/vadimgrn/usbip-win2/files/9094957/maybe_this_is_it.zip)

unrivaledrides, 2022-07-12:
AULTING_SOURCE_LINE:  D:\usbip-win2\driver\vhci\internal_ioctl.cpp

FAULTING_SOURCE_FILE:  D:\usbip-win2\driver\vhci\internal_ioctl.cpp

FAULTING_SOURCE_LINE_NUMBER:  341

FAULTING_SOURCE_CODE:  
No source found for 'D:\usbip-win2\driver\vhci\internal_ioctl.cpp'

SYMBOL_NAME:  usbip_vhci!`anonymous namespace'::control_vendor_class_request+3c

MODULE_NAME: usbip_vhci

IMAGE_NAME:  usbip_vhci.sys

STACK_COMMAND:  .cxr; .ecxr ; kb

BUCKET_ID_FUNC_OFFSET:  3c

FAILURE_BUCKET_ID:  0x1E_C0000420_usbip_vhci!_anonymous_namespace_::control_vendor_class_request

OS_VERSION:  10.0.22000.1

vadimgrn, 2022-07-12:
Don't know why you concatenate log from several WinDbg sessions. I see what causes this bug, will fix it later. Maybe all synchronous calls have to be modified to asynchronous, will see.

vadimgrn, 2022-07-12:
This is the assertion failure because IRQ level is higher than expected. I haven't seen this on my "usual" devices.

vadimgrn, 2022-07-13:
Try new build, pls
[usbip-win2-0.9.1.0-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/9102384/usbip-win2-0.9.1.0-Debug.zip)

unrivaledrides, 2022-07-13:
This looks promising. No crashes, at least so far. The connection was a bit fussy, but I was doing a "sanity" test (and it might even be on the host side).  More to come.

Many thanks!

vadimgrn, 2022-07-13:
I wonder why "usbip attach" command stucks in your environment. This means the driver is resolving host or connecting to it. If it can't do that, attach fails after ~30sec timeout. Have you tried use ip address instead of hostname?

unrivaledrides, 2022-07-13:
Yes -- that does happen a lot during my testing. I don't have DNS set up (yet), so I'm using a hard-coded IP address. (192.168.10.1 if that means anything).

unrivaledrides, 2022-07-13:
Here's a log of connection I had with the AiM "Smarty Cam". It worked fine for quite a while, though it eventually froze. I was able to download big chunks of "live camera" data.  I thought this might be useful for you to see how this USB device behaves, and how it differs from other devices. Lots of unresolved trace symbols, but I think that's because I'm only providing a formatter for the usbip stuff. 

[usbip-AimCamera.zip](https://github.com/vadimgrn/usbip-win2/files/9104264/usbip-AimCamera.zip)

unrivaledrides, 2022-07-14:
Just a comment on the "sticking attach": the host thinks everything is ok, and seems to be waiting on the client to begin requesting data. Here's all the data (not even worth creating a zip file:

Client:
[2]4eb8 TRACE_LEVEL_VERBOSE `anonymous namespace'::vhci_complete:VDEV_VHCI: irql !0x00000000(Low)
[2]4eb8 TRACE_LEVEL_VERBOSE vhci_ioctl:VDEV_VHCI: enter irql 0x00000000(Low), USBIP_VHCI_PLUGIN_HARDWARE(0X2A4000)

vadimgrn, 2022-07-14:
Use these *.bat files for testing.
dbg.bat installs the driver and starts tracing session
di.bat - driver install
dr.bat - driver remove
va.bat - start tracing session
vz.bat stops tracing session and makes trace log

The usual sequence is dbg.bat, testing (attach, ..., detach), vz.bat

Sometimes the driver can't be uninstalled due to corruption of unins000.exe. Run commands from uninstall.bat is such case.

I used to use [Far Manager](https://www.farmanager.com/download.php?l=en) for working with console commands. I run in with Admin rights
[bin.zip](https://github.com/vadimgrn/usbip-win2/files/9109624/bin.zip)
.

vadimgrn, 2022-07-14:
You have to do correct log files, otherwise I can't help you. usbip-AimCamera.txt is useless due to massive lack of information.

unrivaledrides, 2022-07-14:
Yikes! Sorry for all the trouble! Fixing the unbind BSOD will help a lot, since that should mean I can fuss with the usb connections from the host side.

unrivaledrides, 2022-07-14:
Here's the camera capture again, using your va/vz scripts. They do work better than my start_log/stop_log scripts (I changed nothing else). I gotta figure out what I was doing wrong!

Anyway this feed ran fine for 5 minutes or so, then froze. I detached the port, then captured the data.

[usbip-aimCamera.zip](https://github.com/vadimgrn/usbip-win2/files/9113584/usbip-aimCamera.zip)

unrivaledrides, 2022-07-14:
Here's the hanging attach, also with full symbols now. btw, there's a "sed' line in the vz.bat file that doesn't work. As near as I can tell, it has never worked... anyway, I hope this is more useful:

[usbip-AiMDash-Attach.zip](https://github.com/vadimgrn/usbip-win2/files/9115241/usbip-AiMDash-Attach.zip)

vadimgrn, 2022-07-15:
Install sed and it will work :)

vadimgrn, 2022-07-15:
usbip-AiMDash-Attach.zip - there is no hang on the client side, the driver sent request and was waiting for response that did't happen. Server side debugging is required. I recommend debug your devices one by one and do not use usb splitter, just plug in cam or whatever directly in the usb port of the router.

unrivaledrides, 2022-07-15:
ok, working on sed install. :) On the hang, if it matters: it works fine when I use the old usbip-win3 (https://github.com/cezanne/usbip-win). I'll see about adding debug to the host to see if I can pick anything up -- and I'll discontinue using the hub for the time being.

unrivaledrides, 2022-07-15:
Still no sed (no path in shell, I guess). I tried a different-ish experiment. It might be useful, it might not. I removed the hub from the host side, and tried to attach to the AiMDash. The attach hangs, as it has been. Then I unplugged the USB cable from the router.  The client side then said "successfully attached", and returned to the cmd prompt. Of course, it wasn't really attached, but I snagged the log from the host and the client. Maybe there's a clue here?  

Client:
[usbip-AiMDash-Attach-RemovePlug.zip](https://github.com/vadimgrn/usbip-win2/files/9121889/usbip-AiMDash-Attach-RemovePlug.zip)

vadimgrn, 2022-07-16:
In this case attempt to read non-existent string descriptor failed with -EPROTO error instead of -EPIPE. After that all requests were failed with EPROTO. This can be the issue with firmware that must return EPIPE for invalid requests. To avoid this I have to change reading of string descriptors at start.

vadimgrn, 2022-07-16:
Meanwhile I'm being trying to recover win pc for third day. I've tried bunch of Intel RST drivers, but the installer refuses to load them. It's weird that the latest Win11 installer has no required driver.

vadimgrn, 2022-07-16:
usbip attach just initiates creation of usb device on virtual hub. After that the driver returns control to usbip. The system issues PnP requests to new device and then starts it. If there is a driver in the system that can handle this device and there weren't errors, the device will appear on system.

unrivaledrides, 2022-07-16:
I think I understand: you're saying that the client has made a legitimate request, to which the correct host response should be EPROTO (host says device doesn't support this feature). Instead, the host responds with EPIPE. Client sees the EPIPE, but continues to wait for the EPROTO, which will never come.

vadimgrn, 2022-07-16:
Default control pipe returns EPIPE for erroneous requests. But this device returns EPROTO and fails all next requests too.

unrivaledrides, 2022-07-16:
Oh: I had it backward. And once it returns EPROTO, it continues to stay in that state. Got it. But it does go to "stub up", which must only mean it has begun talking to a client, not that it has "connected" the device to the client.

vadimgrn, 2022-07-17:
I've managed to reinstall Windows, but I can't work on this until Friday.

unrivaledrides, 2022-07-17:
No problem! This is supposed to be fun, right?  I'm trying to turn on host debug so that I can figure out where the EPROTO call and decision comes from. Talk with you soon.

vadimgrn, 2022-07-17:
It's not necessary, I'll rework reading of string descriptors and the issue will gone.

vadimgrn, 2022-07-17:
There are good chances that your devices will work, I haven't been spotted unresolvable issues so far.

unrivaledrides, 2022-07-17:
Thanks Vadym, that's good to hear. I'll work on securing the router in the vehicle, routing wires and power, etc.

vadimgrn, 2022-07-19:
Try new build, pls
[usbip-win2-0.9.1.0-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/9138924/usbip-win2-0.9.1.0-Debug.zip)

unrivaledrides, 2022-07-19:
OK, tried the new build and the AiMDash now works correctly. I even did some housekeeping on the Dash with it. The Camera also works, but I had to fuss a bit to get the Camera up after I was done working with the Dash. I can work with that, as is. 

However: 

Now the PMU doesn't work (it worked fine before this latest build). The attach/detach works fine and there is no BSOD. But the PMU configuration software says it can't find the PMU and doesn't display its data. I captured a log of the attach, 3-4 attempts by the software to talk with its device, and then a detach. 

[usbip-pmu.zip](https://github.com/vadimgrn/usbip-win2/files/9142835/usbip-pmu.zip)

vadimgrn, 2022-07-23:
[usbip-win2-0.9.1.0-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/9173651/usbip-win2-0.9.1.0-Debug.zip)

unrivaledrides, 2022-07-23:
Very nice! This works! I'm able to connect to all three devices now. The only caveat is that I had to hard connect the PMU once, and then after that I was able to connect the PMU via usbip. At this point I have a working solution, though there is one issue with the Camera: it drops and freezes occasionally. I'll snag a log of that and post it. But, for now, I'm thrilled. Many thanks!!

vadimgrn, 2022-07-23:
Excellent, thank you for cooperation! I'm interesting in making this driver robust.

unrivaledrides, 2022-07-23:
Here's a camera hang. It happens when I enable "Live View" mode, where the camera image is displayed on the tuning computer in near real-time. So lots of data coming very fast.  The PMU and Dash seem to work all day long now -- only the camera in Live View mode seems to be an issue.

[usbip-cameraHang.zip](https://github.com/vadimgrn/usbip-win2/files/9174002/usbip-cameraHang.zip)

vadimgrn, 2022-07-23:
FYI some devices as usb drives don't work after the second "usbip attach". The first attach is OK, device works, you run "usbip detach" and after the next "usbip attach" it doesn't work. The solution is to unbind/bind the device on server side before each next "usbip attach".

unrivaledrides, 2022-07-23:
That's good advice, and I'll follow that. I was still able to get the Camera Live View to hang. I'm checking with the manufacturer to see if that's something endemic to the device, and not the communications path.

vadimgrn, 2022-07-24:
This bug is harder to fix, the cam stops working after EPROTO error. I'm going to rewrite receiving of server's responses, this could fix it. This work will take several days, but now I'm free on weekends only.

unrivaledrides, 2022-07-24:
Sounds good. As I said, the current solution is pretty good. Should I pick up your newest release? Or is it the same as what I have?

vadimgrn, 2022-07-24:
It almost the same, nothing significant was added.

vadimgrn, 2022-08-10:
Could you please try new pre-release?
[usbip-win2-0.9.2.0-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/9299967/usbip-win2-0.9.2.0-Debug.zip)

unrivaledrides, 2022-08-10:
Sounds promising! I have to shuffle the cars around in the shop. I'll get to it later today or tomorrow.

unrivaledrides, 2022-08-13:
Sorry for the delay. This version is a definite improvement. I still lose the "Live View" video feed after awhile, but now the connection survives, so that I can re-do the Live Feed, or do other connected functions. Let me know if you want a log of the Live View.

vadimgrn, 2022-08-13:
I could analyze a log to find out what's wrong.

unrivaledrides, 2022-08-13:
OK, working on a log now.

unrivaledrides, 2022-08-13:
Here's a nice test case/bug recreation. I started logging, then attached to the remote host. Then I opened the configuration software for the camera and let all the ordinary data flow to the PC. Then I opened a "Live View". It came up for a bit and then died. Then I snagged this log:

[usbip-LiveViewHang.zip](https://github.com/vadimgrn/usbip-win2/files/9332098/usbip-LiveViewHang.zip)

[5 more replies]

closed 2022-08-17 by vadimgrn