vadimgrn/usbip-win2 #13: Signed driver

josephdunne-eaton, 2022-07-26:
Is there any hope of getting a signed driver?  My organization has policies set which disallow use of test signed drivers.

```
C:\Program Files\usbip-win2>bcdedit.exe /set testsigning on
An error has occurred setting the element data.
The value is protected by Secure Boot policy and cannot be modified or deleted.
```

vadimgrn, 2022-07-26:
I don't think so. A certification costs money which I'm not going to spend. The second reason is that the driver is actively developing and there are still a lot of changes.

vadimgrn, 2022-07-26:
The certification will be the primary goal when the driver will became stable.

josephdunne-eaton, 2022-07-26:
I see.  That is unfortunate.  Thanks for the prompt reply.

paulpv, 2022-09-21:
Consider that many games with ant-cheat enabled won't run if they detect Windows is in Test Mode.

Is there any reasonable workaround for PCs that want to trust this driver to not have to be run in Windows Test Mode?

It seems like test mode is required whenever actively attaching a binded device.

vadimgrn, 2022-09-21:
There is no workaround.

paulpv, 2022-09-25:
Thanks @vadimgrn!

Do you think it is possible for a fork **with the willingness to pay for its own certificate** [and set up a MS account and other obligatory tasks] and get this to work?
I am, and it looks like others are too, willing to help fund a 1 or multi-year cert.

I had recently been reading those and related links before you posted them. :)
I'll trod through them, but would love to find a simple but complete ~1 page readme on how to go through the signing process.

Is there any such a thing as "evaluation" EV code signing certs to try this out with to confirm if it will work and would be worth paying the money for? I suspect not, thinking the ability to legitimately sign code with an evaluation cert would defeat the purpose of code signing.

vadimgrn, 2022-09-26:
I'm sure that a certificate evaluation isn't available. The simplest way is to make [Windows 10 attestation signed drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-reqs#windows-10-attestation-signed-drivers). This means a driver can be used without enabling Test Signing Mode. You don't have to do Hardware Certification process which could take much time and efforts.

DzzD, 2023-02-08:
Hello, 

I am looking for a stable & easy version of USBIP for windows, the last "cezanne" release works great for me (used for laser engraver and its camera), except once disconnection/reconnection wich is unstable and does require to reboot  either windows and sometime the raspberry server, but this version does not require Test mode wich is great, from what I understood if the version require test mode it is require to stay in test mode as long as we need to use it ? that's it ? 

what rour plan for the futur of your version ? this is a really great product, do you have any stable release date ?

vadimgrn, 2023-02-08:
I do not have any release dates. This driver is pretty stable. Signing requires money to buy certificate which I'm not going to spend. Develop branch has UDE driver which is stable too.

DzzD, 2023-02-08:
>which I'm not going to spend
Seems logical, maybe it could be funded via GitHub sponsorship ? or a different kind of crowdfunding ? (or even selling this product once it is no more stamped as " probable BSOD"), I would understand to have to pay a bit to use it in a non-test mode

vadimgrn, 2023-03-11:
Those interested in a signed driver can donate to purchase EV Code Signing Certificate. I added two sponsorship methods.

nefarius, 2023-03-26:
An EV certificate is only sold to companies, not individuals, in case you didn't know. Also you will need an Azure AD tenant and register a Microsoft Partner account, those at least come with no costs as of writing.

Also the EV cert only grants you submission to Microsoft, it is no longer possible to self-sign kernel drivers with the EV cert directly, it has been killed off quite a while ago.

Cheers

vadimgrn, 2023-03-26:
Thank you for the information. If I can't buy an EV certificate, that driver will never be signed.

nefarius, 2023-03-26:
I shall keep my eye on this project then 😉

vadimgrn, 2023-06-18:
https://community.osr.com/discussion/292357/driver-signing-options-for-an-independent-developer

nefarius, 2023-06-18:
> Microsoft is not interested in hobbyist drivers. Drivers significantly impact system stability and security. They want drivers done by professionals.

This had me spit out my coffee laughing 🤣 I have followed the official fails of components by Microsoft, the "professionals" in this case, and as an open source developer myself have reported countless bugs and fixes to them over the years. This is why I avoid OSR, the elitism is so cringy, it hurts.

alexmi256, 2023-06-20:
> There is no hope to release a signed driver :( https://community.osr.com/discussion/292357/driver-signing-options-for-an-independent-developer

I may try to use a driver signing service once you feel it's stable enough that it's worth investigating

vadimgrn, 2023-06-21:
As far as I know, such services are no longer an option since Win10.

nefarius, 2023-06-21:
> As far as I know, such services are no longer an option since Win10.

Correct, only one way is left and that is via EV through Microsoft.

levelad, 2024-02-24:
Cheapest EV code signing certificate I have found: €749.00 gross for 3 years.

https://shop.certum.eu/certum-ev-code-sigining-code.html

nefarius, 2024-02-24:
Also make sure the cert provider is listed as supported by MS [as documented here](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-cert-manage#get-or-renew-a-code-signing-certificate) which Certum appears to be.

levelad, 2024-02-26:
Oh, I overlooked IdenTrust, it's even cheaper.

```
TrustID EV Code Signing | Organization Identity | Hardware Storage
3 Year
SafeNet USB Token use existing

Certificate $497.00
Storage $0.00

Total $497.00
Free USPS shipping within
the U.S. Additional fees may apply
for shipping outside of the U.S.
Expedited delivery is available.
```

But my problem is that there is no Windows server stub driver (cezanne) which is signed. And I don't want to set a production system in test signing mode. The cezanne attestation signed vhci driver for the client works fine.

Trying to pass a SmartCard reader from Server 2022 Hyper-V host to Server 2016 Hyper-V guest. Guess I have to buy a software like USB Redirector.

The whole certification process seems to be quite the hassle according to this blog post:

https://billauer.co.il/blog/2021/05/windows-drivers-attestation-signing/

Schuwi, 2024-03-19:
While working on something completely different I just stumbled upon [WinBtrfs](https://github.com/maharmstone/btrfs) and was reminded of usbip-win2 when I saw that their driver is apparently signed.

I searched/skimmed the relevant issues and found these:
https://github.com/maharmstone/btrfs/issues/35 (especially https://github.com/maharmstone/btrfs/issues/35#issuecomment-331540153)
https://github.com/maharmstone/btrfs/issues/270

Thought I'd just drop these here, maybe they help?

nefarius, 2024-03-19:
My two cents on this topics since I've walked the walk since Windows 7...

- You only need a code-signing certificate, not - contrary to what certain companies might say - a more expensive EV certificate.
  - This claim was at one point correct but is now completely obsolete and does not apply. What you could do in the past was so called cross-sign the driver binaries. You could get a code signing certificate from a CA Microsoft had offloaded trust to which you could sign with their cross cert and your own, so the kernel could use that chain of trust to load the driver without MS's involvement. This is a dead approach and doesn't work on any recent vanilla Windows 10/11 installation. Also these cross certificates have not been renewed and have all expired so you couldn't even sign new drivers for older OSs without trickery of forging timestamps, which makes you look like a malware author and is not practical for production releases.
- An EV certificate is only required on the latest versions of Windows 10, but you can get round this by disabling Secure Boot
  - Well, Windows 10 and 11 are the defacto-standard now since all older major editions have gone EOL and expecting users to change some settings they might not even have the permissions to (like company devices) is not realistic for production releases of drivers. It may suffice in a small lab environment but not for the masses.
- Even if you have an EV certificate, you don't (contrary to what Microsoft say) need to go through WHQL to get it to work (according a thread on Dokan's pages, anyway)
  - EV or not never had anything to do with the WHQL process and IIRC wasn't claimed anywhere on the official docs.
- WHQL apparently rejects out-of-hand any driver released under an open-source licence (because they can)
- You are unlikely to be able to get an EV certificate unless you have a company
  - The whole idea behind EV is to only be purchasable by corporate entities, if we ignore a black market situation you can not acquire one as an individual or hobbyist at all.
- If you are writing open-source drivers, the best place to get a certificate is Certum. Expect to pay roughly €100 for a card reader, and €30 a year for the certificate.
  - You can sign your driver binaries, they simply will just not load on 99% of the machines of your customers/users then 😉 So why even bother.
- You need to sign both your sys file and your cat file. Your cat file is created from your inf file by the Microsoft tool inf2cat. There's a good but wordy guide at [http://www.davidegrayson.com/signing/](http://www.davidegrayson.com/signing/).
  - Windows 10/11 no longer accepts self/cross-signed CAT files without at least displaying a warning box to the user, or depending on SecureBoot etc. will outright deny installation altogether.

**TL;DR:** there only exist two official ways to get a kernel driver signed for modern Windows editions (ignoring exploits, stolen certificates, timestamp tampering, UEFI hacks and whatnot):
- You (your company) gets an EV cert which gives you the power to submit your driver to the Microsoft Partner Portal where you can request it to get attestation signed, which will generate a new CAT file and sign both the CAT and the SYS file with Microsoft's own internal keys. You may or may not sign the .SYS with your EV cert, you can do that for some extra authenticity but the kernel will not care if you do, it will only accept that Microsoft attestation signature anyway.
- You (your company) gets an EV cert which gives you the power to submit your **WHQL (HLK) test results package** to the Microsoft Partner Portal where you then get back files signed for all operating system versions you ran the tests for. This is almost impossible for a hobbyist or FOSS creator to achieve, unless you have the time, equipment and insights into this verbose process. And yes, each time one bit of your driver changes you need to run the entire test battery, upload to portal, react to errors etc. all again.

**Extra TL;DR:** IMHO any approach requiring your user to change stuff like SecureBoot or Code Integrity registry settings or policies, you have already lost. Anti Virus and Anti Cheat solutions more and more look for these options and will cause collateral problems which will force the user to either remove your driver or to give up on their game or whatever DRM protected solution. You can take a guess what most people will abandon first if they're forced to do so 😉 

Cheers

cooljimy84, 2024-03-19:
Pass the device to the host then give pass it through to the guest ? Then test mode is on on the host allowing the driver/device, then share it using hyper-v to the guest ? (RemoteFX USB device or something)

nefarius, 2024-03-19:
Using Hyper-V implicitly turns on features like Code Integrity, Memory Integrity etc. I assume you'd need to have your hypervisor host run in test mode then (I have no idea if that is even possible, never tried it before). Even if so, not practical except for the 5 people world-wide who'd be fine running such a setup.

ashleyw-gh, 2024-04-27:
The lack of a signed driver is sadly a no go for most people. Is there a possibility for the code to be signed by this organisation?
https://signpath.org/

fredemmott, 2024-04-29:
> The whole idea behind EV is to only be purchasable by corporate entities, if we ignore a black market situation you can not acquire one as an individual or hobbyist at all.

This isn't really the case: like IV code signing certificates more broadly, several CAs have been willing to issue EV code signing certs to sole proprietorships, but they've not advertised it. The flow usually goes "buy it, then contact support", though ssl.com is now advertising this: <https://www.ssl.com/certificates/iv-ev-code-signing/buy/>

While this doesn't require an LLC/Ltd./other distinct corporate entity, it can require registering "I am doing business under this name" or similar; for example, you can register a 'doing business as' name with Texas with form 503 for $25 every 10 years.

fredemmott, 2024-04-29:
> The lack of a signed driver is sadly a no go for most people. Is there a possibility for the code to be signed by this organisation? https://signpath.org/ https://about.signpath.io/product/open-source "Under the umbrella of the SignPath Foundation, open source projects can apply for a free code signing certificate. In order to use the free certificate, the build process has to be fully automated and integrated with SignPath.io, to ensure that the resulting binary results directly from the source code checked into the repository"

SignPath is not (yet?) an option: https://about.signpath.io/product/editions

[image]

ashleyw-gh, 2024-04-30:
thanks for seeing that. Because of all of this and no easy solution, I've taken a different approach.
For anyone interested...
I dug out an old RPI 3B I had in one of my IT cupboards, and I've stuck 2 dongles on that (ANT+ and Bluetooth for cycling equipment) and then used Virtualhere software with the RPI as the USB "server" and my remote GPU accelerated machine as the client. That way I can use a combination of Moonlight as a UI frontend to a Sunshine service (Nvidia Gamestream equivalent) on a remote windows VM with RTX3070 GPU passthrough on ESXi.
I was nervous of using VirtualHere because the license is locked to a single server, but I thought using the RPI just for that purpose means the worst that can happen is the SD card in RPI might die but the license is tied to the RPI serial number which persists for the life of the device even after SD card swaps.
I always prefer to use opensource for this type of connectivity, but ultimately US$49 to me was worth the price rather than spending hours of time and money battling with signing kernel drivers for windows.
If at some stage in the future this usbip-win2 fork can be signed I'll re-look to see how it compares to VirtualHere.

If anyone is interested about 14 years ago we were dealing with hardware security dongles for bank software development and at the time we ended up running with;
a Digi AnywhereUSB (G2) but then found a Belkin F5L009 that was about 1/5th of the price.
but again these solutions are dependent on the continued availability of signed kernel drivers for windows and even back then problems of device drop outs etc were common.

eriklundh, 2024-04-30:
I am looking into attestation signing for another project. 
I found this recent microsoft post: https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-attestation

alexmi256, 2024-05-06:
For those using Windows 11 below 23H2 you might be able to use https://github.com/Flerov/TS-Changer to change signing mode on the fly.
I also found a seller on Taobao supposedly offering EV Signing (782309659071) for $320US but I did't ask for details.

eebssk1, 2024-10-25:
> For those using Windows 11 below 23H2 you might be able to use https://github.com/Flerov/TS-Changer to change signing mode on the fly. I also found a seller on Taobao supposedly offering EV Signing (782309659071) for $320US but I did't ask for details.

I think they only sign the binary with the certificate itself. Not through windows hardware certifcating lab. You nned to sign through the WHQL lab for the driver to load.

eriklundh, 2024-10-25:
I recently learned from an independent dev, and verified it with a senior
MS engineer, that you can sign up for developer accounts for Microsoft
Store that brings you some kind of EV signing path for apps.
It seems to be a one time fee of about 20 USD for an individual dev, about
100 USD for a company.
The trick is that the "Microsoft Store" signs the package, even if the
price is 0 USD.
Once your identity is verified, you, or your buildbot, can upload your
installer package to Microsoft Store, there is some automated vetting, then
it gets signed, apparently with a one-year certificate.
You can point to Microsoft Store, or download the package with WinGet and
put it as a signed package on your own website, - or Github.
Just beware of the expiration date of each installer package.

https://learn.microsoft.com/en-us/windows/apps/publish/partner-center/account-types-locations-and-fees

I have read elsewhere that Microsoft plans to stop distributing third party
drivers with Windows Update.
Instead MS wants vendors to distribute through Microsoft Store, without any
requirement that a package should be a cost to the end user.

But I have not yet tried the Microsoft Store route to code signing myself,
since I am currently on my third month trying to get Azure Trusted Signing
to work for signing my own apps, but that is about 10USD per month -
forever.

/Erik

delebash, 2024-11-30:
As another user stated https://signpath.org/ is free for open source and looks like it is easy to use with github @vadimgrn   Any plans for this?  TY.

eebssk1, 2024-11-30:
> As another user stated https://signpath.org/ is free for open source and looks like it is easy to use with github @vadimgrn Any plans for this? TY.

kernel signing requires a EV certificate and it's only for company.

eebssk1, 2024-11-30:
@vadimgrn consider lock or move the issue to a discussion and clarify that personal certification is useless for driver signing.

vadimgrn, 2024-11-30:
There is nothing to discuss only the company may purchase EV certificate.

nefarius, 2024-12-01:
> The most realistic way that someone who owns the company will build and sign the driver out of goodwill or for the price of an EV certificate.

Maybe I know someone who knows someone...

eebssk1, 2024-12-01:
In the meantime,there are other way to load the driver without a ev signer.

Allow you load unsigned/self signed drivers without test mode: https://github.com/Mattiwatti/EfiGuard
Become the certificate root of your system and load any drivers signed by yourself: https://github.com/valinet/ssde

Fjox, 2024-12-08:
> In the meantime,there are other way to load the driver without a ev signer.
> 
> Allow you load unsigned/self signed drivers without test mode: https://github.com/Mattiwatti/EfiGuard Become the certificate root of your system and load any drivers signed by yourself: https://github.com/valinet/ssde

# DO NOT USE ANY OF THESE OPTIONS. THEY BRICKED MY SYSTEM!

eebssk1, 2024-12-08:
> # DO NOT USE ANY OF THESE OPTIONS. THEY BRICKED MY SYSTEM!

This only means you didn't follow the guide correctly.

Me and the hack makers are using it without any issue.
And only saying "THEY BRICKED MY SYSTEM" without any steps or logs or reproduces, no one can help you.

Fjox, 2024-12-08:
> > > In the meantime,there are other way to load the driver without a ev signer.
> > > Allow you load unsigned/self signed drivers without test mode: https://github.com/Mattiwatti/EfiGuard Become the certificate root of your system and load any drivers signed by yourself: https://github.com/valinet/ssde
> > 
> > 
> > # DO NOT USE ANY OF THESE OPTIONS. THEY BRICKED MY SYSTEM!
> 
> This only means you didn't follow the guide correctly or fully check the readme.
> 
> Me and the hack makers are using it without any issue. And only saying "THEY BRICKED MY SYSTEM" without any steps or logs or reproduces, no one can help you.

I ran the executable, it prompted me to restart and then started boot looping and somehow messed up my grid for that vm which shouldn't be possible

eebssk1, 2024-12-09:
> I ran the executable, it prompted me to restart and then started boot looping and somehow messed up my grid for that vm which shouldn't be possible

I suppose you are trying ssde? check https://github.com/valinet/ssde/issues/4
pls report any problems to corresponding repo.

nefarius, 2024-12-09:
> Hello I don´t understand in which way this version should be better than the cezanne/usbip-win version which runs perfectly with no need to go into server testsigning mode. And if there are advantages why is it not possible to use the driver from this project?

Start you own **Discussion** then instead of hijacking unrelated conversations, thank you.

vadimgrn, 2024-12-09:
> Hello I don´t understand in which way this version should be better than the cezanne/usbip-win version which runs perfectly with no need to go into server testsigning mode. And if there are advantages why is it not possible to use the driver from this project?

https://github.com/vadimgrn/usbip-win2/tree/develop?tab=readme-ov-file#differences-with-cezanneusbip-win

First of all,use what works for you.
cezanne/usbip-win is abandoned, the development was halted three years ago. It has tons of bugs, poor performance, poor quality of code, absence of GUI and installer, etc. The only reason to use it today is that the driver is signed.

SummerSigh, 2024-12-10:
Hi there, im from the Project Babble team https://github.com/Project-Babble/ProjectBabble which is an application for VR mouth tracking. We'd be more than happy to put the money upfront for a signed driver.

vadimgrn, 2024-12-11:
If you can build and sign the driver, it will be the best option.

f8ith, 2024-12-30:
User-mode drivers don't need to be signed by Microsoft. Porting the project from KMDF to UMDF would be pretty tedious I assume - but it should run on vanilla Windows without needing an EV certificate then

nefarius, 2024-12-30:
> User-mode drivers don't need to be signed by Microsoft. Porting the project from KMDF to UMDF would be pretty tedious I assume - but it should run on vanilla Windows without needing an EV certificate then

User-mode driver signing requirements will only further tighten as time and Windows Updates advance. Also UDE is and never will be available to user-mode, so no, 90% of this project in fact can not be ported to UMDF.

ambitiousCC, 2025-01-13:
> If you can build and sign the driver, it will be the best option.

I am not very familiar with the code of this project. There are multiple drivers in the project, do I need to sign all of them?

vadimgrn, 2025-01-13:
usbip2_ude.sys and usbip2_filter.sys

puetzk, 2025-01-17:
It looks like you may have an offer (thanks @SummerSigh!), but if it doesn't pan out here's another idea I haven't seen proposed above. There are RedHat folks involved with https://www.linux-kvm.org/page/WindowsGuestDrivers / https://github.com/virtio-win/kvm-guest-drivers-windows who regularly make signed releases. Interest in usbip device pass-through to windows guests seems like it might overlap quite a bit with other kvm paravirtualization needs, so perhaps they would be interested in a similar collaboration.

Redhat provides WHQL signed signed stable releases of those virtio drivers to paying customers as part of RHEL, and donates attestation-signed builds of each upstream release to the public via https://fedorapeople.org/groups/virt/virtio-win. This at least means they already have all the certificates and other Microsoft account setup for driver signing, and some operation in that space. I don't have any direct connection, but @vrozenfe and @crobinso are names I in the https://fedorapeople.org/groups/virt/virtio-win/CHANGELOG making releases for virtio. It might be worth trying to reach out to them and see if any similar collaboration would be possible...

Or if anyone here looking for usbip is already an customer for the RHEL virtualization tools, customers making feature requests would probably carry more weight than github discussion...

scheibling, 2025-04-19:
@vadimgrn We're currently going through and verifying some stuff before attestation signing the driver, quick question, have you run any of the HLK tests on this previously? (Or anyone else) 

And once we have the signed driver, if you'd be willing and able to just test it as well before we release it here that would be amazing, just as an extra peace of mind for us so we don't release a bad build.

vadimgrn, 2025-04-20:
Here is result of today's HLK test.

[image]

But I suspect this is not enough. I don't know what must be selected for UDE-based driver, see screenshot.

[image]

The other issue is usbip2_filter driver. I must be tested too, I haven't tested it yet. But it is much more simple than the usbip2_ude driver.

Sure thing, I will test signed driver if you build it.

scheibling, 2025-04-20:
Amazing, that gives us a great starting point! I'll get back to you sometime around end of next week hopefully

It's mostly about identifying potential large/fatal issues for us at this point, attestation signing has a lighter requirement than full-on WHQL. But you've done a great job with this from what I can see up to this point, thank you for your dedication!

scheibling, 2025-04-21:
@vadimgrn Could you send me an email at github@scheibling.se when you get a chance? No hurry

scheibling, 2025-04-27:
So in case someone missed it, there is now a package with the signed drivers, meaning you can run them without the testsigning option enabled.. A couple of notes regarding the current state:

## ARM
We are currently not in a position to be able to test the ARM drivers, when I get a chance I'll spin up an ARM VM and run through everything there as well.

## Attestation signing
The drivers are attestation signed, which means they are **not** certified under WHQL but Windows will trust the source. This also means there are no guarantees regarding compatibility - this is still a pre-stable release.

There are some limitations with attestation signing, see here for more information: 
https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-reqs#windows-10-attestation-signed-drivers-for-testing-scenarios

The current release supports **only x64** and the following Windows releases.

[image]

**Windows server installations do not accept attestation signed drivers**, for a complete compatibility table see here: https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-reqs#windows-driver-signing-requirements

The next step would be to have a go at actually certifying these drivers through the WHQL program.

## Company Name
Since the drivers are signed via our Microsoft partner account, you will se our company name on the emulated host controller:

[image]

## Installer & binaries
The installer in the release and the binaries are still signed with the repository certificate as usual, in most cases having those unsigned is not an issue. If you for some reason need a signed installer/signed binaries, we have an alternate build of the installer and the individually signed files on [ossign/vadimgrn--usbip-win2](https://github.com/ossign/vadimgrn--usbip-win2/releases).

## Trust
Since the signatures happen locally in our environment before we manually upload the files, there is of course no guarantee we haven't made any changes before signing. But we are working on that part, the plan on our end is to set up a public build action for compiling everything before we sign so that you can follow the build process on Github. But for now - all the files are signed with our EV certificate, and even though the trust we place in those certificates is debatable at best if you have any issues or would like to have any more information about how we do the build process feel free to reach out.

And a short final note (and shameful plug), we are opening up for applications on [ossign.org](https://ossign.org), if you have an open source project with Windows binaries, DLLs or drivers that you would like signed feel free to open a PR on our Github or send us an application through the contact form on the website.

vadimgrn, 2025-04-27:
Actually drivers pass WHLK tests. The issue is that an installation of test  server and client for each Windows version is time consuming and should be done eight times (from win10 1809 to win11 24h2). I have only one laptop for dedicated test client and must reinstall Windows for each test. I don't have Win11 ARM64 either.

scheibling, 2025-04-27:
@vadym would it help to have access to a VM or Msft Devbox? Just haven't had time to set up an ARM server since we don't have those templates ready

[33 more replies]