vadimgrn/usbip-win2 #14: Why prefer WDM over UDE?

tfgstudios, 2022-09-05:
Hey,

I'm by no means a driver developer or such, but I was wondering why did you choose to focus on WDM, as UDE is the tech that supersedes it, according to [wikipedia](https://en.wikipedia.org/wiki/Windows_Driver_Model#Criticism).
Cheers

vadimgrn, 2022-09-06:
Hi, because WDM driver was working for devices that I need. Every driver is a WDM driver. UDE is just a layer above KMDF which is a layer above WDM :)

Currently I have just one concern about WDM driver. I don't know how to implement USB 3.0 hub. UDE driver implements it for you. Maybe I will write brand new UDE driver, I don't want to deal with this driver from parent repo.

tfgstudios, 2022-09-07:
> Hi, because original WDM driver was working for devices that I need. Every driver is a WDM driver. UDE is just a layer above KMDF which is a layer above WDM :)
> 
> Currently I have just one concern about WDM driver. I don't know how to implement USB 3.0 hub. UDE driver implements it for you for free. Maybe I will write brand new UDE driver, I don't want to deal with this driver from the parent repo.
> 
> Beside that I like WDM for simplicity and usage of pure kernel APIs.

That makes sense. Thank you for taking the time to explain!