vadimgrn/usbip-win2 #15: Win10 BSOD when connecting to mouse on Win10 host running https://github.com/dorssel/usbipd-win

paulpv, 2022-09-20:
I am trying a mix of your `usbip` client talking to https://github.com/dorssel/usbipd-win's `usbipd` host.

On the host I bind the mouse and start the daemon:
```
C:\Program Files\usbipd-win>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
8-2    1b1c:1b8b  USB Input Device                                              Not shared

C:\Program Files\usbipd-win>usbipd bind -b 8-2

C:\Program Files\usbipd-win>usbipd server
....
```

On the client I list and attach:
```
C:\Program Files\usbip-win2>usbip.exe list -r 10.0.0.11
Exportable USB devices
======================
 - 10.0.0.11
        8-2: Corsair : unknown product (1b1c:1b8b)
           : USB\VID_1B1C&PID_1B8B\1602D03FAF7A04C55F40C0F0F5001C06
           : (Defined at Interface level) (00/00/00)
           :  0 - Human Interface Device/Boot Interface Subclass/Mouse (03/01/02)
           :  1 - (Defined at Interface level) (00/00/00)

C:\Program Files\usbip-win2>usbip.exe attach -r 10.0.0.11 -b 8-2
```
The connection is established, but soon after the client PC BSODs in (saving Issue here and will edit after reproing again...)

paulpv, 2022-09-20:
I followed @vadimgrn 's instructions at https://github.com/vadimgrn/usbip-win2/issues/11#issuecomment-1181589639 and here is the log.
[BSOD_usbip2_vhci.txt](https://github.com/vadimgrn/usbip-win2/files/9611892/BSOD_usbip2_vhci.txt)
Hope it is good enough.

paulpv, 2022-09-21:
Duh, I now also see the comments in the readme.md (RTFM) about how to collect logs.
Let me know if the above logs are OK; if not I will get a better one.

vadimgrn, 2022-09-21:
You should make a log as described in README.md

vadimgrn, 2022-09-21:
Fixed, works. It could be considered as the issue of usbipd because genuine Linux server behaves differently. The workaround has applied.

[usbip-win2-0.9.3.1.zip](https://github.com/vadimgrn/usbip-win2/files/9614713/usbip-win2-0.9.3.1.zip)

vadimgrn, 2022-09-21:
Run as Admininstrator these commands to remove the driver manually
```
pnputil /remove-device /deviceid USBIP\root /subtree
FOR /F %P IN ('findstr /m "CatalogFile=usbip2_vhci.cat" C:\Windows\INF\oem*.inf') DO pnputil.exe /delete-driver %~nxP /uninstall
```

After that remove folder "C:\Program Files\usbip-win2\".

paulpv, 2022-09-25:
Thanks @vadimgrn Looking in to this today....

paulpv, 2022-09-25:
@vadimgrn Confirmed works! Thanks!

vadimgrn, 2022-09-26:
Fixed

closed 2022-09-26 by vadimgrn