vadimgrn/usbip-win2 #16: Driver not loading - Driver Entry Point Not Found

pro-spero, 2022-09-21:
Installing the driver on Win10 I cannot get the driver to load with the error:
```
Windows cannot load the device driver for this hardware. The driver may be corrupted or missing. (Code 39)

{Driver Entry Point Not Found}
The %hs device driver could not locate the entry point %hs in driver %hs.
```
I tried with several of the release packages, 9.0.1 up to 9.3.0.
Am I doing something wrong or is something missing from the driver or system?

vadimgrn, 2022-09-21:
What version of Windows?

pro-spero, 2022-09-21:
It identifies itself as Windows 10 2016 LTSB. It should be 64bit, but I shall recheck as soon as I have access again, if this could be the issue (I would expect a more brutal message in case of bit mismatches ;) ).
Is there some way in which I could debug this further or provide more detailed information about this without installing development tools?
Thank you in advance for your time, it is very much! :)

pro-spero, 2022-09-21:
I have confirmed that it is a x64 version.

vadimgrn, 2022-09-21:
I haven't found what is Win 10 2016, but it seems that it isn't compatible with the driver.

https://learn.microsoft.com/en-us/windows/release-health/release-information

vadimgrn, 2022-09-21:
It will work for Windows 21H2 or 2004.

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-rtlisntddiversionavailable

vadimgrn, 2022-09-21:
[image]

pro-spero, 2022-09-21:
I only found this:
https://learn.microsoft.com/en-us/lifecycle/products/windows-10-2016-ltsb
I shall have to ask around if it can be mapped to "more regular" versions.

vadimgrn, 2022-09-21:
Find out OS build number

vadimgrn, 2022-09-22:
Make sure that Windows Test Signing Mode is enabled

pro-spero, 2022-09-22:
The OS build number is 14393.1198, Version 1607.
Test Signing is currently blocked by policy and will take quite some time to alter.
I running without enforcement of driver signatures, e.g., I am able to load test signed drivers. If the system boots with enforced driver signatures, the error code is 52; for this and but also other drivers.

vadimgrn, 2022-09-22:
Supported version is 2004(OS build 19041) and later. Your OS is not supported.

pro-spero, 2022-09-22:
I understand, thank you for helping me finding the cause of the error message!

closed 2022-09-22 by pro-spero