vadimgrn/usbip-win2 #18: Working device report

qmega, 2022-11-23:
I'm not sure what the criteria are for listing a device as "known to work" in the wiki. For what it's worth, I've used usbip-win2 0.9.3.2 to attach `1b35:1019 Anthem ARC Microphone` and it worked without a hitch as far as I could tell, so perhaps that can be listed as a working device. I was able to complete room correction with [ARC Genesis](https://anthemarc.com/arc-genesis/index.php) -- which is the device's purpose -- on a Windows VM using a microphone exported by a Linux box with usbipd.

vadimgrn, 2022-11-23:
Thank you! I'll add your device to the list. There is no special criteria. If someone claims that it works, I can't prove that anyway without this device.

vadimgrn, 2022-11-23:
Added as audio device

closed 2022-11-23 by qmega