vadimgrn/usbip-win2 #19: I tried installing on Windows Server 2019 - Getting compatibility error

migur0sa, 2022-12-16:
So I tried installing it and I get "This program is not compatible with this version of Windows."

- I tried running it as Windows 9 since that was the latest option on the "Compatibility Tab"

Is there any fix for this? Or is it just not compatible with Windows Server 2019

Thank you :)

vadimgrn, 2022-12-17:
Try this build, please
[usbip-win2-0.9.3.2-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/10251041/usbip-win2-0.9.3.2-Debug.zip)

I removed this declaration from InnoSetup
MinVersion=10.0.19041

The minimal supported version is Win10 v.2004(10.0.19041). I don't know how this version is corresponded with Server versions.

vadimgrn, 2022-12-17:
I see that the latest release is 10.0.17763 / October 2, 2018

This version is not supported.

migur0sa, 2022-12-19:
I will try it today and let you know how it went! Thank you so much :) I went into the source code and I was trying to change it myself, but I felt I was going to break something.

vadimgrn, 2022-12-20:
You should just replace ExAllocatePool2 calls with ExAllocatePoolWithTag taking into account ExAllocatePool2's flags.

closed 2023-01-20 by vadimgrn