vadimgrn/usbip-win2 #20: usbip: error: can't connect to domain.com

adelin-mcbsoft, 2022-12-18:
Hi,

I have an USBIP Server running on *Ubuntu Core 22.04* on *Raspberry PI 3 Model B+*, with 2 peripherals attached (a *CH340 Serial Converter* and a *CP2102 USB to UART Converter*).

In the very beginning, I used the [cezanne/usbip-win](https://github.com/cezanne/usbip-win) binaries on *Windows 10*, and it was working with *CH340*, but whenever I attached the *CP2102* the *USB 3.0 Hub* ceased to function -- I was seeing an exclamation mark in *Device Manager* and the ports were not available. Running only with *CH340* was working flawlessly).

So I decided to give your binaries a try.
Server-wise the configuration remained the same.

I used a fresh install of Windows and ran all the necessary steps to install the USBIP Client.

Whenever I try to list the devices, it works.
On Windows I run `C:\Windows\system32>usbip.exe --tcp-port 16125 list -r domain.com` and I receive the device list:

```
usbip: info: using port "16125"
Exportable USB devices
======================
 - domain.com
    1-1.1.3: Silicon Labs : CP210x UART Bridge (10c4:ea60)
           : /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3
           : (Defined at Interface level) (00/00/00)

    1-1.1.2: QinHeng Electronics : CH340 serial converter (1a86:7523)
           : /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2
           : (Defined at Interface level) (00/00/00)
```

Also, since I run the server with `-d` option, I also see that it's exported correctly:
```
usbipd: debug: usbipd.c:575:[do_standalone_mode] heartbeat timeout on ppoll()
usbipd: debug: usbipd.c:569:[do_standalone_mode] read event on fd[0]=3
usbipd: info: connection from XX.XX.XX.XX:63887
usbipd: info: received request: 0x8005(4)
usbipd: info: exportable devices: 2
libusbip: debug: usbip_common.c:114:[dump_usb_device] path                 = /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3
libusbip: debug: usbip_common.c:115:[dump_usb_device] busid                = 1-1.1.3
libusbip: debug: usbip_common.c:121:[dump_usb_device] Device(C/SC/P)       = (Defined at Interface level) (00/00/00)
libusbip: debug: usbip_common.c:123:[dump_usb_device] bcdDevice            = 100
libusbip: debug: usbip_common.c:128:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (10c4:ea60)
libusbip: debug: usbip_common.c:130:[dump_usb_device] bNumConfigurations   = 1
libusbip: debug: usbip_common.c:131:[dump_usb_device] bNumInterfaces       = 0
libusbip: debug: usbip_common.c:133:[dump_usb_device] speed                = Full Speed(12Mbps)
libusbip: debug: usbip_common.c:136:[dump_usb_device] busnum               = 1
libusbip: debug: usbip_common.c:137:[dump_usb_device] devnum               = 5
libusbip: debug: usbip_common.c:114:[dump_usb_device] path                 = /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2
libusbip: debug: usbip_common.c:115:[dump_usb_device] busid                = 1-1.1.2
libusbip: debug: usbip_common.c:121:[dump_usb_device] Device(C/SC/P)       = unknown class / unknown subclass / unknown protocol (ff/00/00)
libusbip: debug: usbip_common.c:123:[dump_usb_device] bcdDevice            = 264
libusbip: debug: usbip_common.c:128:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (1a86:7523)
libusbip: debug: usbip_common.c:130:[dump_usb_device] bNumConfigurations   = 1
libusbip: debug: usbip_common.c:131:[dump_usb_device] bNumInterfaces       = 0
libusbip: debug: usbip_common.c:133:[dump_usb_device] speed                = Full Speed(12Mbps)
libusbip: debug: usbip_common.c:136:[dump_usb_device] busnum               = 1
libusbip: debug: usbip_common.c:137:[dump_usb_device] devnum               = 4
usbipd: info: request 0x8005(4): complete
usbipd: debug: usbipd.c:569:[do_standalone_mode] read event on fd[0]=3
usbipd: info: connection from XX.XX.XX.XX:63889
usbipd: info: received request: 0x8003(4)
usbipd: info: found requested device: 1-1.1.2
usbip: info: connect: 1-1.1.2
usbipd: debug: usbipd.c:152:[recv_request_import] import request busid 1-1.1.2: complete
usbipd: info: request 0x8003(4): complete
usbipd: debug: usbipd.c:575:[do_standalone_mode] heartbeat timeout on ppoll()
```

But when I try to attach a device, I receive the following error:
```
C:\Windows\system32>usbip.exe --tcp-port 16125 attach -r domain.com -b 1-1.1.2
usbip: info: using port "16125"
usbip: error: can't connect to domain.com:16125
```

I also enabled debug mode `-d` to see if I get more info:
```
C:\Windows\system32>usbip.exe -d --tcp-port 16125 attach -r domain.com -b 1-1.1.2
usbip: info: using port "16125"
usbip: debug: usbip.cpp:104:[run_command] running command: attach
usbip: debug: vhci.cpp:51:[vhci_driver_open] device path: \\?\usbip#ehci#1&79f5d87&0#{b8b60941-cacb-454a-a8d1-35acb8fa1f1e}
usbip: error: can't connect to domain.com:16125
```

The weird thing, is that the server reports that the device was successfully exported:
```
usbipd: debug: usbipd.c:575:[do_standalone_mode] heartbeat timeout on ppoll()
usbipd: debug: usbipd.c:569:[do_standalone_mode] read event on fd[0]=3
usbipd: info: connection from XX.XX.XX.XX:63894
usbipd: info: received request: 0x8003(4)
usbipd: info: found requested device: 1-1.1.2
usbip: info: connect: 1-1.1.2
usbipd: debug: usbipd.c:152:[recv_request_import] import request busid 1-1.1.2: complete
usbipd: info: request 0x8003(4): complete
usbipd: debug: usbipd.c:575:[do_standalone_mode] heartbeat timeout on ppoll()
```

I get the same thing for the second device as well (`1-1.1.3`).
If I try to attach a non-existent device just to see if I get the same message, it tells me that the device does not exist indeed:
```
C:\Windows\system32>usbip.exe -d --tcp-port 16125 attach -r domain.com -b 1-1.1.5
usbip: info: using port "16125"
usbip: debug: usbip.cpp:104:[run_command] running command: attach
usbip: debug: vhci.cpp:51:[vhci_driver_open] device path: \\?\usbip#ehci#1&79f5d87&0#{b8b60941-cacb-454a-a8d1-35acb8fa1f1e}
usbip: error: device not found by bus id: 1-1.1.5
```

The server also reports that the client tried to attach a non-existent device:
```
usbipd: debug: usbipd.c:575:[do_standalone_mode] heartbeat timeout on ppoll()
usbipd: debug: usbipd.c:569:[do_standalone_mode] read event on fd[0]=3
usbipd: info: connection from XX.XX.XX.XX:63918
usbipd: info: received request: 0x8003(4)
usbipd: info: requested device not found: 1-1.1.5
usbipd: debug: usbipd.c:139:[recv_request_import] import request busid 1-1.1.5: failed
usbipd: info: request 0x8003(4): failed
usbipd: debug: usbipd.c:575:[do_standalone_mode] heartbeat timeout on ppoll()
```
...so it's clear that it isn't a TCP connection problem (such as no route, firewall or similar).
Even more... the listing works, it just does not connect.

As said, it (partially) works with the [cezanne/usbip-win](https://github.com/cezanne/usbip-win) implementation.

Things I tried:
- Fresh Windows 10 Install (with all the updates)
- Removing the port forwarding (currently using TCP port 16125) and leaving the default one 3240.
- Installing the debug version (I received no further error)
- Adding USBIP to the Windows Firewall or disabling the firewall completely

None of them changed anything.

I also tried to install the mentioned SDK and WDK in the readme, but I couldn't find the `tracelog.exe` binary for *x64*, for some reason it only installed it for *arm* and *arm64*.
I tried with an older version of WDK, but unfortunately I got no relevant information that could lead me to anything.

Server is running `usbip (usbip-utils 2.0)`, kernel is `Linux version 5.15.0-1021-raspi (buildd@bos02-arm64-062) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #23-Ubuntu SMP PREEMPT Fri Nov 25 15:27:43 UTC 202`.

Any help is greatly appreciated.
Thank you,
A.

*Note: I've hidden the domain and IP for privacy purposes.*

adelin-mcbsoft, 2022-12-18:
In the end I managed to enable debugging as well, here's the `usbip-vhci.txt` contents:

```
[0]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::vhci_complete:VDEV_VHCI
[0]0a10 TRACE_LEVEL_VERBOSE vhci_ioctl:HCI_USB2, VDEV_VHCI USBIP_VHCI_PLUGIN_HARDWARE(0X2A4000), inlen 1348, outlen 4
[0]0a10 TRACE_LEVEL_VERBOSE plugin_vpdo:domain.com:16125, busid 1-1.1.2, serial '<NULL>'
[0]0a10 TRACE_LEVEL_VERBOSE vdev_create:ba07eb90 HCI_USB2, VDEV_VPDO
[0]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::set_options:get keepalive: idle(7200 sec) + cnt(10)*intvl(1 sec) => 7210 sec
[0]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::set_options:set keepalive: idle(30 sec) + cnt(9)*intvl(10 sec) => 120 sec
[0]0a10 TRACE_LEVEL_INFORMATION plugin_vpdo:Connected to domain.com:16125
[1]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::log:usbip_usb_device(path '/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2', busid 1-1.1.2, busnum 1, devnum 4, USB_SPEED_FULL,vid 0x1a86, pid 0x7523, rev 0x264, class/sub/proto ff/0/0, bConfigurationValue 0, bNumConfigurations 1, bNumInterfaces 0)
[1]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::read_descr_hdr:OUT 48{seqnum 3, devid 0x10004, in[0]}, cmd_submit: flags 0, length 18, start_frame 0, isoc[-1], interval 0, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x12(18)}
[1]0a10 TRACE_LEVEL_ERROR `anonymous namespace'::read_descr_hdr:Recv header of 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE) 0x8000000d(STATUS_PARTIAL_COPY)
[1]0a10 TRACE_LEVEL_VERBOSE destroy_device:ba07ece0 HCI_USB2, VDEV_VPDO
[1]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::destroy:HCI_USB2 ba07ece0, port 0
[1]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::cancel_pending_irps:HCI_USB2 vpdo ba07ece0
[1]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::PeekNextIrp:0000
[1]0a10 TRACE_LEVEL_VERBOSE vhub_detach_vpdo:ba07ece0, port 0
[1]0a10 TRACE_LEVEL_VERBOSE vhci_ioctl:HCI_USB2, VDEV_VHCI -> STATUS_SUCCESS, outlen 4
[1]0a10 TRACE_LEVEL_VERBOSE `anonymous namespace'::vhci_complete:VDEV_VHCI
```

vadimgrn, 2022-12-18:
Hi, the driver connects to the server and fails to read configuration #0. This critical error causes the disconnect.

adelin-mcbsoft, 2022-12-18:
Ok… and how can I address this and fix the issue?

vadimgrn, 2022-12-18:
This error is returned by
NTSTATUS wsk::send(_In_ SOCKET *sock, _In_ WSK_BUF *buffer, _In_ ULONG flags)

I need to analyze this code, it can has a bug.

vadimgrn, 2022-12-18:
If you can build the solution, try to comment out the check where STATUS_PARTIAL_COOY is returned.

vadimgrn, 2022-12-19:
I haven't found errors in the code that receives usb device descriptor.
Could you please run this build and get me drivers' logs on Windows and Linux. Logs from userspace (daemon, etc.) are not required.
[usbip-win2-0.9.3.2-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/10259606/usbip-win2-0.9.3.2-Debug.zip)

adelin-mcbsoft, 2022-12-19:
By driver logs you mean the same `tracelog.exe` generated errors, right? From WDK.
And on Linux, what kind of logs you need? The only one I had so far were the ones generated by `usbipd -d` (`--debug`).
Do you need anything else?

vadimgrn, 2022-12-19:
usbip-vhci.txt on Windows, right.
For linux see https://github.com/vadimgrn/usbip-win2/tree/develop#obtaining-usbip-log-on-linux. I need a log from the module.

adelin-mcbsoft, 2022-12-19:
Ok, I no longer have access to that particular Raspberry Pi, but I will get a new one soon and come back with the required info. Thanks.

closed 2023-05-28 by vadimgrn