vadimgrn/usbip-win2 #23: Why doesn't this project use the VHCI(UDE) driver?

WuXian2021, 2023-01-09:
Hi, thanks for developing this project.
I'd like to know why this project doesn't use the VHCI(UDE) driver.
I'm using the VHCI(UDE) driver of the https://github.com/cezanne/usbip-win project in my project, but the project has many problems.
So, I'm considering this project.
But, because this project uses the VHCI(WDM) driver, I'm struggling to properly change my project.
Please help me.

vadimgrn, 2023-01-09:
Hi, because WDM driver works well. The major issue is that it can't create virtual USB 3.0 controller. Microsoft is used KMDF for USB 3.0 drivers. For this reason I'm developing UDE driver. if you look in develop branch, you will see that UDE driver is almost complete. 

The major issue with UDE is that it doesn't call a callback for SELECT_CONFIGURATION/SELECT_INTERFACE. For that reason upper filter driver is used that catches these requests and should pass these data to UDECXUSBDEVICE (currenlty I'm working on this). If you know how to do that, you are welcome.

vadimgrn, 2023-03-02:
"master" branch now has UDE driver.

closed 2023-03-02 by vadimgrn