vadimgrn/usbip-win2 #24: Communications issue with the device

JackSlateur, 2023-01-09:
Hello,

We are using cezanne's WDM driver with a specific USB device and are trying to move to usbip-win2 (hopefully that will fix many of our issues ..)

When plugged in and powered, that device is in a specific mode called "diag mode"
usbip-win2 bind correctly the device in that first mode
We can then use a tool to switch the device's mode to a second mode call "passthru mode" : it will disconnect, do a soft reboot, and can be rebind in the second mode
The mode switch is physically visible on the device via a colored led blinking

Anyway, when bound with usbip-win2, the passthru mode cannot be reached : the led never blinks, the device is never disconnected, because there is no soft reboot
It seems that whatever communication happens between the soft and the device is somehow denied by the driver
There is no crash or whatever, the device is still seen in its first mode

I've fetched the logs using the [procedure](https://github.com/vadimgrn/usbip-win2#obtaining-usbip-logs-on-windows):
[usbip-vhci.txt](https://github.com/vadimgrn/usbip-win2/files/10373479/usbip-vhci.txt)

I know this is not much information to work with ..
What could it be ?

Best regards,

vadimgrn, 2023-01-09:
Hello, you should start logging session, connect to device and do everything to reproduce the issue. After that log should be gathered. Your log is incomplete.

vadimgrn, 2023-01-09:
It's weird to see that the device's driver sends bulk IN\OUT requests with zero TransferBufferLength. This is meaningless.

```
[3]2484 VRB bulk_or_interrupt_transfer:irp 18871540 -> PipeHandle 0x20102, OUT|SHORT_OK, TransferBufferLength 0 
[0]1260 VRB bulk_or_interrupt_transfer:irp 1b204d80 -> PipeHandle 0x20181, IN|SHORT_OK, TransferBufferLength 0 
```

JackSlateur, 2023-01-10:
Hello,
What are the differences between VDEV_VPDO and VDEV_HPDO ?

It seems that cezanne's driver bind our device as HPDO while usbip-win2 bind as VPDO

vadimgrn, 2023-01-10:
Hub PDO, virtual device PDO. I'm finishing development of brand new UDE driver. Today I implemented the last missing part of it. After testing I could give a build to you. I just need to make sure that it works for flash drives and webcams.

vadimgrn, 2023-01-16:
You could try the build of brand new UDE driver, it works for flash drivers, webcams. It doesn't work for audio devices. May be your device will work.
[usbip-win2-0.9.5.0-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/10423530/usbip-win2-0.9.5.0-Debug.zip)

vadimgrn, 2023-01-18:
Try this
[usbip-win2-0.9.5.0-Debug.zip](https://github.com/vadimgrn/usbip-win2/files/10445185/usbip-win2-0.9.5.0-Debug.zip)

JackSlateur, 2023-01-18:
Awesome, the issue is gone

We'll pursue our tests

JackSlateur, 2023-02-05:
We'll open a new issue if required

Thank you

closed 2023-02-05 by JackSlateur