vadimgrn/usbip-win2 #29: Question about driver Installation and "Test mode"

DzzD, 2023-02-11:
Wouldn't be possible to install unsigned driver using one of the following methods (not test mode): 

https://www.minitool.com/news/install-unsigned-drivers-windows-10.html

And then use it in a windows "normal mode" environment ?

vadimgrn, 2023-02-11:
I haven't tried another methods except the test mode.

vadimgrn, 2023-02-11:
Unsigned driver can't be loaded by definition. You must disable integrity check permanently.

DzzD, 2023-02-11:
> Unsigned driver can't be loaded by definition. You must disable integrity check permanently if you want to use it.

This is what I try to do... It works only when I use UEFI menu, using BCDEDIT dont works at reboot

bcdedit give :
```
nointegritychecks       Yes
testsigning             No
```

But that's dont work either :/, I wtill need to reboot and use UEFI "7 - disable integrity check"

DzzD, 2023-02-12:
> Unsigned driver can't be loaded by definition. You must disable integrity check permanently if you want to use it.

And do you know how to do that "disable integrity check permanently" ? I would prefer to run my computer without integrity check rather than in test mode.

Some personal thoughts :
_Sorry for my insistance on signing drivers, but USBIP allow to make any 3D printer or Laser (including their contrôle cameras) easily remotetly controlable over LAN/WIFI/even WAN, and that's exacly why I use it, I think it have a great potential, and that a lot of customers could be interrested, societies too... with only a raspberrypi 3 (less than 50$) it enable to make a lot of thing remote controlable by well know softwares without modifications  (eg: Lightburn for laser). 

This could be sellable as a box containing a raspberrypi and two USB port, but IIt would requiere easy install (this is already the case with your version, just missing signing :/ and a bit of work on hardware reboot/unplug)

I am convinced that there is a market that is far away bigger than the cost of signing drivers and which would be of great service for customers (ir is the case for me)._

vadimgrn, 2023-02-12:
I agree that it is worth to add sponsorship for the project. I'm going to do that despite some difficulties.

Currently I'm considering to use liberapay or lfx_crowdfunding. If you have better options, let me know, pls.

DzzD, 2023-02-13:
You could use GitHub sponsorship so all will be concentrated in one place, you may explain on the README of the repository that to keep the drivers signed it is required to grab XXX$ / years

DzzD, 2023-02-13:
Also I suppose that giving some real world usage examples (as laser, CNC, 3d printer) could help to motivate people

vadimgrn, 2023-03-11:
I added two sponsorship methods.

closed 2023-03-11 by vadimgrn