vadimgrn/usbip-win2 #30: Unable to detach devices

DzzD, 2023-02-11:
Seems that detaching a device does not works in my laptop, even rebooting the server does not correct the problem, the client still cant detach remote connected USB devices, any advice ?

Once server reboot, those device are still available in "phantom mode"

```
C:\Users\bruno>usbip port
Imported USB devices
====================
Port 01: <Device in Use> at Full Speed(12Mbps)
       QinHeng Electronics : CH340 serial converter (1a86:7523)
           -> usbip://192.168.0.101:3240/1-1.5
           -> remote bus/dev 001/005
Port 02: <Device in Use> at Low Speed(1.5Mbps)
       Sunplus Innovation Technology Inc. : unknown product (1bcf:28c4)
           -> usbip://192.168.0.101:3240/1-1.3
           -> remote bus/dev 001/004
Port 03: <Device in Use> at Low Speed(1.5Mbps)
       Sunplus Innovation Technology Inc. : unknown product (1bcf:28c4)
           -> usbip://192.168.0.101:3240/1-1.3
           -> remote bus/dev 001/004

C:\Users\bruno>usbip detach -p 01
usbip: error: failed to detach

C:\Users\bruno>usbip detach -p 1
usbip: error: failed to detach

C:\Users\bruno>usbip detach -p 02
usbip: error: failed to detach

C:\Users\bruno>usbip detach -p 03
usbip: error: failed to detach

C:\Users\bruno>usbip detach -p -1
all ports are detached

C:\Users\bruno>usbip port
Imported USB devices
====================
Port 01: <Device in Use> at Full Speed(12Mbps)
       QinHeng Electronics : CH340 serial converter (1a86:7523)
           -> usbip://192.168.0.101:3240/1-1.5
           -> remote bus/dev 001/005
Port 02: <Device in Use> at Low Speed(1.5Mbps)
       Sunplus Innovation Technology Inc. : unknown product (1bcf:28c4)
           -> usbip://192.168.0.101:3240/1-1.3
           -> remote bus/dev 001/004
Port 03: <Device in Use> at Low Speed(1.5Mbps)
       Sunplus Innovation Technology Inc. : unknown product (1bcf:28c4)
           -> usbip://192.168.0.101:3240/1-1.3
           -> remote bus/dev 001/004
```

vadimgrn, 2023-02-11:
This is a known issue for some devices. "usbip unbind" on server side could help.

DzzD, 2023-02-12:
Thanks for the tips, everything works fine with your version (and seems a lot more stable) as long as there are no unexpected things : power off of the server or client, hardware USB disconnection,  without software disconnection (usbip unbind and uspip detach -p -1) before.

vadimgrn, 2023-02-12:
Thanks for your feedback! I'm being working on this project since Nov 2021. 

As far as I understand you are using WDM driver. There is UDE driver in "develop" branch. It is stable, but I still can't solve some issues, for example audio devices do not work. You could try it if you can build it yourself. I'm going to release it AS IS soon. It could be helpful if someone has devices that don't work with WDM driver.

vadimgrn, 2023-05-28:
Fixed in release 0.9.5.3

closed 2023-05-28 by vadimgrn