vadimgrn/usbip-win2 #33: Project Status

HeadHodge, 2023-03-04:
hi,

if i understand your project properly, i think its fantastic and hope youre project is still active. i think your 'very' smart to conform to the linux USB/IP protocol.

so in theory i should be able to create an embedded host that acts as an usb hid keyboard host that connects to your client and also to a linux client too. that sound right?

im, creating a small embedded device to emulate usb host  hid devices that connect to the clients via physical usb, bluetooth, and ip.

i have the usb and bluetooth working. so ready to focus on wifi.

thanks again for this project. happy to help if possible.

-bob

vadimgrn, 2023-03-05:
Hi, thank you! As far as I understand, you are running usb/ip server on your Linux device. You plug in some USB devices to it (keyboard, etc.) and want to use them on Windows over the network. If it is correct, this project is exactly what you need.

HeadHodge, 2023-03-06:
> you are going to run usb/ip server on your Linux device.

hi,

same concept as you describe, except the server needs to run embedded on my esp32-s3-usb-otg board run with arduino os (no linux, but coding is c/c++).

so now i'm looking for such an existing server or a way to port to arduino, one thats coded in c/c++.

any thoughts are appreciated. 🤓👍

vadimgrn, 2023-03-07:
So, you should port usbip server from Linux kernel to your OS. I'm not sure it's as easy as you might think. You should port userspace tools too or write new ones. Simply put, you need to write the usbip server implementation for Adruino almost from scratch.

HeadHodge, 2023-03-07:
🍺🤓👍

antton, 2023-04-18:
@HeadHodge Maybe this link can be useful for you https://robots.nootrix.com/diy-tutos/usb-wifi-arduino/

HeadHodge, 2023-04-19:
@antton 🤓👍

closed 2023-05-26 by vadimgrn