vadimgrn/usbip-win2 #34: USBHub3.sys BSOD on device disconnect

nraggett, 2023-03-15:
When running ~130mb/s of data over the link for a long enough period of time, I get a BSOD. It appears to be from the device disconnecting and reconnecting.
The stack dump is:
```
SYSTEM_THREAD_EXCEPTION_NOT_HANDLED (7e)
This is a very common BugCheck.  Usually the exception address pinpoints
the driver/function that caused the problem.  Always note this address
as well as the link date of the driver/image that contains this address.
Arguments:
Arg1: ffffffffc0000005, The exception code that was not handled
Arg2: fffff80028a8f47c, The address that the exception occurred at
Arg3: fffff900b2a84288, Exception Record Address
Arg4: fffff900b2a83ac0, Context Record Address

Debugging Details:
------------------
KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 29983

    Key  : Analysis.DebugAnalysisManager
    Value: Create

    Key  : Analysis.Elapsed.mSec
    Value: 30298

    Key  : Analysis.IO.Other.Mb
    Value: 1

    Key  : Analysis.IO.Read.Mb
    Value: 0

    Key  : Analysis.IO.Write.Mb
    Value: 26

    Key  : Analysis.Init.CPU.mSec
    Value: 2655

    Key  : Analysis.Init.Elapsed.mSec
    Value: 26247

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 112

    Key  : Bugcheck.Code.DumpHeader
    Value: 0x7e

    Key  : Bugcheck.Code.KiBugCheckData
    Value: 0x7e

    Key  : Bugcheck.Code.Register
    Value: 0x7e

    Key  : WER.OS.Branch
    Value: vb_release

    Key  : WER.OS.Timestamp
    Value: 2019-12-06T14:06:00Z

    Key  : WER.OS.Version
    Value: 10.0.19041.1

FILE_IN_CAB:  MEMORY.DMP

VIRTUAL_MACHINE:  VirtualBox

BUGCHECK_CODE:  7e

BUGCHECK_P1: ffffffffc0000005

BUGCHECK_P2: fffff80028a8f47c

BUGCHECK_P3: fffff900b2a84288

BUGCHECK_P4: fffff900b2a83ac0

EXCEPTION_RECORD:  fffff900b2a84288 -- (.exr 0xfffff900b2a84288)
ExceptionAddress: fffff80028a8f47c (nt!ExFreeHeapPool+0x000000000000076c)
   ExceptionCode: c0000005 (Access violation)
  ExceptionFlags: 00000000
NumberParameters: 2
   Parameter[0]: 0000000000000000
   Parameter[1]: 0000000000000000
Attempt to read from address 0000000000000000

CONTEXT:  fffff900b2a83ac0 -- (.cxr 0xfffff900b2a83ac0)
rax=0000000000000000 rbx=ffffb68305bddd70 rcx=0000000000000016
rdx=0000000000000000 rsi=0000000000000000 rdi=a2e64eada2e64ead
rip=fffff80028a8f47c rsp=fffff900b2a844c0 rbp=0000000000000000
 r8=0000000000000000  r9=0000000000000000 r10=0000000000000000
r11=fffff8002d2484e8 r12=0000000000000000 r13=0000000000000200
r14=0000000000000000 r15=0000000000000001
iopl=0         nv up ei ng nz na pe nc
cs=0010  ss=0018  ds=002b  es=002b  fs=0053  gs=002b             efl=00010282
nt!ExFreeHeapPool+0x76c:
fffff800`28a8f47c 488b18          mov     rbx,qword ptr [rax] ds:002b:00000000`00000000=????????????????
Resetting default scope

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXNTFS: 1 (!blackboxntfs)

BLACKBOXPNP: 1 (!blackboxpnp)

BLACKBOXWINLOGON: 1

EXCEPTION_STR:  0xc0000005

STACK_TEXT:  
fffff900`b2a844c0 fffff800`291b5019     : 00000000`00000000 00000000`00000000 00000000`00000000 01000000`00100000 : nt!ExFreeHeapPool+0x76c
fffff900`b2a845a0 fffff800`2d1b5b0d     : 00000000`ffffffff 00000000`00000001 00000000`1c341c5c 00000000`00000000 : nt!ExFreePool+0x9
fffff900`b2a845d0 fffff800`2d1b40cb     : 00000000`00000000 fffff900`b2a84680 00000000`00000004 00000000`00000000 : Wdf01000!FxRegKey::`scalar deleting destructor'+0x8d
fffff900`b2a84610 fffff800`2d1b2a46     : ffffb683`012b7d60 fffff800`2d1ba67a 00000000`c0000034 0000497c`fa421248 : Wdf01000!FxObject::SelfDestruct+0x1b [minkernel\wdf\framework\shared\inc\private\common\fxobject.hpp @ 453] 
fffff900`b2a84640 fffff800`2d1b5f0d     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000001 : Wdf01000!FxObject::Release+0x106 [minkernel\wdf\framework\shared\inc\private\common\fxobject.hpp @ 884] 
fffff900`b2a84690 fffff800`2d1b3c08     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : Wdf01000!FxObject::DeleteObject+0x19d [minkernel\wdf\framework\shared\object\fxobjectstatemachine.cpp @ 124] 
fffff900`b2a84710 fffff800`30b8e55e     : 00000000`00000000 ffffb683`05bddd70 00000000`00000000 00000000`00000000 : Wdf01000!imp_WdfRegistryClose+0x78 [minkernel\wdf\framework\shared\support\fxregistryapi.cpp @ 327] 
fffff900`b2a84760 fffff800`30b8c708     : ffffb683`05769894 00000000`00000fa0 ffffb683`057692c8 00000000`0000afd1 : UsbHub3!HUBREG_QueryUsbflagsValuesForDevice+0xb4e
fffff900`b2a84920 fffff800`30b2e9e0     : 00000000`00000000 ffffb683`03a08410 ffffb683`057692c8 00000000`ffffffff : UsbHub3!HUBMISC_QueryAndCacheRegistryValuesForDevice+0x54
fffff900`b2a84990 fffff800`30b1a737     : ffffb683`00000000 00000000`0000afd1 fffff900`00000000 01000000`00100000 : UsbHub3!HUBDSM_QueryingRegistryValuesForAlternateDeviceEnumeration+0x10
fffff900`b2a849c0 fffff800`30b1aecf     : 00000000`00000000 00000000`00000000 fffff800`00000fa0 fffff800`30b76b80 : UsbHub3!HUBSM_ExecuteEntryFunctionsAndPushPopStateMachinesForCurrentState+0x87
fffff900`b2a84a60 fffff800`3095a770     : ffffb683`03a08410 fffff800`30b1aeb0 ffffb683`0138eb60 fffff800`30b1aeb0 : UsbHub3!HUBSM_EvtSmWorkItem+0x1f
fffff900`b2a84aa0 fffff800`3095a6f9     : ffffb683`034aee50 ffffb683`0138eb60 00000000`00000000 fffff800`3095a6f0 : ucx01000!Controller_ForwardProgressWorkItemCallback+0x60
fffff900`b2a84ad0 fffff800`28a07d53     : 00000000`00000000 00000000`00000000 00000000`00000000 fffff800`2fd95440 : ucx01000!Controller_ForwardProgressWorkItemWdmCallback+0x9
fffff900`b2a84b00 fffff800`28abdef5     : ffffb682`fc245080 ffffb682`fc245080 fffff800`28a07cc0 ffffb682`00000000 : nt!IopProcessWorkItem+0x93
fffff900`b2a84b70 fffff800`28a55485     : ffffb682`fc245080 00000000`00000080 ffffb682`fbc7e080 000fa005`b8bb3dfe : nt!ExpWorkerThread+0x105
fffff900`b2a84c10 fffff800`28c02d48     : ffff8e80`2e3a0180 ffffb682`fc245080 fffff800`28a55430 00000000`00000000 : nt!PspSystemThreadStartup+0x55
fffff900`b2a84c60 00000000`00000000     : fffff900`b2a85000 fffff900`b2a7f000 00000000`00000000 00000000`00000000 : nt!KiStartSystemThread+0x28

SYMBOL_NAME:  UsbHub3!HUBREG_QueryUsbflagsValuesForDevice+b4e

MODULE_NAME: UsbHub3

IMAGE_NAME:  UsbHub3.sys

IMAGE_VERSION:  10.0.19041.1202

STACK_COMMAND:  .cxr 0xfffff900b2a83ac0 ; kb

BUCKET_ID_FUNC_OFFSET:  b4e

FAILURE_BUCKET_ID:  AV_UsbHub3!HUBREG_QueryUsbflagsValuesForDevice

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {9865597a-443d-22dc-6f2b-3f61bcf3e013}

Followup:     MachineOwner
---------
```

The issue appears to be from UCX handling a WorkItem. I'll look further into their use to see if there's a possible faulty WorkItem being passed.

nraggett, 2023-03-15:
Unfortunately the `!wdfkd.wdflogdump usbip2_ude -f` and `!wdfkd.wdflogdump usbip2_ude -d`  didn't provide anything from this dump.

vadimgrn, 2023-03-15:
As you see, there is no usbip driver in the backtrace. It could be the bug in UsbHub3 or Wdf01000 driver that calls ExFreePool with NULL pointer. Have you tried to get usbip logs as described [here](https://github.com/vadimgrn/usbip-win2/tree/develop#obtaining-usbip-logs-on-windows)?

nraggett, 2023-03-15:
I did, unfortunately the logs are empty every time I get a BSOD.
I haven't attempted to grab the logs with !wdfkd.wdflogdump usbip2_ude -f while the trace was running, I'll attempt that now

nraggett, 2023-03-15:
Unfortunately I only get this when I attempt to grab the logs from the BSOD dump:
```1: kd> !wdfkd.wdfsearchpath %TEMP%\USBip
Trace searchpath is: %TEMP%\USBip
1: kd> !wdfkd.wdfsettraceprefix [%9]%3!04x! %!LEVEL! %!FUNC!:
SetTracePrefix: "[%9]%3!04x! %!LEVEL! %!FUNC!:"
1: kd> !wdfkd.wdflogdump usbip2_ude -d
Warning: It looks like you're using the WDF debugger extension on a WDM
         driver. Framework logs will not be available

Unable to load image \SystemRoot\System32\DriverStore\FileRepository\usbip2_ude.inf_amd64_c58d60c2ac484616\usbip2_ude.sys, Win32 error 0n2
warning: could not retrieve framework image for client driver usbip2_ude
Trace searchpath is: %TEMP%\USBip

Trace format prefix is: [%9]%3!04x! %!LEVEL! %!FUNC!:
warning: could not retrieve framework image for client driver usbip2_ude
hint: Are symbols available for this driver?
hint: Did you provide the correct .sys/.dll extension in the drivername parameter?

Could not find usbip2_ude in client list
hint: Are symbols available for this driver?
hint: Did you provide the correct .sys/.dll extension in the drivername parameter?

Could not find usbip2_ude in client list
1: kd> !wdfkd.wdflogdump usbip2_ude -f
Warning: It looks like you're using the WDF debugger extension on a WDM
         driver. Framework logs will not be available

warning: could not retrieve framework image for client driver usbip2_ude
Trace searchpath is: %TEMP%\USBip

Trace format prefix is: [%9]%3!04x! %!LEVEL! %!FUNC!:
warning: could not retrieve framework image for client driver usbip2_ude
hint: Are symbols available for this driver?
hint: Did you provide the correct .sys/.dll extension in the drivername parameter?

Could not find usbip2_ude in client list
hint: Are symbols available for this driver?
hint: Did you provide the correct .sys/.dll extension in the drivername parameter?

Could not find usbip2_ude in client list```

nraggett, 2023-03-15:
I'm currently using the UDE driver. I've ensured there are no usbip-vhci driver entries in my system.
My current usb directory is as follows:
[image]

The WDF/WDM message is easily repeatable across multiple memory dumps.

nraggett, 2023-03-16:
Going through with a live kernel debugger I was able to trace down the issue to when it gets the descriptors from the device
```[3]0004.1CCC::03/16/2023-16:21:33.365 [ude]dev 8ab248c8, UdecxEndpointsConfigureTypeEndpointsReleasedOnly: unplugged, NewConfigurationValue 0, InterfaceNumber 0, NewInterfaceSetting 0
[3]0004.1CCC::03/16/2023-16:21:33.366 [ude]WDFREQUESTCOMPLETE HERE line: 383
[0]0688.0638::03/16/2023-16:21:35.470 [ude]vhci_plugin_hardware(0X22E000), OutputBufferLength 8, InputBufferLength 1100
[0]0688.0638::03/16/2023-16:21:35.471 [ude]192.168.4.1:3240, busid 2-1
[3]0688.0638::03/16/2023-16:21:35.473 [ude]get keepalive: idle(7200 sec) + cnt(10)*intvl(1 sec) => 7210 sec
[3]0688.0638::03/16/2023-16:21:35.475 [ude]set keepalive: idle(30 sec) + cnt(9)*intvl(10 sec) => 120 sec
[3]0688.0638::03/16/2023-16:21:35.479 [ude]Connected to 192.168.4.1:3240
[3]0688.0638::03/16/2023-16:21:35.487 [ude]usbip_usb_device(path '/sys/devices/platform/axi/ff9e0000.usb1/fe300000.dwc3/xhci-hcd.1.auto/usb2/2-1', busid 2-1, busnum 2, devnum 76, USB_SPEED_SUPER,vid 0x403, pid 0xff0b, rev 0x205, class/sub/proto 0/0/0, bConfigurationValue 0, bNumConfigurations 1, bNumInterfaces 0)
[3]0688.0638::03/16/2023-16:21:35.490 [ude]wsk workitem 8ac56e88
[3]0688.0638::03/16/2023-16:21:35.491 [ude]dev 8ab164a8, queue 8c477808
[3]0688.0638::03/16/2023-16:21:35.492 [ude]dev 8ab164a8
[3]0688.0638::03/16/2023-16:21:35.494 [ude]Referencing a new handle (WdfObjectReference)
[3]0688.0638::03/16/2023-16:21:35.495 [ude]dev 8ab164a8, port 6
[3]0688.0638::03/16/2023-16:21:35.496 [ude]dev 8ab164a8 -> port 6
[3]0688.0638::03/16/2023-16:21:35.498 [ude]STATUS_SUCCESS, Information 8
[3]0688.0638::03/16/2023-16:21:35.499 [ude]WDFREQUESTCOMPLETE HERE line: 529
[3]0004.0010::03/16/2023-16:21:35.500 [ude]dev 8ab164a8, endp 8eafb0e8{Address 0000: Ctrl Out[0], MaxPacketSize 0, Interval 0}, queue 8c7adf68
	0000  07050000 000000                      .......
	
[3]0004.0010::03/16/2023-16:21:35.502 [ude]dev 8ab164a8, EndpointsToConfigure[1]
	0000  E8B0AF8E F8350000                    .....5..
	
[3]0004.0010::03/16/2023-16:21:35.504 [ude]dev 8ab164a8, DeviceInitialize
[3]0004.0010::03/16/2023-16:21:35.506 [ude]req 8c15c2d8 -> PipeHandle 6cad1950, IN|SHORT_OK|DEFAULT_PIPE, TransferBufferLength 18, Timeout 5000, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x12(18)}
[3]0004.0010::03/16/2023-16:21:35.509 [ude]req 8c15c2d8 -> 48{seqnum 3, devid 0x2004c, in[0]}, cmd_submit: flags 0, length 18, start_frame 0, isoc[-1], interval 0, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x12(18)}
[3]0000.0000::03/16/2023-16:21:35.515 [ude]req 8c15c2d8 <- 66{seqnum 3, devid 0, in[0]}, ret_submit: status 0, actual_length 18, start_frame 0, isoc[0], error_count 0
[3]0000.0000::03/16/2023-16:21:35.517 [ude]bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE)
	0000  12010003 00000009 03040BFF 05020102  ................
	0010  0301                                 ..
	
[3]0000.0000::03/16/2023-16:21:35.519 [ude]DeviceDescriptor(bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE), bcdUSB 0x300, bDeviceClass 0, bDeviceSubClass 0, bDeviceProtocol 0, bMaxPacketSize0 9, idVendor 0x403, idProduct 0xff0b, bcdDevice 0x205, iManufacturer 1, iProduct 2, iSerialNumber 3, bNumConfigurations 1)
KDTARGET: Refreshing KD connection

*** Fatal System Error: 0x0000007e
                       (0xFFFFFFFFC0000005,0xFFFFF8032C88F47C,0xFFFFF481FB362168,0xFFFFF481FB3619A0)
```

I'll report back if I find what's causing the issue

vadimgrn, 2023-03-16:
There is nothing wrong here. Device descriptor is returned as usual. May be this happens when hub or device is in wrong state.

nraggett, 2023-03-22:
Unfortunately it had no effect, I still get the same BSOD and dump:
```STACK_TEXT:  
fffff185`e75774c0 fffff802`369b5019     : 00000000`00000000 00000000`00000000 00000000`00000000 01000000`00100000 : nt!ExFreeHeapPool+0x76c
fffff185`e75775a0 fffff802`391b5b0d     : 00000000`ffffffff 00000000`00000001 00000000`25102538 00000000`00000000 : nt!ExFreePool+0x9
fffff185`e75775d0 fffff802`391b40cb     : 00000000`00000000 fffff185`e7577680 00000000`00000004 00000000`00000000 : Wdf01000!FxRegKey::`scalar deleting destructor'+0x8d
fffff185`e7577610 fffff802`391b2a46     : ffffce83`8ab19d00 fffff802`391ba67a 00000000`c0000034 0000317c`7305c3e8 : Wdf01000!FxObject::SelfDestruct+0x1b [minkernel\wdf\framework\shared\inc\private\common\fxobject.hpp @ 453] 
fffff185`e7577640 fffff802`391b5f0d     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000001 : Wdf01000!FxObject::Release+0x106 [minkernel\wdf\framework\shared\inc\private\common\fxobject.hpp @ 884] 
fffff185`e7577690 fffff802`391b3c08     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : Wdf01000!FxObject::DeleteObject+0x19d [minkernel\wdf\framework\shared\object\fxobjectstatemachine.cpp @ 124] 
fffff185`e7577710 fffff802`3ca2e55e     : 00000000`00000000 ffffce83`8cfa7d10 00000000`00000000 00000000`00000000 : Wdf01000!imp_WdfRegistryClose+0x78 [minkernel\wdf\framework\shared\support\fxregistryapi.cpp @ 327] 
fffff185`e7577760 fffff802`3ca2c708     : ffffce83`8d3e9ce4 00000000`00000fa0 ffffce83`8d3e9718 00000000`0000afd1 : UsbHub3!HUBREG_QueryUsbflagsValuesForDevice+0xb4e
fffff185`e7577920 fffff802`3c9ce9e0     : 00000000`00000000 ffffce83`8efbc110 ffffce83`8d3e9718 00000000`ffffffff : UsbHub3!HUBMISC_QueryAndCacheRegistryValuesForDevice+0x54
fffff185`e7577990 fffff802`3c9ba737     : ffffce83`00000000 00000000`0000afd1 fffff185`00000000 01000000`00100000 : UsbHub3!HUBDSM_QueryingRegistryValuesForAlternateDeviceEnumeration+0x10
fffff185`e75779c0 fffff802`3c9baecf     : 00000000`00000000 fffff185`e7577aa0 fffff802`00000fa0 fffff802`3ca16b80 : UsbHub3!HUBSM_ExecuteEntryFunctionsAndPushPopStateMachinesForCurrentState+0x87
fffff185`e7577a60 fffff802`3c7fa770     : ffffce83`8efbc110 fffff802`3c9baeb0 ffffce83`8ab21b60 fffff802`3c9baeb0 : UsbHub3!HUBSM_EvtSmWorkItem+0x1f
fffff185`e7577aa0 fffff802`3c7fa6f9     : ffffce83`8e6c4f00 ffffce83`8ab21b60 00000000`00000000 fffff802`3c7fa6f0 : ucx01000!Controller_ForwardProgressWorkItemCallback+0x60
fffff185`e7577ad0 fffff802`36207d53     : 00000000`00000000 00000000`00000000 00000000`00000000 fffff802`3bcd5440 : ucx01000!Controller_ForwardProgressWorkItemWdmCallback+0x9
fffff185`e7577b00 fffff802`362bdef5     : ffffce83`8aac8040 ffffce83`8aac8040 fffff802`36207cc0 ffffce83`00000000 : nt!IopProcessWorkItem+0x93
fffff185`e7577b70 fffff802`36255485     : ffffce83`8aac8040 00000000`00000080 ffffce83`84c69080 000fa005`b8bb3dfe : nt!ExpWorkerThread+0x105
fffff185`e7577c10 fffff802`36402d48     : ffff9601`daba0180 ffffce83`8aac8040 fffff802`36255430 00000000`00000000 : nt!PspSystemThreadStartup+0x55
fffff185`e7577c60 00000000`00000000     : fffff185`e7578000 fffff185`e7572000 00000000`00000000 00000000`00000000 : nt!KiStartSystemThread+0x28

SYMBOL_NAME:  UsbHub3!HUBREG_QueryUsbflagsValuesForDevice+b4e

MODULE_NAME: UsbHub3

IMAGE_NAME:  UsbHub3.sys

IMAGE_VERSION:  10.0.19041.1202

STACK_COMMAND:  .cxr 0xfffff185e7576ac0 ; kb

BUCKET_ID_FUNC_OFFSET:  b4e

FAILURE_BUCKET_ID:  AV_UsbHub3!HUBREG_QueryUsbflagsValuesForDevice

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {9865597a-443d-22dc-6f2b-3f61bcf3e013}

Followup:     MachineOwner
---------

nraggett, 2023-03-22:
Here's a wireshark pcap of a test, ending on a crash. You can see at the bottom what looks like the descriptors getting polled, prior to a network disconnect from the PC BSODing.
[usbip_dump.zip](https://github.com/vadimgrn/usbip-win2/files/11035139/usbip_dump.zip)

I recommend compiling this version of Wireshark with its USBIP dissector fixed, as the current version does not correctly parse isochronous packets: https://gitlab.com/wireshark/wireshark/-/merge_requests/10094

nraggett, 2023-03-28:
Reviving this, it turns out my VM didn't have enough memory to do a full dump with logging output. Here's the output with all the BSOD debug info requested:
[BSOD.txt](https://github.com/vadimgrn/usbip-win2/files/11084986/BSOD.txt)

nraggett, 2023-05-26:
Closing as this issue is now outdated.

closed 2023-05-26 by nraggett

vadimgrn, 2023-05-26:
I did a lot of work last month and fixed several issues that caused BSODs. A new release will be issued soon.