vadimgrn/usbip-win2 #35: Insights on why audio devices don't work with UDE

nefarius, 2023-03-26:
Ahoy!

Thought this project might be interested [in my findings](https://fosstodon.org/@Nefarius/110089542325783941) as it took me and a colleague months to figure it out 🤪

Cheers, and keep up the good work!

vadimgrn, 2023-03-26:
Thank you very much! I'll implement QueryBusTime as WDM driver does. WDM driver implements many similar stuff, that's why most devices work.

nefarius, 2023-03-26:
Great! here's a snippet on what worked for me:

```c
static
NTSTATUS
USB_BUSIFFN
UsbInterfaceQueryBusTime(
	IN PVOID BusContext,
	IN OUT PULONG CurrentUsbFrame
)
{
	FuncEntry(TRACE_DEVICE);

	const PCHILD_DEVICE_CONTEXT pCtx = ChildDeviceGetContext(BusContext);

	NTSTATUS status = pCtx->OriginalInterface.QueryBusTime(
		pCtx->OriginalInterface.BusContext,
		CurrentUsbFrame
	);

	if (CurrentUsbFrame)
	{
		TraceInformation(
			TRACE_DEVICE,
			"UsbInterfaceQueryBusTime.CurrentUsbFrame = %d",
			*CurrentUsbFrame
		);
	}

	// fake-succeed the call
	if (CurrentUsbFrame && status == STATUS_NOT_SUPPORTED)
	{
		*CurrentUsbFrame = 0;
		status = STATUS_SUCCESS;
	}

	FuncExit(TRACE_DEVICE, "status=%!STATUS!", status);

	return status;
}
```

vadimgrn, 2023-04-10:
I my case audio device does not work for another reason.
The kernel sets interface 1.1 and issues ISOCH transfers, but UDE has not called EVT_UDECX_USB_DEVICE_ENDPOINTS_CONFIGURE to set 1.1 yet. My driver does not receive ISOCH transfers at all, because UDE fails them.
I still can't comprehend a logic of  EVT_UDECX_USB_DEVICE_ENDPOINTS_CONFIGURE.

```
# UDE
endpoint_add:dev bbdfbdc8, endp cc2b4798{Address 0000: Ctrl Out[0], MaxPacketSize 0, Interval 0}, queue c701d758
	0000  07050000 000000                      .......

# filter driver
select_interface:ConfigurationHandle 4d1a1320
Interface(Length 24, InterfaceNumber 1, AlternateSetting 0, Class 0x1, SubClass 0x2, Protocol 00, InterfaceHandle 465b8aa0, NumberOfPipes 0)

# UDE
endpoint_add:dev bbdfbdc8, endp cc2b4c18{Address 0x06: Isoch Out[6], MaxPacketSize 192, Interval 1}, queue c790c4d8
	0000  09050609 C0000100 00                 .........

# UDE
endpoints_configure:dev bbdfbdc8, EndpointsToConfigure[1]
	0000  184C2BCC 7C590000                    .L+.|Y..

# UDE
endpoints_configure:dev bbdfbdc8, InterfaceNumber 1, NewInterfaceSetting 0

# filter driver
select_interface:ConfigurationHandle 4d1a1320
Interface(Length 48, InterfaceNumber 1, AlternateSetting 1, Class 0x1, SubClass 0x2, Protocol 00, InterfaceHandle 4ac3dda0, NumberOfPipes 1)
Pipes[0](MaximumPacketSize 0xc0, EndpointAddress 0x6 OUT[6], Interval 0x1, Isoch, PipeHandle 39cc5be0, MaximumTransferSize 0x17700, PipeFlags 0x8)

# UDE, reset endpoint ISOCH OUT[6]
control_transfer:req bf66a4d8 -> PipeHandle 3afaf3d0, OUT|SHORT_OK|DEFAULT_PIPE, TransferBufferLength 3, Timeout 0, {OUT|CLASS|ENDPOINT, CLEAR_FEATURE(0x1), wValue 0x100, wIndex 0x06, wLength 0x03(3)}

# filter driver, ISOCH transfers did not reach the main driver (because UDE has not selected interface 1.1 yet?)
ISOCH: PipeHandle 39cc5be0, OUT|ISO_ASAP, TransferBufferLength 1920, StartFrame 0, NumberOfPackets 10, ErrorCount 0
dev 4e64d7d0, ISOCH_TRANSFER, USBD_STATUS_INVALID_PARAMETER, 0xc000000d(STATUS_INVALID_PARAMETER)

# filter driver
dev 4e64d7d0, SYNC_RESET_PIPE_AND_CLEAR_STALL, PipeHandle 39cc5be0

...

# UDE never receieved these ISOCH

endpoint_purge:dev bbdfbdc8, endp cc2b4c18, queue c790c4d8
endpoint_start:endp cc2b4c18, queue c790c4d8
...

endpoint_purge:dev bbdfbdc8, endp cc2b4c18, queue c790c4d8

endpoints_configure:dev bbdfbdc8, ReleasedEndpoints[1]
	0000  184C2BCC 7C590000                    .L+.|Y..

# too late
endpoints_configure:dev bbdfbdc8, InterfaceNumber 1, NewInterfaceSetting 1

# filter driver
select_interface:ConfigurationHandle 4d1a1320
Interface(Length 24, InterfaceNumber 1, AlternateSetting 0, Class 0x1, SubClass 0x2, Protocol 00, InterfaceHandle 465b5ea0, NumberOfPipes 0)
```

nefarius, 2023-04-11:
I haven't yet used dynamic endpoints myself so there might be other lingering issues nobody has addressed yet 🤔

vadimgrn, 2023-04-13:
Here I'm describing results of my work on this issue. It stubbornly does not work, no matter what I do. Sequence of events:

- Select interface 1.1 which has ISOCH OUT (address 0x6) endpoint
- OUT|CLASS|ENDPOINT, CLEAR_FEATURE(0x1), wValue 0x100, wIndex 0x06, wLength 3
- ucx01000 fails ISOCH OUT transfer with USBD_STATUS_INVALID_PARAMETER, STATUS_INVALID_PARAMETER
- ABORT_PIPE
- SYNC_RESET_PIPE_AND_CLEAR_STALL
- ISOCH OUT ...

vadimgrn, 2023-04-13:
It does not work even if complete ISOCH transfers in the filter driver. The result is the same, the loop of {ABORT_PIPE, SYNC_RESET_PIPE_AND_CLEAR_STALL, ISOCH OUT}. I have no more ideas what to do to avoid this.

nefarius, 2023-04-13:
> Here I'm describing results of my work on this issue. It stubbornly does not work, no matter what I do. Sequence of events in the filter driver:
> 
> ```
> - Select interface 1.1 which has ISOCH OUT (address 0x6) endpoint
> - OUT|CLASS|ENDPOINT, CLEAR_FEATURE(0x1), wValue 0x100, wIndex 0x06, wLength 3
> - ucx01000 fails ISOCH OUT transfer with USBD_STATUS_INVALID_PARAMETER, STATUS_INVALID_PARAMETER
> - ABORT_PIPE
> - SYNC_RESET_PIPE_AND_CLEAR_STALL
> - ISOCH OUT ...
> ```

This sequence ocurres because of the missing `QueryBusTime` implementation.

vadimgrn, 2023-04-13:
QueryBusTime is implemented, it doesn't affect on this issue.

vadimgrn, 2023-04-13:
The question is what argument is incorrect, all members of UrbIsochTransfer look the same as in WDM driver. By the way, transfer flags has ISOCH_ASAP, thus StartFrame is set to zero.

vadimgrn, 2023-04-20:
QueryBusTime affected only USB Audio card which is works now. All headsets don't work, I've spent a huge amount of time trying in vain to figure out what's wrong. If I had sources of ucx1000, the reason of the error would be detected instantly.

vadimgrn, 2023-06-04:
This is a configuration descriptor of my audio speaker
```	0000  09027400 02010080 FA090400 00000101  ..t.............
	0010  00000924 0100012B 0001010C 24020101  ...$...+....$...
	0020  01000203 0000000D 24060D01 02010202  ........$.......
	0030  00020000 09240303 0103000D 00090401  .....$..........
	0040  00000102 00000904 01010101 02000007  ................
	0050  24010101 01000E24 02010202 100244AC  $......$......D.
	0060  0080BB00 09050609 C0000100 00072501  ..............%.
	0070  01000000                             ....
```

USBD_ValidateConfigurationDescriptor returns 
```
USBD_STATUS_BAD_DESCRIPTOR_BLEN, offset 0x64(100)
	0000  09050609 C0000100 00072501 01000000  ..........%.....
```
It does not like audio endpoint length 9. I tried to return a corrected version of the descriptor, but it didn't help, the same USBD_STATUS_INVALID_PARAMETER for isoch transfers.

bozax, 2024-03-12:
udecx default set usb2.0 port status is HighSpeedDeviceAttached, but usb audio device is full speed device, UsbHub3 check isoch_transfer packets number alignment is not match 64 (for High speed ) , so irp complete status USBD_STATUS_INVALID_PARAMETER, these urb packets is not send to usbip-ude drivers.

vadimgrn, 2024-03-12:
Thank you very much, this is very valuable info for me! I'll deal with that as soon as will have free time.

vadimgrn, 2024-03-12:
Do you know how the best way to fix that?

nefarius, 2024-03-12:
> udecx default set usb2.0 port status is HighSpeedDeviceAttached, but usb audio device is full speed device, UsbHub3 check isoch_transfer packets number alignment is not match 64 (for High speed ) , [...]

This post is golden; I remember now that I noticed the UDE child device starting to act differently when I changed the device speed flags on creation and until now didn't know the details why that happened, or that it in fact *did* happen and I didn't just hallucinate 😅

nefarius, 2024-03-12:
> Do you know how the best way to fix that?

When calling [`UdecxUsbDeviceInitSetSpeed`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/udecxusbdevice/nf-udecxusbdevice-udecxusbdeviceinitsetspeed) set it to `UdecxUsbHighSpeed` instead of `UdecxUsbFullSpeed` ([see](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/udecxusbdevice/ne-udecxusbdevice-_udecx_usb_device_speed))

vadimgrn, 2024-03-14:
> UdecxUsbHighSpeed

It does not work, the same error occurs. You cannot just change the device speed because it has corresponding usb descriptors. For example, device descriptor has bcdUSB, etc.

vadimgrn, 2024-03-14:
> When calling [`UdecxUsbDeviceInitSetSpeed`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/udecxusbdevice/nf-udecxusbdevice-udecxusbdeviceinitsetspeed) set it to `UdecxUsbHighSpeed` [...]

It does not work, the same error occurs. You cannot just change the device speed because it has corresponding usb descriptors. For example, device descriptor has bcdUSB, etc.

nefarius, 2024-03-14:
> It does not work, the same error occurs. You cannot just change the device speed because it has corresponding usb descriptors. For example, device descriptor has bcdUSB, etc.

What `bInterval` value are you using in your configuration descriptor? I assume 1? Try it with 4.

vadimgrn, 2024-03-14:
Thank you, I'll try. usbip means that the driver receives usb transfers from the device driver on Windows and sends them to Linux server. The received results from Linux are returned back. I do nothing like you mentioned.

nefarius, 2024-03-14:
> Thank you, I'll try. usbip means that the driver receives usb transfers from the device driver on Windows and sends them to Linux server. The received results from Linux are returned back. I do nothing like you mentioned.

Let me rephrase that then; the device you currently use for your testing; what `bInterval`, uh, values, does it report in its configuration descriptor?

vadimgrn, 2024-03-14:
I'll answer tomorrow, today is too late.

vadimgrn, 2024-03-15:
```

    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 3-1
Properties               : 0x01
 IsUserConnectable       : yes
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x01 (Port 1)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#ROOT_HUB30#1&2b53a856&2e&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x1F (Port 31)
 -> CompanionPortChain   : 3-31

      ========================== Summary =========================
Vendor ID                : 0x0D8C (C-MEDIA ELECTRONICS INC.)
Product ID               : 0x0103
USB Version              : 1.1
Port maximum Speed       : High-Speed (Companion Port 3-31 supports SuperSpeed)
Device maximum Speed     : High-Speed
Device Connection Speed  : High-Speed
Self powered             : no
Demanded Current         : 500 mA
Used Endpoints           : 1

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_0D8C&PID_0103#2&2b7483fe&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-6
Device ID                : USB\VID_0D8C&PID_0103\2&2B7483FE&0&1
Hardware IDs             : USB\VID_0D8C&PID_0103&REV_0010 USB\VID_0D8C&PID_0103
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0007 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.22621.3155  Date: 2024-02-14)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0003
Container ID             : {33b69b3a-e201-11ee-9dc2-c018035e89da}
Manufacturer Info        : (Standard USB Host Controller)
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 1
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : USB Sound Device (USB Audio Device)
  Device Path 1          : \\?\USB#VID_0D8C&PID_0103&MI_00#3&879833&0&0000#{6994ad04-93ef-11d0-a3cc-00a0c9223196}\global (AM_KSCATEGORY_AUDIO)
  Device Path 2          : \\?\USB#VID_0D8C&PID_0103&MI_00#3&879833&0&0000#{65e8773e-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_RENDER)
  Kernel Name            : \Device\00000114
  Device ID              : USB\VID_0D8C&PID_0103&MI_00\3&879833&0&0000
  Class                  : MEDIA
  Driver KeyName         : {4d36e96c-e325-11ce-bfc1-08002be10318}\0006 (GUID_DEVCLASS_MEDIA)
  Service                : usbaudio
  Location               : 0000.0000.0000.001.000.000.000.000.000
   Child Device 1        : Speakers (USB Sound Device ) (Audio Endpoint)
    Device ID            : SWD\MMDEVAPI\{0.0.0.00000000}.{28E0363A-A464-4FCB-9E42-EE16ACC6B73A}
    Class                : AudioEndpoint
[... 249 lines cut ...]
wLockDelay               : 0x0000
Data (HexDump)           : 07 25 01 01 00 00 00                              .%.....

      -------------------- String Descriptors -------------------
             ------ String Descriptor 0 ------
bLength                  : 0x04 (4 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language ID[0]           : 0x0409 (English - United States)
Data (HexDump)           : 04 03 09 04                                       ....
             ------ String Descriptor 1 ------
bLength                  : 0x1A (26 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "C-Media INC."
Data (HexDump)           : 1A 03 43 00 2D 00 4D 00 65 00 64 00 69 00 61 00   ..C.-.M.e.d.i.a.
                           20 00 49 00 4E 00 43 00 2E 00                      .I.N.C...
             ------ String Descriptor 2 ------
bLength                  : 0x32 (50 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "USB Sound Device        "  *!*CAUTION  trailing space characters
Data (HexDump)           : 32 03 55 00 53 00 42 00 20 00 53 00 6F 00 75 00   2.U.S.B. .S.o.u.
                           6E 00 64 00 20 00 44 00 65 00 76 00 69 00 63 00   n.d. .D.e.v.i.c.
                           65 00 20 00 20 00 20 00 20 00 20 00 20 00 20 00   e. . . . . . . .
                           20 00                                              .

```

vadimgrn, 2024-03-15:
This is real info for the real device
```

    =========================== USB Port4 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 2-4
Properties               : 0x01
 IsUserConnectable       : yes
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x04 (Port 4)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#ROOT_HUB30#4&1474365&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x11 (Port 17)
 -> CompanionPortChain   : 2-17

      ========================== Summary =========================
Vendor ID                : 0x0D8C (C-MEDIA ELECTRONICS INC.)
Product ID               : 0x0103
USB Version              : 1.1
Port maximum Speed       : High-Speed (Companion Port 2-17 supports SuperSpeed)
Device maximum Speed     : Full-Speed
Device Connection Speed  : Full-Speed
Self powered             : no
Demanded Current         : 500 mA
Used Endpoints           : 1

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_0D8C&PID_0103#5&45549b7&0&4#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-7
Device ID                : USB\VID_0D8C&PID_0103\5&45549B7&0&4
Hardware IDs             : USB\VID_0D8C&PID_0103&REV_0010 USB\VID_0D8C&PID_0103
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0009 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.22621.3155  Date: 2024-02-14)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0004.Hub_#0002
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4), ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS04)
Container ID             : {33b69b86-e201-11ee-9dc2-c018035e89da}
Manufacturer Info        : (Standard USB Host Controller)
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 4
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : USB Sound Device (USB Audio Device)
  Device Path 1          : \\?\USB#VID_0D8C&PID_0103&MI_00#6&3169eb15&0&0000#{65e8773e-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_RENDER)
  Device Path 2          : \\?\USB#VID_0D8C&PID_0103&MI_00#6&3169eb15&0&0000#{6994ad04-93ef-11d0-a3cc-00a0c9223196}\global (AM_KSCATEGORY_AUDIO)
  Kernel Name            : \Device\00000118
  Device ID              : USB\VID_0D8C&PID_0103&MI_00\6&3169EB15&0&0000
  Class                  : MEDIA
  Driver KeyName         : {4d36e96c-e325-11ce-bfc1-08002be10318}\0007 (GUID_DEVCLASS_MEDIA)
  Service                : usbaudio
  Location               : 0000.0014.0000.004.000.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USBMI(0)  PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USB(4)  ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS04)#USBMI(0)  ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS04)#USB(4)
[... 248 lines cut ...]
bLockDelayUnits          : 0x00 (Undefined)
wLockDelay               : 0x0000
Data (HexDump)           : 07 25 01 01 00 00 00                              .%.....

      -------------------- String Descriptors -------------------
             ------ String Descriptor 0 ------
bLength                  : 0x04 (4 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language ID[0]           : 0x0409 (English - United States)
Data (HexDump)           : 04 03 09 04                                       ....
             ------ String Descriptor 1 ------
bLength                  : 0x1A (26 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "C-Media INC."
Data (HexDump)           : 1A 03 43 00 2D 00 4D 00 65 00 64 00 69 00 61 00   ..C.-.M.e.d.i.a.
                           20 00 49 00 4E 00 43 00 2E 00                      .I.N.C...
             ------ String Descriptor 2 ------
bLength                  : 0x32 (50 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "USB Sound Device        "  *!*CAUTION  trailing space characters
Data (HexDump)           : 32 03 55 00 53 00 42 00 20 00 53 00 6F 00 75 00   2.U.S.B. .S.o.u.
                           6E 00 64 00 20 00 44 00 65 00 76 00 69 00 63 00   n.d. .D.e.v.i.c.
                           65 00 20 00 20 00 20 00 20 00 20 00 20 00 20 00   e. . . . . . . .
                           20 00                      
```                        .

vadimgrn, 2024-03-15:
The difference is that real device connection speed is Full-Speed, but virtual device is connected as High-Speed, as you mentioned above.

nefarius, 2024-03-15:
This is most probably your issue on the endpoint descriptors:

```
bInterval                : 0x01 (1 ms)
```

I sadly do not remember where I exactly got that Eureka Moment from but I do recall that UDE (perhaps due to its dependency on USBHUB3) does not support 1ms polling, it **always** treats `bInterval` as 0.125ms intervals, not 1ms, and it doesn't care if everything else in the device descriptor or speed is correct, I had to set it to **at least** 4 to get my device(s) to work.

---

So the unholy trinity of getting a virtual USB audio device to work with UDE:
- Present the device as **High-Speed**, even if the physical one is Full-Speed
- The USB host controller device needs to provide an implementation of **QueryBusTime** that can report nonsense like always 0 but *has to* report `STATUS_SUCCESS`
- The `bInterval` of each ISOCH endpoint in the configuration descriptor **must not be smaller than 4**, even if the physical device reports 1

vadimgrn, 2024-04-28:
The issue has [fixed](6e7af2e84bca1a0f6480a17be5219274627fdda0)

closed 2024-04-28 by vadimgrn