vadimgrn/usbip-win2 #36: UDE 9.5.0 Bug

Geramy, 2023-04-01:
Hi, I have a NodeMCU ESP8266 with https://github.com/thevoidnn/esp8266-wifi-cmsis-dap installed on it which turns it in to a usb wifi device as a dap link probe. But it seems that when I actually attach the usb device it doesn't show up in device manager on windows. This device works fine on linux usbip.

Here is my COM output thanks so much!

s1 handling dev attach request...
s1 sending header...
s1 sending device info...
* GET 0x01 DEVICE DESCRIPTOR
* GET 0x01 DEVICE DESCRIPTOR
* GET 0x02 CONFIGURATION DESCRIPTOR
Sending ALL CONFIG

Here is my device manager for you to see not sure why  usbip root is not installed with a driver either.

[image]

vadimgrn, 2023-04-01:
Hi, the driver log is required.

Geramy, 2023-04-01:
How do I get the driver log for you?

vadimgrn, 2023-04-01:
https://github.com/vadimgrn/usbip-win2#obtaining-usbip-logs-on-windows

Geramy, 2023-04-01:
So I don't see it crashing but I also don't see any new device in my device manager or by using openocds included USB Driver Tool.

[%9]%3!04x! %!LEVEL! %!FUNC!:vhci_plugin_hardware(0X22E000), OutputBufferLength 8, InputBufferLength 1100
[%9]%3!04x! %!LEVEL! %!FUNC!:10.21.1.164:3240, busid 1-1
[%9]%3!04x! %!LEVEL! %!FUNC!:get keepalive: idle(7200 sec) + cnt(10)*intvl(1 sec) => 7210 sec
[%9]%3!04x! %!LEVEL! %!FUNC!:set keepalive: idle(30 sec) + cnt(9)*intvl(10 sec) => 120 sec
[%9]%3!04x! %!LEVEL! %!FUNC!:Connected to 10.21.1.164:3240
[%9]%3!04x! %!LEVEL! %!FUNC!:usbip_usb_device(path '/sys/devices/pci0000:00/0000:00:01.2/usb1/1-1', busid 1-1, busnum 1, devnum 2, USB_SPEED_FULL,vid 0x16c0, pid 0x5df, rev 0x130, class/sub/proto 0/0/0, bConfigurationValue 1, bNumConfigurations 1, bNumInterfaces 1)
[%9]%3!04x! %!LEVEL! %!FUNC!:wsk workitem 4dab97e8
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58, queue 4f2b8f68
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58, port 1
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58 -> port 1
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 8
[%9]%3!04x! %!LEVEL! %!FUNC!:PreviousState 0x4
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58, endp 3ffbbed8{Address 0000: Ctrl Out[0], MaxPacketSize 0, Interval 0}, queue 35522f68
	0000  07050000 000000                      .......
	
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58, EndpointsToConfigure[1]
	0000  D8BEFB3F 741A0000                    ...?t...
	
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58, DeviceInitialize
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 -> PipeHandle cace66a0, IN|SHORT_OK|DEFAULT_PIPE, TransferBufferLength 64, Timeout 5000, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x40(64)}
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 -> 48{seqnum 3, devid 0x10002, in[0]}, cmd_submit: flags 0, length 64, start_frame 0, isoc[-1], interval 0, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x40(64)}
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 <- 66{seqnum 3, devid 0x10002, in[0]}, ret_submit: status 0, actual_length 18, start_frame 0, isoc[0], error_count 0
[%9]%3!04x! %!LEVEL! %!FUNC!:bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE)
	0000  12011001 00000040 C016DF05 30010102  .......@....0...
	0010  0301                                 ..
	
[%9]%3!04x! %!LEVEL! %!FUNC!:DeviceDescriptor(bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE), bcdUSB 0x110, bDeviceClass 0, bDeviceSubClass 0, bDeviceProtocol 0, bMaxPacketSize0 64, idVendor 0x16c0, idProduct 0x5df, bcdDevice 0x130, iManufacturer 1, iProduct 2, iSerialNumber 3, bNumConfigurations 1)
[%9]%3!04x! %!LEVEL! %!FUNC!:dev 320f8e58, endp 3ffbbed8, queue 35522f68
[%9]%3!04x! %!LEVEL! %!FUNC!:endp 3ffbbed8, queue 35522f68
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 -> PipeHandle cace66a0, IN|SHORT_OK|DEFAULT_PIPE, TransferBufferLength 18, Timeout 5000, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x12(18)}
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 -> 48{seqnum 5, devid 0x10002, in[0]}, cmd_submit: flags 0, length 18, start_frame 0, isoc[-1], interval 0, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x12(18)}
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 <- 66{seqnum 5, devid 0x10002, in[0]}, ret_submit: status 0, actual_length 18, start_frame 0, isoc[0], error_count 0
[%9]%3!04x! %!LEVEL! %!FUNC!:bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE)
	0000  12011001 00000040 C016DF05 30010102  .......@....0...
	0010  0301                                 ..
	
[%9]%3!04x! %!LEVEL! %!FUNC!:DeviceDescriptor(bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE), bcdUSB 0x110, bDeviceClass 0, bDeviceSubClass 0, bDeviceProtocol 0, bMaxPacketSize0 64, idVendor 0x16c0, idProduct 0x5df, bcdDevice 0x130, iManufacturer 1, iProduct 2, iSerialNumber 3, bNumConfigurations 1)
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 -> PipeHandle cace66a0, IN|SHORT_OK|DEFAULT_PIPE, TransferBufferLength 255, Timeout 5000, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x200, wIndex 0000, wLength 0xff(255)}
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 -> 48{seqnum 7, devid 0x10002, in[0]}, cmd_submit: flags 0, length 255, start_frame 0, isoc[-1], interval 0, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x200, wIndex 0000, wLength 0xff(255)}
[%9]%3!04x! %!LEVEL! %!FUNC!:req 5988d208 <- 303{seqnum 7, devid 0x10002, in[0]}, ret_submit: status 0, actual_length 255, start_frame 0, isoc[0], error_count 0
[%9]%3!04x! %!LEVEL! %!FUNC!:vhci_get_imported_devices(0X22E008), OutputBufferLength 4436, InputBufferLength 4
[%9]%3!04x! %!LEVEL! %!FUNC!:1 device(s) reported
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 1112
[%9]%3!04x! %!LEVEL! %!FUNC!:vhci_get_imported_devices(0X22E008), OutputBufferLength 4436, InputBufferLength 4
[%9]%3!04x! %!LEVEL! %!FUNC!:1 device(s) reported
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 1112
[%9]%3!04x! %!LEVEL! %!FUNC!:vhci_get_imported_devices(0X22E008), OutputBufferLength 4436, InputBufferLength 4
[%9]%3!04x! %!LEVEL! %!FUNC!:1 device(s) reported
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 1112
[%9]%3!04x! %!LEVEL! %!FUNC!:vhci_get_imported_devices(0X22E008), OutputBufferLength 4436, InputBufferLength 4
[%9]%3!04x! %!LEVEL! %!FUNC!:1 device(s) reported
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 1112
[... cut ...]
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 1112
[%9]%3!04x! %!LEVEL! %!FUNC!:vhci_get_imported_devices(0X22E008), OutputBufferLength 4436, InputBufferLength 4
[%9]%3!04x! %!LEVEL! %!FUNC!:1 device(s) reported
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 1112
[%9]%3!04x! %!LEVEL! %!FUNC!:vhci_get_imported_devices(0X22E008), OutputBufferLength 4436, InputBufferLength 4
[%9]%3!04x! %!LEVEL! %!FUNC!:1 device(s) reported
[%9]%3!04x! %!LEVEL! %!FUNC!:STATUS_SUCCESS, Information 1112

Geramy, 2023-04-02:
Also whats going on with the usbip root devices being unknown? I suspect maybe that could have something to do with it?
[image]

vadimgrn, 2023-04-02:
Make log exactly as described, please.

vadimgrn, 2023-05-26:
Malformed log

vadimgrn, 2023-05-26:
Malformed log

closed 2023-05-26 by vadimgrn