vadimgrn/usbip-win2 #37: Multiple bug &  "low speed"

DzzD, 2023-04-30:
Hello I have tried the last three version : 
- 9.5.2 : usbip with or without parameters command output nothing
- 9.3.4 : usbip with or without parameters command output nothing
- 9.3.3 : usbip can list and attach but use "low speed" on USP cam where before it was using "fulll speed"

Let's try 9.3.2 !

vadimgrn, 2023-04-30:
> * usbip with or without parameters command output nothing

I don't get it. Use --debug option. Empty output is OK if nothing to output.

DzzD, 2023-04-30:
wait, I unistall/reinstall (i try multiple version)

DzzD, 2023-04-30:
I mean this : 

[image]

DzzD, 2023-04-30:
this happen with the latest version 9.5.2 (9.3.4 works, but still use low speed where it should not)

DzzD, 2023-04-30:
Ok installing the debug version I got this : 
[image]

this explain why the release version did not output anything to the console

vadimgrn, 2023-04-30:
I'll check it tomorrow, thank you.

DzzD, 2023-04-30:
For information, if it can help, with the previous version (9.3.3) I got : 

[image]

Thanks to you !

DzzD, 2023-04-30:
Some more informations : 
With the version 9.3.3 I managed to make it works at full speed, it seems there is a maybe a mess between the two drivers, dont know.

If I deactivate xhci driver I fot error on usbip pot, but my camera is working well, I can use it maximum resolution.

[image]

vadimgrn, 2023-05-01:
When 0.9.3.X and 0.9.5.X are installed, make sure that you are calling right usbip.exe. It is binary incompatibility. I checked the latest release, all is OK.

DzzD, 2023-05-01:
I always unsinstall the previous version, I use the one that come with the installer is that ok ?

DzzD, 2023-05-01:
Re-tested, same, uninstall/reinstall and I got : 
[image]

DzzD, 2023-05-01:
Another question, why some device appear in "low speed" ? 

They was working well before, but now they are set as low speed

[image]

vadimgrn, 2023-05-01:
You should completely uninstall all usbip drivers. You should do it manually as described in readme.

DzzD, 2023-05-01:
Well I ll try, in the meantime I have found a cheap software that works well for me and wich look stable (wich shows that you have great potential to sell this one once packager & reliable)

vadimgrn, 2023-05-01:
I am not aware about commercial usbip implementations for Windows.

DzzD, 2023-05-01:
I just tried Virtualhere, it works flawlessly for me : 

[image]

I have virtualised three USB of a raspberrypi3 to control my laser from lightburn (+ a controle camera) using WIFI. 

Everything works perfectly, it is free for one USB over IP on a device and 50$ for unlimited (I have bought their license)

DzzD, 2023-05-01:
It use it own server (server installation requiere only one line, and is finished in less than 30s on raspberry)

Client side everything is in a single program.

Full server/client installation took less than 5 minutes.

vadimgrn, 2023-05-02:
There are a lot of proprietary solutions for sharing USB devices. VirtualHere is one of them, it can't work with Linux usbip server.

DzzD, 2023-05-02:
Actually, I would have prefer to use your solution as it is open source and already pretty advanced, but for now Virtualhere do the job very well and very reliabily without requiring windows test mode. I will keep an eye here. Keep up the good work !

closed 2023-05-26 by vadimgrn