vadimgrn/usbip-win2 #39: WDM: Failure to receive HID reports

nraggett, 2023-05-26:
Hey Vadym!
I've been experimenting around with a device which has its config loaded through repetitive HID reports.
When loading through UDE, the reports come through as expected, with no issues.
When loading through WDM, the reports appear to first come through with null data, then are double-received, leading to a total report twice the expected size.

When capturing this operation through Wireshark, it appears the process is exactly the same, with no incorrect actions taken. So the issue must be between the WDM driver and its handling of the userspace USB requests.

The actual userspace USB request is handled through this windows HIDAPI function:
```
BOOL ret = DeviceIoControl(dev->device_handle,
		IOCTL_HID_GET_FEATURE,
		data, (DWORD)length,
		data, (DWORD)length,
		&bytes_returned, &overlap);
```

Attached below is:
- [The kernel log of the WDM driver](https://github.com/vadimgrn/usbip-win2/files/11571602/usbip-vhci.txt)
- [Wireshark capture of the UDE](https://github.com/vadimgrn/usbip-win2/files/11571638/USB.dump.UDE.pcapng.gz)
    - [and WDM operations](https://github.com/vadimgrn/usbip-win2/files/11571644/USB.dump.WDM.responses.pcapng.gz)
- [The good output of the userspace application capturing the HID reports](https://github.com/vadimgrn/usbip-win2/files/11571655/survive_good.log)
    - [and the WDM output of the userspace application capturing the HID reports](https://github.com/vadimgrn/usbip-win2/files/11571662/survive_wdm.log)

Is there anywhere I can look within the WDM drivers to trace back how HID reports are handled?

vadimgrn, 2023-06-02:
IRP_MJ_DEVICE_CONTROL, IOCTL_HID_GET_FEATURE is not implemented.
See ioctl_vhci.cpp, vhci_ioctl_vhci

The question is how to obtain this report on Linux using usb transfers, if it is possible.

vadimgrn, 2024-04-28:
WDM driver has retired

closed 2024-04-28 by vadimgrn