vadimgrn/usbip-win2 #42: Windows server 2019 support

srdimitroff, 2023-06-27:
Interested installing this on a VM running Windows Server 2019 edition. When I try running the installer it says that "This Program does not support the version of windows your computer is running" 

Windows Server 2019 is more or less similar enough to Windows 10/11. Thanks!

vadimgrn, 2023-06-28:
What exact version of Windows?

srdimitroff, 2023-06-28:
Thanks for the quick reply. Under windows edition names it:

Windows Server 2019 Standard

vadimgrn, 2023-06-28:
It should be like 
Windows 10 Version 2004 (OS build 19041). I use API that is available since that release for Win10.

srdimitroff, 2023-06-28:
Yeah I thought it should be similar to that with Server 2019, however the .exe to install it will not start due to  the error:

"This program does not support the version of Windows your computer is running."

[image]

vadimgrn, 2023-06-28:
I see, but I want you to get full version info about your Win Server. It contains version and OS build numbers. Win Server 2019 is not enough, I suppose.

vadimgrn, 2023-06-28:
It seems so. It's possible to replace ExAllocatePool2 as described [here](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exallocatepool2). If you are a programmer, you could do that.

vadimgrn, 2023-06-29:
There are several places where Win version should be changed as well.
[image]
[image]
[image]

vadimgrn, 2023-06-29:
It won't be so easy. There are some other functions that available since Windows 10, version 2004. For example, RtlUTF8StringToUnicodeString and RtlInitUTF8String.

vadimgrn, 2023-07-01:
I did it :) It should run on Windows 10 x64 Version 1709(OS build 16299) and later.
Could you build master branch and try it, please?

srdimitroff, 2023-07-02:
Awesome work, I appreciate this! I did end up building it from the master last night and was having some issues with the drivers after the installation. I tried your build and it was doing the same. When I install it, the USBip 3.X Emulated Host Controller driver installs but says:

"Windows cannot start this hardware device because its configuration information (in the registry) is incomplete or damaged. (Code 19)"

[image]

[image]

So I tried restarting the VM and then all the USB controllers have that same error message:

[image]

I am currently testing it out on Win10 LTSC version 1809 build 17763.3887 which will be similar to the Windows Server 2019 I eventually want to run it on.

vadimgrn, 2023-07-02:
I don't have Win10 to test. Collect logs, pls.

srdimitroff, 2023-07-03:
Which logs can I provide? I ran the verifier for the driver and can provide that?

vadimgrn, 2023-07-03:
Today I installed Win10 1709 on VM. There are issues with .inf files for drivers. Other than that "pnputil /remove-device" available starting in Windows 10 version 2004. Errors for .inf files are very generic, it is not possible to understand what exactly wrong. I'm going to move these changes to a new branch, master branch will be reverted to Win 10 2004.

vadimgrn, 2023-07-03:
https://github.com/vadimgrn/usbip-win2/tree/win10_1709

srdimitroff, 2023-07-05:
Does the driver need that "pnputil /remove-device" to run properly? I'm not sure if that is only needed when uninstalling the driver or is it for detaching usbip device?

vadimgrn, 2023-07-05:
It is required during the uninstallation.

vadimgrn, 2023-07-11:
I found the solution, downgrade of Win version will be implemented in develop branch soon.

vadimgrn, 2023-07-17:
The branch [win10_1809](https://github.com/vadimgrn/usbip-win2/tree/win10_1809) is completed, usbip can be installed/removed correctly. I'm thinking if it worth to merge this branch into develop.

The only issue is that usbip2_ude service remains in registry after the uninstallation on Win10 1809, but it is harmless.

vadimgrn, 2023-07-18:
The branch has merged into develop.

closed 2023-07-18 by vadimgrn