vadimgrn/usbip-win2 #43: HID Device not working after installing v.0.9.5.5

xshrinix, 2023-08-10:
After Installation and rebooting the device (Windows 10 pro). Both Mouse and Keyboard connected to that machine stopped working. I tried connecting other Mouse keyboard and also tried to plug in on all USB port, Mouse and keyboard just not working after installation.

vadimgrn, 2023-08-10:
Requirements: Windows 10 Version 1809 (OS build 17763) and later.
https://github.com/vadimgrn/usbip-win2#enable-test-signing-without-removing-usbip

xshrinix, 2023-08-10:
So one must not restart machine immediately after installation. We first disable the usbip2_filter then we can reboot the machine.
Is that the correct order?

vadimgrn, 2023-08-10:
What Win version do you have?

xshrinix, 2023-08-11:
Windows 10 pro OS build 19045.xx. version - 22H2. After USB driver installation at the end, system asks for reboot, so i rebooted.

vadimgrn, 2023-08-11:
If usbip2_filter driver can't be loaded, all usb devices will not work. It should work on your system, but something went wrong. I'll try to help you resolve this issue, but slightly later, I'm busy right now.

estintax, 2023-08-11:
Today I also encountered this problem, I have been trying to restore the operation of HID devices for two hours, but to no avail

vadimgrn, 2023-08-13:
Device Manager, Universal Serial Bus Controllers, pay attention to
- USB Root Hub (USB 3.0), Location: on USBip 3.X Emulated Host Controller
- USBip 3.X Emulated Host Controller
[image]
[image]
[image]
[image]

vadimgrn, 2023-08-13:
If these devices are not working, most likely usbip2_filter.sys can't be loaded. This is a filter driver, see HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}, UpperFilters. If you remove usbip2_filter
 from UpperFilters and everything will work after the reboot, this is the reason.

vadimgrn, 2023-08-13:
To find out what is going on, you could use C:\Windows\INF\setupapi.*.log files (setupapi.app.log, setupapi.dev.log). Remove these files before the installation of USBip, and post them here after the installation.

vadimgrn, 2023-08-14:
> Today I also encountered this problem, I have been trying to restore the operation of HID devices for two hours, but to no avail. I can't interact with the system in any way

Start your PC in safe mode in Windows (press F11 during early boot process) and uninstall USBip.

sylviiu, 2023-09-19:
I'm also having this problem. I'm still unable to interact with the system in safe mode, unfortunately. Only way I can mess with my system right now is through the command prompt in that "Automatic Repair" prompt.

vadimgrn, 2023-09-19:
Any Windows PC can be booted in safe mode, just find out how to do that.
After that you will be able to uninstall USBip.
https://support.microsoft.com/en-us/windows/start-your-pc-in-safe-mode-in-windows-92c27cff-db89-8644-1ce4-b3e5e56fe234

sylviiu, 2023-09-19:
Sorry, originally deleted comment because I decided to go for another approach.

I was able to successfully boot into safe mode, but I want to say that the driver still attempted to load in safe mode because the other physically connected devices were still not able to function.

I ended up doing a system restore, which restored functionality.

vadimgrn, 2023-09-19:
That means that Windows is trying to start UpperFilters drivers even in Safe Mode for
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}. As a result, USB keyboard/mouse do not work in safe mode too, I guess.

closed 2024-05-03 by vadimgrn