vadimgrn/usbip-win2 #45: bsod when trying to forward smart card or yubikey

paweljasinski, 2023-11-10:
Hi,

I have tried the following combinations. In all cases trying to connect smartcard reader or yubikey triggers bsod in windows client.

Status  Server      Client               Device
-------------------------------------------------------------------
bsod    Android    Windows          Omnikey Smart Card Reader
bsod    Android    Windows          Yubikey
bsod    Linux        Windows          Yubikey
ok        Linux        Windows          Mac keyboard
ok        Android    Linux                Yubikey
ok        Linux        Linux                Yubikey

Cheers
--pawel

vadimgrn, 2023-11-10:
https://github.com/vadimgrn/usbip-win2/tree/develop#debugging-bsod

vadimgrn, 2023-12-15:
Crash log is missing

closed 2023-12-15 by vadimgrn