vadimgrn/usbip-win2 #46: How can i improve quality of usb audio?

Kobain-Seo, 2023-11-29:
Hi, I am very impressed by what you've done. This project is the most stable and has the best performance within usb ip softwares. 

I have been trying to pass USB headset to remote PC. When the driver is ude, I does not work at all. But in wdm mode, it works pretty. But there is cracking sound intermittently. Is there anything I can do for resolving crackling sound?

vadimgrn, 2023-11-29:
Hi, thanks for the feedback! Unfortunately, this cannot be improved upon as the sound is very sensitive to the latency. TCP/IP is a poor choice for network audio, but USBIP protocol is based on it, and there is currently no alternative.

Kobain-Seo, 2023-11-30:
> Hi, thanks for the feedback! Unfortunately, this cannot be improved upon as the sound is very sensitive to the latency. TCP/IP is a poor choice for network audio, but USBIP protocol is based on it, and there is currently no alternative.
> 
> Non working sound in the IDE driver is a problem I've been dealing with for a year. But I'm stuck and have no more ideas how to fix it for now.

Then, What if I increase latency, will it be improved?

vadimgrn, 2023-11-30:
You can't do that, it's a network. The lower latency the better.

Kobain-Seo, 2023-11-30:
> You can't do that, it's a network. The lower latency the better.

What i meant is using buffer to increase latency, not physical latency. If it is latency from buffer, isn't it still a shot?

closed 2024-04-28 by vadimgrn

makcorner, 2024-12-26:
Hi, any chance to use UDP as a transport for audio devices?

vadimgrn, 2024-12-27:
usb/ip protocol uses tpc/ip

makcorner, 2024-12-27:
Thanks for your prompt answer. Yes, I know it uses TCP from the previous posts. I'm just wondering if implementing UDP for some devices, like audio, would improve the final quality. It is quite obvious that transferring audio/video stream over UDP is often a better fit. But maybe it is quite difficult to implement for some reason?

vadimgrn, 2024-12-27:
It is not possible. First of all, Linux server does not have and will not have such implementation. But the major issue is that usb/ip must forward usb transfers bit by bit.