vadimgrn/usbip-win2 #47: some experiences using usbip win-win and win-lin

comsyspro, 2023-12-11:
i'm working on a gui for the usbip client side for windows:
[image]
here you can easily attach and detach usb devices (usbip client gui working for windows and linux server)

here i made different experiences connecting from 
-windows (client) to windows (server)
and 
-windows (client) to linux (server)

windows (client) to windows (server):
-here attaching the usb devices works very easy, but some usb devices like usb storage sticks are only working with wdm drivers
-so you have to adapt the devices to different usbip.exe versions (wdm or ude driver)

windows (client) to linux (server):
-attaching usb devices sometimes needs fixes to work properly, so you have to disable and re-enable the usb device after attaching. so you have to create rules for some vendors or devices to get it work. these fixes are not needed in windows (client) to windows (server) scenario.
-ude drivers often leeds to bluescreen

results:
-crossplatform usage needs some tweaking and fixes with rules for wdm and ude usage
-it is not possible to use only one driver version, you have to mix both drivers and have to decide which usb device will work with which driver by testing and creating rules
-it is important to have the wdm driver version, because many usb devices will work only with that driver

questions:
-can you also say or comment something to these observations?
-would it be possible to have only one version of usbip-win2 which will work crossplatform or do you need both driver versions (wdm and ude)?
-can this info help you in developping further the usbip-win2 project?

vadimgrn, 2023-12-11:
I have several usb flash drives and all work without issues.

comsyspro, 2023-12-11:
from **windows to windows** the usb flash works also with ude for me, but from **windows to linux** i have to use wdm as described. i don't know why that's the case.

vadimgrn, 2023-12-11:
I don't care about Win server, I use Linux server only.

vadimgrn, 2023-12-11:
I use Linux Mint with kernel 6.2 currently. Previous kernels work well too.

comsyspro, 2023-12-11:
thank you. overall it works now either with wdm or ude drivers. there ar some devices like card readers which i can't get work, but that is not so important for me. i wonder how other usb redirection software solutions can handle all types of usb devices.

vadimgrn, 2023-12-12:
Are you sure about that? I worked for major developer of such proprietary software with its own protocol instead of usbip. It had, has and will always have tons of bugs. I can't solve one real [problem](https://github.com/vadimgrn/usbip-win2/issues/35) in UDE driver

comsyspro, 2023-12-12:
i tried different software solutions and they do support for example out of the box some of my usb devices.  (usb storage sticks, graphic tablet, 3d mouse, multi card reader...). so i got the impression that they do soemthing different or have an own development. now i can get the most usb devices running with your usbip-win2 project on windows client. as decribed i have to experiment which driver to use best for which usb device. also making tests with win to win and win to lin until i got something running stable.

i have also another question. in the latest code you have a folder with ude and wdm driver. when compiling this is then only the ude driver included in the innosetup? or is it possible to add both drivers in one innosetup or is wdm only in wdm branch and development is stopped for wdm?

vadimgrn, 2023-12-12:
WDM is in "wdm" branch, I will delete its directory from UDE branch to not confuse people.

vadimgrn, 2023-12-12:
SDK will install C++ headers, PDB are required for debugging BSOD.

closed 2024-01-29 by comsyspro