vadimgrn/usbip-win2 #48: can you remove testsigning after installing winusb?

comsyspro, 2023-12-11:
is it possible to disable testsigning with `bcdedit.exe /set testsigning off` after you have installed winusb with the test certificates? is this possible for both ude and wdm driver releases or only for ude?

vadimgrn, 2023-12-11:
You can (disable)[https://github.com/vadimgrn/usbip-win2#disable-windows-test-signing-mode-without-removing-usbip] test signing, but usbip will not work until it will be enabled.

comsyspro, 2023-12-11:
yes, i did try that and also deleted the UpperFilters value in the registry. after rebooting you get a yellow warning mark on "usbip root" and USBip 3.X" in device manager and usbip is out of function. what would it cost to get a signed driver?

vadimgrn, 2023-12-11:
https://github.com/vadimgrn/usbip-win2/issues/13

closed 2023-12-15 by vadimgrn

Oleszkiewicz, 2024-01-12:
Hi, if the problem is the business entity - I can help with that, I have a company in Poland that is already a Microsoft partner, Provided we get funding for the cert I can go through EV code signing validation. From what I see it is possible to get one from Comodo a bit south of USD 900 or I can get one locally from Certum at approx USD 1050 for 3 years.

vadimgrn, 2024-01-12:
Hi, it would be nice.

Oleszkiewicz, 2024-01-12:
So how would we proceed ?

vadimgrn, 2024-01-12:
Sorry, I'm busy right now.I'll answer later.

vadimgrn, 2024-01-15:
So the question of money to buy an EV certificate?

Oleszkiewicz, 2024-01-15:
yup, I can donate my time to go through the certification process, recompile and sign, but no possibility to fund this from the company right now.

vadimgrn, 2024-01-15:
Donations have been very rare in recent months, I don't even remember the last time. I don't want to spend my own money for that during the war.

Oleszkiewicz, 2024-01-15:
Sure , I don’t expect that, is there actually a possibility to set some kind of a goal on github or somewhere else? Are you safe in the war?

comsyspro, 2024-03-21:
what is the actual best offer for a 3 three year driver certificate? could you create a kickstarter project https://www.kickstarter.com/ to collect money?

vadimgrn, 2024-03-22:
There are two options to sponsor this project, the third is not required.

comsyspro, 2024-03-25:
i mentioned kickstarter as you can see how much is already collected and you can define a timespan for collecting the reuired money. at the moment you can not know how much money is collected from some interested people and you can't know when the certificate will be ready. so can you publish this somehow or with such a platform like kickstarter.

eebssk1, 2024-04-01:
Maybe ask https://about.signpath.io/product/open-source ?

fredemmott, 2024-04-28:
SignPath's open source offering does not include EV certificates, so isn't useful here.

FirstS0ul, 2024-10-24:
How is the Status and how can i donate? :-)

vadimgrn, 2024-10-25:
[image]