vadimgrn/usbip-win2 #49: error: No connection could be made because the target machine actively refused it.

Petros0000, 2023-12-14:
Hi,

I/m using a Raspberry Zero with a clean install of Debian Bullseye Lite and a laser engraver connected to its USB port. USBip makes a successful bind on that side.

On windows 11, after installing USBip I try to access it but get the following error:

error: No connection could be made because the target machine actively refused it.

I can successfully ssh to the raspberry, its IP address is correct, I also used Virtualhere and successfully connected to the laser(but communication freezes after initial connection, that's why I'm trying USBip).

Do you have any idea how I can proceed? 

Thanks!

vadimgrn, 2023-12-15:
Hi, usbip daemon must be running on server side.

Petros0000, 2023-12-17:
You were right, I somehow missed that. 

Unfortunately a connection with the laser engraver cannot be established. The device appears in Device Manager but the control software(Lightburn) either hangs or simply waits forever for a connection. The engraver uses an ESP32 so it's USB-> Serial, I don't know if that helps at all(Creality Falcon). I know there are many people online that have successfully connected their lasers but I believe I've done everything right. I connected a thumb drive and have no issues there, I can freely browse and play videos from it. I tried both UDE and WDM drivers.

vadimgrn, 2023-12-17:
You could gather driver log, sometimes it is possible to fix issues without having a device.

Petros0000, 2023-12-17:
[usbip-flt.txt](https://github.com/vadimgrn/usbip-win2/files/13697956/usbip-flt.txt)
[usbip-ude.txt](https://github.com/vadimgrn/usbip-win2/files/13697957/usbip-ude.txt)
[CMD Log.txt](https://github.com/vadimgrn/usbip-win2/files/13697970/CMD.Log.txt)

After I started logging, I attached the device, ran Lightburn and issued a couple of commands(Go to Home, Frame, Device settings etc). Lightburn never managed to connect(I get a "Waiting for connection" message). In device manager it shows up successfully under COM/LPT ports tab as USB3, and the esspresif Serial to USB embedded in ESP32 under one of the properties tab.

I've attached the log files, I wouldn't know what to make of them but you might find something in there. Given that a thumb drive connects just fine I think this has to do with how this particular laser engraver communicates which might be different to other engravers that communicate successfully through usbip.

vadimgrn, 2023-12-27:
Unfortunately, there is nothing unusual in the log. Transfers are successful, can't find out the cause of the issue.