vadimgrn/usbip-win2 #50: Calling usbip in cmd resulted in app crash in windows, but calling usbip.exe is working fine

shaoboon, 2023-12-15:
usbip app crash event details:-
Faulting application name: usbip.exe, version: 0.9.5.7, time stamp: 0x654e04b0
Faulting module name: ucrtbase.dll, version: 10.0.19041.3636, time stamp: 0x81cf5d89
Exception code: 0xc0000409
Fault offset: 0x000000000007286e
Faulting process id: 0x1fec
Faulting application start time: 0x01da2f67cc8b8e7c
Faulting application path: C:\Program Files\USBip\usbip.exe
Faulting module path: C:\Windows\System32\ucrtbase.dll
Report Id: 9c6180f4-8941-49ce-93d8-5faf1a3ff571
Faulting package full name: 
Faulting package-relative application ID: 

C:\Users\Administrator>usbip --help     <<<  no output

C:\Users\Administrator>usbip.exe --help
USB/IP client
Usage: usbip.exe [OPTIONS] SUBCOMMAND

Options:
  -h,--help                   Print this help message and exit
  -V,--version                Display program version information and exit
  -d,--debug                  Debug output
  -t,--tcp-port TEXT:INT in [1024 - 65535] [3240]
                              TCP/IP port number of USB/IP server

Subcommands:
  attach                      Attach to a remote/stashed USB device(s)
  detach                      Detach a remote USB device
  list                        List exportable/stashed USB devices
  port                        Show/stash imported USB devices

vadimgrn, 2023-12-15:
The bug has fixed in master/develop branches, can't attach executable due to the github restrictions.

vadimgrn, 2023-12-26:
fixed

closed 2023-12-26 by vadimgrn