vadimgrn/usbip-win2 #52: usbip_ude.sys crash

375583810a, 2023-12-18:
connect xbox gamepad will definitely crash，

who can help me
[image]

375583810a, 2023-12-19:
1: kd> !wdfkd.wdfsearchpath %TEMP%\USBip
Trace searchpath is: %TEMP%\USBip
1: kd> !wdfkd.wdfsettraceprefix [%9]%3!04x! %!LEVEL! %!FUNC!:
SetTracePrefix: "[%9]%3!04x! %!LEVEL! %!FUNC!:"
1: kd> !wdfkd.wdflogdump usbip2_ude -d
Trace searchpath is: %TEMP%\USBip

Trace format prefix is: [%9]%3!04x! %!LEVEL! %!FUNC!:
Trying to extract TMF information from - e:\git\usbip\usbip-win2\x64\debug\usbip2_ude.pdb
--- start of log ---
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send:req b902b898 -> wsk irp 41ff3e10, 48 bytes, 0x00000103(STATUS_PENDING)
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send_complete:req b902b898 -> wsk irp 41ff3e10, STATUS_SUCCESS, Information 48
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive_complete:req 0000, STATUS_SUCCESS, Information 48
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::ret_command:req b902b898 <- 66{seqnum 33, devid 0, in[0]}, ret_submit: status 0, actual_length 18, start_frame 0, isoc[0], error_count 0
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive:wsk irp 446498c0, 18 bytes, 0x00000103(STATUS_PENDING)
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive_complete:req b902b898, STATUS_SUCCESS, Information 18
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::post_control_transfer:bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE)
	0000  12010002 FF47D040 5E04120B 09050102  .....G.@^.......
	0010  0301                                 ..
	
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::log:DEV: bLength 18, bcdUSB 0x200, bDeviceClass 0xff, bDeviceSubClass 0x47, bDeviceProtocol 0xd0, bMaxPacketSize0 64, idVendor 0x45e, idProduct 0xb12, bcdDevice 0x509, iManufacturer 1, iProduct 2, iSerialNumber 3, bNumConfigurations 1
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive:wsk irp 446498c0, 48 bytes, 0x00000103(STATUS_PENDING)
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::endpoint_purge:dev baea6708, endp be8dac28, queue bc895f68
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::endpoint_start:endp be8dac28, queue bc895f68
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::control_transfer:req c0841f68 -> PipeHandle 4786d760, IN|SHORT_OK|DEFAULT_PIPE, TransferBufferLength 18, Timeout 5000, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x12(18)}
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send:req c0841f68 -> 48{seqnum 35, devid 0x20001, in[0]}, cmd_submit: flags 0, length 18, start_frame 0, isoc[-1], interval 0, {IN|STANDARD|DEVICE, GET_DESCRIPTOR(0x6), wValue 0x100, wIndex 0000, wLength 0x12(18)}
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send:req c0841f68 -> wsk irp 41ff3e10, 48 bytes, 0x00000103(STATUS_PENDING)
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send_complete:req c0841f68 -> wsk irp 41ff3e10, STATUS_SUCCESS, Information 48
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive_complete:req 0000, STATUS_SUCCESS, Information 48
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::ret_command:req c0841f68 <- 66{seqnum 35, devid 0, in[0]}, ret_submit: status 0, actual_length 18, start_frame 0, isoc[0], error_count 0
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive:wsk irp 446498c0, 18 bytes, 0x00000103(STATUS_PENDING)
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive_complete:req c0841f68, STATUS_SUCCESS, Information 18
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::post_control_transfer:bLength 18, 0x00000001(USB_DEVICE_DESCRIPTOR_TYPE)
	0000  12010002 FF47D040 5E04120B 09050102  .....G.@^.......
	0010  0301                                 ..
	
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::log:DEV: bLength 18, bcdUSB 0x200, bDeviceClass 0xff, bDeviceSubClass 0x47, bDeviceProtocol 0xd0, bMaxPacketSize0 64, idVendor 0x45e, idProduct 0xb12, bcdDevice 0x509, iManufacturer 1, iProduct 2, iSerialNumber 3, bNumConfigurations 1
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::receive:wsk irp 446498c0, 48 bytes, 0x00000103(STATUS_PENDING)
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::endpoints_configure:dev baea6708, EndpointsToConfigure[2]
	0000  28918DBE FC290000 A8AA8DBE FC290000  (....).......)..
	
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::endpoints_configure:dev baea6708, InterfaceNumber 2, NewInterfaceSetting 0
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::endpoint_start:endp be8d9128, queue b5551478
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::endpoint_start:endp be8daaa8, queue c084ca58
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::bulk_or_interrupt_transfer:req c0841f68 -> PipeHandle 41f9fc30, IN|SHORT_OK, TransferBufferLength 64 
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send:req c0841f68 -> 48{seqnum 37, devid 0x20001, in[2]}, cmd_submit: flags 0, length 64, start_frame 0, isoc[-1], interval 4
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send:req c0841f68 -> wsk irp 41ff3e10, 48 bytes, 0x00000103(STATUS_PENDING)
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::bulk_or_interrupt_transfer:req bdbee7f8 -> PipeHandle 41f9fc30, IN|SHORT_OK, TransferBufferLength 64 
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send:req bdbee7f8 -> 48{seqnum 39, devid 0x20001, in[2]}, cmd_submit: flags 0, length 64, start_frame 0, isoc[-1], interval 4
[0]0000 TRACE_LEVEL_VERBOSE `anonymous namespace'::send:req bdbee7f8 -> wsk irp 483e1790, 48 bytes, 0x00000103(STATUS_PENDING)
[... cut ...]
[0]0000 TRACE_LEVEL_VERBOSE FxRequest::_Create:Irp 0000000000000000 Ownership FxRequestOwnsIrp FxRequest FFFFD60342210530, status STATUS_SUCCESS
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFD6033FF6C040 is processing WDFQUEUE 0x000029FCB5551478
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchRequestToDriver:Calling driver EvtIoInternalDeviceControl for WDFREQUEST 0x000029FCBDDEFAC8
[0]0000 TRACE_LEVEL_WARNING FxObjectGetTypedContext:Attempting to get context type request_ctx from FxObject 0xFFFFD60342210530
---- end of log ----
1 error events encountered in logs

375583810a, 2023-12-19:
this is call task
BLACKBOXWINLOGON: 1

PROCESS_NAME:  System

EXCEPTION_STR:  0xc0000005

STACK_TEXT:  
fffff884`3a97f508 fffff802`2c5d1924     : 00000000`00000000 ffffd603`00000000 ffffd603`4451fdb8 fffff802`2c5edeb0 : usbip2_ude!InitializeListHead+0xf [C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\km\wdm.h @ 12589] 
fffff884`3a97f510 fffff802`2c5d379c     : ffffd603`451599a0 000029fc`bddefac8 000029fc`be8d9128 00000000`0000002e : usbip2_ude!`anonymous namespace'::add_egress_request+0x34 [E:\git\usbip\usbip-win2\drivers\ude\device_ioctl.cpp @ 140] 
fffff884`3a97f550 fffff802`2c5d1d03     : 000029fc`be8d9128 fffff884`3a97f7d8 ffffd603`451599a0 ffffd603`00000000 : usbip2_ude!`anonymous namespace'::send+0x1bc [E:\git\usbip\usbip-win2\drivers\ude\device_ioctl.cpp @ 179] 
fffff884`3a97f740 fffff802`2c5d456b     : ffffd603`451599a0 000029fc`be8d9128 ffffd603`41726f80 000029fc`bddefac8 : usbip2_ude!`anonymous namespace'::bulk_or_interrupt_transfer+0x363 [E:\git\usbip\usbip-win2\drivers\ude\device_ioctl.cpp @ 300] 
fffff884`3a97f860 fffff802`2c5d26c7     : ffffd603`451599a0 000029fc`be8d9128 ffffd603`41726f80 000029fc`bddefac8 : usbip2_ude!`anonymous namespace'::usb_submit_urb+0x1db [E:\git\usbip\usbip-win2\drivers\ude\device_ioctl.cpp @ 419] 
fffff884`3a97f910 fffff802`153b90fd     : 000029fc`b5551478 000029fc`bddefac8 00000000`42241160 00000000`00000000 : usbip2_ude!usbip::device::internal_control+0x157 [E:\git\usbip\usbip-win2\drivers\ude\device_ioctl.cpp @ 628] 
fffff884`3a97f9a0 fffff802`153b8960     : ffffd603`4aaaeb00 ffffd603`4aaaeb80 ffffd603`48e52800 fffff802`15443820 : Wdf01000!FxIoQueue::DispatchRequestToDriver+0x2cd [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3343] 
fffff884`3a97fa40 fffff802`153b5520     : ffffd603`4aaaeb80 ffffd603`4aaaeb00 00000000`00000000 ffffd603`48e52840 : Wdf01000!FxIoQueue::DispatchEvents+0x520 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3125] 
fffff884`3a97fb10 fffff802`15426cd4     : ffffd603`4aaaeb00 ffffd603`42210530 ffffd603`48e52840 fffff802`141b8074 : Wdf01000!FxIoQueue::QueueRequestFromForward+0x98 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 2496] 
fffff884`3a97fb70 fffff802`153fa3d8     : 000029fc`bdde0001 00000000`00000000 000029fc`b9441918 fffff802`13b4f311 : Wdf01000!FxIoQueue::QueueDriverCreatedRequest+0xd8 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 1895] 
fffff884`3a97fbc0 fffff802`157525a4     : ffffd603`4aaaeb80 fffff802`15751010 ffffd603`42241160 ffffd603`42210530 : Wdf01000!imp_WdfRequestForwardToParentDeviceIoQueue+0x148 [minkernel\wdf\framework\shared\core\fxrequestapi.cpp @ 3335] 
fffff884`3a97fc20 fffff802`153ba976     : ffffd603`46bbe6e0 00000001`00000008 00000000`00000000 00000000`00000000 : ucx01000!RootHub_Pdo_EvtInternalDeviceControlIrpPreprocessCallback+0x3d4
fffff884`3a97fcb0 fffff802`13a10665     : 00000000`00220003 fffff802`161deed8 ffffd603`4221d010 00000000`00000000 : Wdf01000!FxDevice::DispatchWithLock+0x266 [minkernel\wdf\framework\shared\core\fxdevice.cpp @ 1447] 
fffff884`3a97fd10 fffff802`161980a7     : 00000000`00000001 00000000`73556f49 ffffd603`486a06b0 00000000`00000000 : nt!IofCallDriver+0x55
fffff884`3a97fd50 fffff802`153ba976     : ffffd603`4221d010 ffffd603`4d28d8b0 ffffd603`4221d010 ffffd603`486a06b0 : UsbHub3!HUBPDO_EvtDeviceWdmIrpPreprocess+0x11d7
fffff884`3a97fe20 fffff802`13a10665     : 00000000`00000019 00000000`00000000 00000000`00000000 ffffd603`4d28d8c8 : Wdf01000!FxDevice::DispatchWithLock+0x266 [minkernel\wdf\framework\shared\core\fxdevice.cpp @ 1447] 
fffff884`3a97fe80 fffff802`2c641d43     : ffffd603`4d28d8c8 ffffd603`4d28d8c8 fffff884`3a980060 ffffd603`00000000 : nt!IofCallDriver+0x55
fffff884`3a97fec0 fffff802`13a10665     : 00000000`00000005 fffff884`3a97fff8 fffff884`3a980060 ffffd603`486a6df0 : usbip2_filter+0x1d43
fffff884`3a97ff20 fffff802`153b35a7     : 00000000`00000000 ffffd603`486a6df0 fffff884`3a980060 fffff884`3a97fff8 : nt!IofCallDriver+0x55
fffff884`3a97ff60 fffff802`153b1ca2     : ffffd603`486a6d01 ffffd603`4519f901 fffff884`3a980060 00000000`00000000 : Wdf01000!FxIoTarget::Send+0x17 [minkernel\wdf\framework\shared\inc\private\km\FxIoTargetKm.hpp @ 268] 
fffff884`3a97ff90 fffff802`2c684b6c     : ffffd603`3f8b2f00 ffffd603`4519f9e0 ffffd603`486a6df0 00000000`00000019 : Wdf01000!imp_WdfRequestSend+0x1d2 [minkernel\wdf\framework\shared\core\fxrequestapi.cpp @ 2090] 
fffff884`3a980010 fffff802`2c684c30     : ffffd603`3c6fe050 ffffd603`3c7020c0 000029fc`bae63a98 ffffd603`3f8b2de0 : dc1_controller!DccProcessSendRequest+0x280
fffff884`3a9800c0 fffff802`153b8f92     : fffff884`3a9806e0 ffffd603`4aaa3a20 ffffd603`3c70224d 00000000`00000000 : dc1_controller!DccEvtSendControl+0x70
fffff884`3a980100 fffff802`153b8960     : 00000000`00000002 ffffd603`48e52000 ffffd603`452cb960 fffff802`15443820 : Wdf01000!FxIoQueue::DispatchRequestToDriver+0x162 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3325] 
fffff884`3a9801a0 fffff802`153b5520     : ffffd603`48e52060 ffffd603`48e52000 00000000`00000000 ffffd603`432984c0 : Wdf01000!FxIoQueue::DispatchEvents+0x520 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 3125] 
fffff884`3a980270 fffff802`153bba5f     : ffffd603`4519c500 ffffd603`4860c020 ffffd603`4519c690 ffffd603`48e52060 : Wdf01000!FxIoQueue::QueueRequestFromForward+0x98 [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 2496] 
[... cut ...]
>12589:     ListHead->Flink = ListHead->Blink = ListHead;
 12590:     return;
 12591: }
 12592: 
 12593: _Must_inspect_result_
 12594: BOOLEAN

SYMBOL_NAME:  usbip2_ude!InitializeListHead+f

MODULE_NAME: usbip2_ude

IMAGE_NAME:  usbip2_ude.sys

STACK_COMMAND:  .cxr 0xfffff8843a97eb00 ; kb

BUCKET_ID_FUNC_OFFSET:  f

FAILURE_BUCKET_ID:  AV_usbip2_ude!InitializeListHead

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {d86b1194-0392-1b49-412e-d699e2992702}

vadimgrn, 2023-12-19:
Thank you for the log, but please do not attach screenshots, just copy the whole output in WinDbg. This bug definitely can be fixed, but currently I cannot understand what is wrong. There is absent the beginning of the output where the reason of the crash is dumped, but I can guess it is a memory violation.

vadimgrn, 2023-12-19:
Could you please copy ALL output in WinDbg command window, paste it to a file and attach it here?

375583810a, 2023-12-19:
this rebuild

Microsoft (R) Windows Debugger Version 10.0.22621.2428 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Users\Administrator\Desktop\MEMORY.DMP]
Kernel Bitmap Dump File: Kernel address space is available, User address space may not be available.

************* Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*D:\symbolslocal*https://msdl.microsoft.com/download/symbols
OK                                             E:\git\usbip\usbip-win2\x64\Debug
Symbol search path is: srv*D:\symbolslocal*https://msdl.microsoft.com/download/symbols;E:\git\usbip\usbip-win2\x64\Debug
Executable search path is: 
Windows 10 Kernel Version 19041 MP (2 procs) Free x64
Product: WinNt, suite: TerminalServer SingleUserTS
Edition build lab: 19041.1.amd64fre.vb_release.191206-1406
Machine Name:
Kernel base = 0xfffff805`61e00000 PsLoadedModuleList = 0xfffff805`62a2a770
Debug session time: Tue Dec 19 19:07:30.827 2023 (UTC + 8:00)
System Uptime: 0 days 8:00:42.583
Loading Kernel Symbols
...............................................................
................................................................
................................................................
.......................
Loading User Symbols

Loading unloaded module list
...................
For analysis of this file, run !analyze -v
1: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

SYSTEM_THREAD_EXCEPTION_NOT_HANDLED (7e)
This is a very common BugCheck.  Usually the exception address pinpoints
the driver/function that caused the problem.  Always note this address
as well as the link date of the driver/image that contains this address.
Arguments:
Arg1: ffffffffc0000005, The exception code that was not handled
Arg2: fffff8057775468f, The address that the exception occurred at
Arg3: ffff9d8e153cc2c8, Exception Record Address
Arg4: ffff9d8e153cbb00, Context Record Address

Debugging Details:
------------------

Unable to load image \SystemRoot\system32\DRIVERS\usbip2_filter.sys, Win32 error 0n2

KEY_VALUES_STRING: 1

    Key  : AV.Dereference
    Value: NullClassPtr

    Key  : AV.Fault
    Value: Write

    Key  : Analysis.CPU.mSec
    Value: 5171

    Key  : Analysis.DebugAnalysisManager
    Value: Create

    Key  : Analysis.Elapsed.mSec
    Value: 5776

[... 307 lines cut ...]
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::QueueRequest:Queuing WDFREQUEST 0x00003A77EB2132F8 on WDFQUEUE 0x00003A77EB354478
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC588002C8040 is processing WDFQUEUE 0x00003A77EB354478
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchRequestToDriver:Calling driver EvtIoInternalDeviceControl for WDFREQUEST 0x00003A77EB2132F8
[0]0000 TRACE_LEVEL_VERBOSE FxPkgPnp::Dispatch:WDFDEVICE 0x00003A77F98ABF68 !devobj 0xFFFFC587FBD84A70, IRP_MJ_PNP, 0x00000009(IRP_MN_QUERY_CAPABILITIES) IRP 0xFFFFC58803D26270
[0]0000 TRACE_LEVEL_VERBOSE FxPkgPnp::Dispatch:WDFDEVICE 0x00003A77EB5C2558 !devobj 0xFFFFC5880DA312C0, IRP_MJ_PNP, 0x00000009(IRP_MN_QUERY_CAPABILITIES) IRP 0xFFFFC588051CF3F0
[0]0000 TRACE_LEVEL_VERBOSE FxPkgFdo::HandleQueryCapabilities:Entering QueryCapabilities handler
[0]0000 TRACE_LEVEL_VERBOSE FxPkgFdo::HandleQueryCapabilities:Exiting QueryCapabilities handler
[0]0000 TRACE_LEVEL_VERBOSE FxPkgFdo::HandleQueryCapabilitiesCompletion:Entering QueryCapabilities completion handler
[0]0000 TRACE_LEVEL_VERBOSE FxPkgFdo::HandleQueryCapabilitiesCompletion:Exiting QueryCapabilities completion handler
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC587FC503040 is processing WDFQUEUE 0x00003A77FF365F68
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC587FC503040 is processing WDFQUEUE 0x00003A77EB354478
[0]0000 TRACE_LEVEL_VERBOSE FxRequest::Complete:Completing WDFREQUEST 0x00003A77F9C19F68 for IRP 0xFFFFC5880D3F3A20 with Information 0x0, STATUS_SUCCESS
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC587FC503040 is processing WDFQUEUE 0x00003A77FF365F68
[0]0000 TRACE_LEVEL_VERBOSE Vf_VerifyWdfDeviceWdmDispatchIrpToIoQueue:WDFDEVICE 0x00003A77F98ABF68 !devobj 0xFFFFC587FBD84A70 0x0000000f(IRP_MJ_INTERNAL_DEVICE_CONTROL), IRP_MN 0, IRP 0xFFFFC5880D3F3A20
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::QueueRequest:Queuing WDFREQUEST 0x00003A77F9C19F68 on WDFQUEUE 0x00003A77EB354478
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC587FC503040 is processing WDFQUEUE 0x00003A77EB354478
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchRequestToDriver:Calling driver EvtIoInternalDeviceControl for WDFREQUEST 0x00003A77F9C19F68
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC587FC503040 is processing WDFQUEUE 0x00003A77FF365F68
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC587FC503040 is processing WDFQUEUE 0x00003A77EB354478
[0]0000 TRACE_LEVEL_VERBOSE FxRequest::_Create:Irp 0000000000000000 Ownership FxRequestOwnsIrp FxRequest FFFFC588111E6DA0, status STATUS_SUCCESS
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchEvents:Thread FFFFC5881117F040 is processing WDFQUEUE 0x00003A77F49B6438
[0]0000 TRACE_LEVEL_VERBOSE FxIoQueue::DispatchRequestToDriver:Calling driver EvtIoInternalDeviceControl for WDFREQUEST 0x00003A77EEE19258
[0]0000 TRACE_LEVEL_WARNING FxObjectGetTypedContext:Attempting to get context type request_ctx from FxObject 0xFFFFC588111E6DA0
---- end of log ----
1 error events encountered in logs

vadimgrn, 2023-12-26:
I still need a full log. Start logging and AFTER THAT attach to the device replicate a BSOD. Do not add pictures.

vadimgrn, 2023-12-26:
Could you please try this build?
[USBip-0.9.5.8-Release.001.zip](https://github.com/vadimgrn/usbip-win2/files/13770906/USBip-0.9.5.8-Release.001.zip)
[USBip-0.9.5.8-Release.002.zip](https://github.com/vadimgrn/usbip-win2/files/13770910/USBip-0.9.5.8-Release.002.zip)
P.S.Rename files to USBip-0.9.5.8-Release.zip.001, USBip-0.9.5.8-Release.zip.002 and use 7-zip to extract the installer.