vadimgrn/usbip-win2 #54: some hid devices do not work immediately after attaching

comsyspro, 2023-12-27:
i hope getting more infos here also concerning the linux usbip server implementation. i have similar problems with some hid devices as described here in usbipd-win project, where they fixed the issue with a timeout:
https://github.com/dorssel/usbipd-win/issues/659#issuecomment-1677977801

after attaching the devices, they are shown in linux os in a connected state but will not work immediately. only after detaching and reattaching 1 to 3 times then the usb devices come to work correctly. so could this be a similar driver issue in linux server usbip or timeout issue? i would like to also mention that these kind of hid devices mostly have 2 bNumInterfaces and this differs from other devices which normally have only 1 bNumInterfaces and perhaps this could also be the reason. any advice or ideas?

vadimgrn, 2023-12-28:
You can do almost nothing with linux usbip implementation. Someone should fix this issue and make a pull request. If maintainer of this subsystem approve it, the poll request will be merged into the kernel.

comsyspro, 2023-12-28:
i also asked this at the developer of usbipd-win project and you can also find some further usbmon captures of the hid device;
https://github.com/dorssel/usbipd-win/issues/816#issuecomment-1870343829

would it be possible to develop an own usbip server like it exists in the origin project https://github.com/cezanne/usbip-win?
why have you decided to remove the server part of usbip in your code?

vadimgrn, 2023-12-28:
https://github.com/torvalds/linux/tree/master/drivers/usb/usbip

closed 2024-01-29 by comsyspro