vadimgrn/usbip-win2 #55: Mac host -> Windows client + libusb claim_interface = reliable BSoD

brandonros, 2024-01-14:
[bsod-logs.txt](https://github.com/vadimgrn/usbip-win2/files/13933091/bsod-logs.txt)

I'm using this as a host: https://github.com/jiegec/usbip on Mac (arm64): `$ RUST_LOG=debug cargo run --example host`

The device I'm working with is super weird. It is exposing 4 interfaces:

```
interface_number = 0 transfer_type = Interrupt direction = In
interface_number = 1 transfer_type = Bulk direction = Out
interface_number = 1 transfer_type = Bulk direction = In
interface_number = 2 transfer_type = Interrupt direction = In
interface_number = 3 transfer_type = Bulk direction = Out
interface_number = 3 transfer_type = Bulk direction = In
```

```rust
// get device
let mut device_handle = rusb::open_device_with_vid_pid(vendor_id, product_id).unwrap();
let device = device_handle.device();
// configuration
let config_desc = device.config_descriptor(config_index).unwrap();
device_handle.set_active_configuration(config_desc.number()).unwrap();
device_handle.claim_interface(interface_number).unwrap();
device_handle.set_alternate_setting(interface_number, 0).unwrap();
```

That `device_handle.claim_interface(interface_number).unwrap();` line makes it blow up.

Let me know if the logs are right or if what I am doing just isn't recommended (I can maybe try Windows host -> Windows client if you think it will help)

vadimgrn, 2024-01-15:
Hi, the log must contain symbolic names instead of something like 'usbip2_ude+0x1fa6', I can't guess what it was.

vadimgrn, 2024-01-15:
Use release build,just follow instructions verbatim.

brandonros, 2024-01-20:
The device is attached:

```
PS C:\Users\star> usbip list -r redacted
Exportable USB devices
======================
   2-6-5   : Realtek Semiconductor Corp. : unknown product (0bda:1100)
           : /sys/bus/2/6/5
           : (Defined at Interface level) (00/00/00)
           :  0 - (Defined at Interface level) (00/00/00)

   2-5-1   : Realtek Semiconductor Corp. : unknown product (0bda:5483)
           : /sys/bus/2/5/1
           : (Defined at Interface level) (00/00/00)
           :  0 - (Defined at Interface level) (00/00/00)

   2-4-2   : Realtek Semiconductor Corp. : RTL8153 Gigabit Ethernet Adapter (0bda:8153)
           : /sys/bus/2/4/2
           : (Defined at Interface level) (00/00/00)
           :  0 - (Defined at Interface level) (00/00/00)
           :  1 - (Defined at Interface level) (00/00/00)

   2-3-1   : Realtek Semiconductor Corp. : unknown product (0bda:0483)
           : /sys/bus/2/3/1
           : (Defined at Interface level) (00/00/00)
           :  0 - (Defined at Interface level) (00/00/00)

   2-2-3   : Realtek Semiconductor Corp. : unknown product (0bda:0483)
           : /sys/bus/2/2/3
           : (Defined at Interface level) (00/00/00)
           :  0 - (Defined at Interface level) (00/00/00)

   2-1-2   : Realtek Semiconductor Corp. : unknown product (0bda:5483)
           : /sys/bus/2/1/2
           : (Defined at Interface level) (00/00/00)
           :  0 - (Defined at Interface level) (00/00/00)

   0-1-1   : unknown vendor : unknown product (18e1:01b3)
           : /sys/bus/0/1/1
           : Miscellaneous Device/?/Interface Association (ef/02/01)
           :  0 - Miscellaneous Device/?/? (ef/04/01)
           :  1 - (Defined at Interface level) (00/00/00)
           :  2 - (Defined at Interface level) (00/00/00)
           :  3 - (Defined at Interface level) (00/00/00)

PS C:\Users\star> usbip port
Imported USB devices
====================
Port 01: device in use at High Speed(480Mbps)
         unknown vendor : unknown product (18e1:01b3)
           -> usbip://redacted:3240/0-1-1
           -> remote bus/dev 000/001
```

Zadig shows the device:

[image]

[image]

device shows up, but when i call setupapi

```
[   2024-01-20T14:59:08.016449-05:00] trace_hooks::setupapi INFO: SetupDiGetClassDevsW ClassGuid = 0x6c4e0b5c (Data1 = f99f6fce Data2 = 03f4 Data3 = 4817 Data4 = [8a, 03, 8c, fd, e4, 74, 3b, 92]) Enumerator = 0x0 hwndParent = 0x0 Flags = 18
[2024-01-20T14:59:08.023907700-05:00] trace_hooks::setupapi INFO: SetupDiGetClassDevsW ClassGuid = 0x6c4e0b5c (Data1 = f99f6fce Data2 = 03f4 Data3 = 4817 Data4 = [8a, 03, 8c, fd, e4, 74, 3b, 92]) Enumerator = 0x0 hwndParent = 0x0 Flags = 18 ret_val = 0x1429ea8 last_error = 0
[2024-01-20T14:59:08.028475800-05:00] trace_hooks::setupapi INFO: SetupDiEnumDeviceInterfaces DeviceInfoSet = 0x1429ea8 DeviceInfoData = 0x0 InterfaceClassGuid = 0x6c4e0b5c (Data1 = f99f6fce Data2 = 03f4 Data3 = 4817 Data4 = [8a, 03, 8c, fd, e4, 74, 3b, 92]) MemberIndex = 0 DeviceInterfaceData = 0xddcef8
[2024-01-20T14:59:08.036703800-05:00] trace_hooks::setupapi INFO: SetupDiEnumDeviceInterfaces DeviceInfoSet = 0x1429ea8 DeviceInfoData = 0x0 InterfaceClassGuid = 0x6c4e0b5c (Data1 = f99f6fce Data2 = 03f4 Data3 = 4817 Data4 = [8a, 03, 8c, fd, e4, 74, 3b, 92]) MemberIndex = 0 DeviceInterfaceData = 0xddcef8 ret_val = 0 last_error = 259
[2024-01-20T14:59:08.043235400-05:00] trace_hooks::setupapi INFO: SetupDiDestroyDeviceInfoList
[2024-01-20T14:59:08.045522400-05:00] trace_hooks::setupapi INFO: SetupDiDestroyDeviceInfoList
```

it isn't showing?

brandonros, 2024-01-20:
[image]

wonder if this is an issue/how to fix?

brandonros, 2024-01-20:
[image]

are you sure release is ok? no Trace entires in usbip.pdb, etc.

vadimgrn, 2024-01-21:
I am sure, WPP tracing is relevant for driver only.

brandonros, 2024-01-27:
[bsod-logs-2024-01-27.txt](https://github.com/vadimgrn/usbip-win2/files/14073889/bsod-logs-2024-01-27.txt)

Is this any better?

Step to reproduce:

```
rem change to your WDK version
set PATH=%PATH%;C:\Program Files (x86)\Windows Kits\10\bin\10.0.22621.0\x64

set NAME=usbip

tracelog.exe -stop %NAME%-flt
tracelog.exe -stop %NAME%-ude

del /F %NAME%-*.*
tracepdb.exe -f "C:\Program Files\USBip\*.pdb" -s -p %TEMP%\%NAME%

tracelog.exe -start %NAME%-flt -guid #90c336ed-69fb-43d6-b800-1552d72d200b -f %NAME%-flt.etl -flag 0x3 -level 5
tracelog.exe -start %NAME%-ude -guid #ed18c9c5-8322-48ae-bf78-d01d898a1562 -f %NAME%-ude.etl -flag 0xF -level 5

# do the thing that makes the system BSOD, it auto restarts
# upon restart, do not run any extra tracelog commands to stop logs/etc., just run C:\Program Files (x86)\Windows Kits\10\Debuggers\x64\windbg.exe as Admin, open C:\Windows\MEMORY.DMP that BugCheck wrote before BSOD reboot, then run the following commands in WinDbg command line:

!analyze -v
!wdfkd.wdfsearchpath %TEMP%\USBip
!wdfkd.wdfsettraceprefix [%9]%3!04x! %!LEVEL! %!FUNC!:
!wdfkd.wdflogdump usbip2_ude -d
!wdfkd.wdflogdump usbip2_ude -f

# then, copy and paste entire WinDbg.exe output as logs and attach
```

If this is wrong process please let me know, or if what I am trying is "impossible/broken by design"

brandonros, 2024-01-27:
Even after all this when I run `!analyze -v` again, there are still no debug symbols...?

vadimgrn, 2024-01-28:
I don't know what is wrong.I tested release builds, names are present in the log.

vadimgrn, 2024-01-28:
Try this sequence, pls

!wdfkd.wdfsearchpath %TEMP%\usbip
analyze -v

brandonros, 2024-01-28:
```
1: kd> !wmitrace.searchpath +%TEMP%\usbip
Trace Format search path is: 'C:\Program Files (x86)\Windows Kits\10\Debuggers;%TEMP%\usbip'
1: kd> !wdfkd.wdfsettraceprefix [%9]%3!04x! %!LEVEL! %!FUNC!:
SetTracePrefix: "[%9]%3!04x! %!LEVEL! %!FUNC!:"
1: kd> !wdfkd.wdflogdump usbip2_ude -d
Unable to load image \SystemRoot\System32\DriverStore\FileRepository\usbip2_ude.inf_amd64_86c507ba2734a391\usbip2_ude.sys, Win32 error 0n2
```

Is this expected/the root of our pain?

brandonros, 2024-01-28:
I got it, needed to do `.sympath+ C:\Program Files\USBip`

brandonros, 2024-01-28:
https://github.com/vadimgrn/usbip-win2/blob/master/drivers/ude/device_ioctl.cpp#L176

Which variable do we think is causing 0xC0000005 there

vadimgrn, 2024-01-28:
Full windbg output is required

brandonros, 2024-01-28:
[bsod-logs-2024-01-28.txt](https://github.com/vadimgrn/usbip-win2/files/14077257/bsod-logs-2024-01-28.txt)

brandonros, 2024-01-28:
[bsod-logs-2024-01-28.txt](https://github.com/vadimgrn/usbip-win2/files/14077260/bsod-logs-2024-01-28.txt)

brandonros, 2024-01-28:
I think you fixed it in https://github.com/vadimgrn/usbip-win2/releases/tag/v.0.9.5.8

closed 2024-01-28 by brandonros

vadimgrn, 2024-01-28:
Yes, it seems the same issue. Have you tried the latest release?

brandonros, 2024-01-28:
Yes, working good on this release, thank you.

[image]

What would you recommend to trace this down why this is happening? `tracelog`?

brandonros, 2024-01-28:
[usbip-ude.txt](https://github.com/vadimgrn/usbip-win2/files/14077372/usbip-ude.txt)
[usbip-flt.txt](https://github.com/vadimgrn/usbip-win2/files/14077373/usbip-flt.txt)

vadimgrn, 2024-01-28:
It could be unsupported Windows version. Try it on Win11 if you can, please.

brandonros, 2024-01-28:
I don't think this is it.

If I do Windows host (https://github.com/dorssel/usbipd-win/releases) to your Windows client (usbip-win2) it is working good.

It is when I try to go Android host or Linux host or Mac OS host (which is using libusb and not "Windows driver implementation") that I run into this issue.

It very well could be "unsupported" but I was looking to figure out why it is unsupported/what is unsupported so I can work around it and fix... :)

brandonros, 2024-01-28:
Are you able to help me determine in which layer/stack the issue is? Let me know how I can reach you, I believe we may be able to help pay you for your time/support.

vadimgrn, 2024-01-28:
I have seen this bug (invalid parameter passed to service or function) when I debugged what is the oldest supported Win version for the driver.

brandonros, 2024-01-28:
You know what, you are right. They are compiling against Windows SDK 8.1 instead of Windows 10 in the driver.

brandonros, 2024-01-28:
Let me ask you this, tell me if I should make a separate issue for this (it might not be in your layer)

```rust
pub fn write_slp_command(&self, slp_command: &SlpCommand) {
        let timeout = Duration::from_secs(0); // android host -> usbip-win2 does not work, blocks indefinitely, if i change it to 1 sec it will unblock
        let message = slp::serialize_slp_command(slp_command);
        println!("out: {message:02x?}");
        self.write_device_handle
            .write_bulk(self.out_endpoint_address, &message, timeout)
            .unwrap();
    }
```

vadimgrn, 2024-01-28:
I don't know what it is, but usbip driver must not block indefinitely anyway.