vadimgrn/usbip-win2 #60: how can i get lcd display work of 3dconnexion spacemouse enterprise?

comsyspro, 2024-01-29:
on this device the usbip is working for all buttons and 6 dof controller but not for the lcd display. also only the wdm driver is stable and when using the ude driver after some seconds there is a bluescreen, so i only use wdm here. but both on wdm and ude driver the lcd display don't show the icons which normally shows an icon and text explaining the button functions on the top. i also tried this with some commercial usbip software products and there the display is working. so what can i do to get the lcd display work?

**here the lcd display is showing the icons which are not shown when connecting with usbip-win2**
[image]

vadimgrn, 2024-01-29:
https://github.com/vadimgrn/usbip-win2#debugging-bsod
Collect BSOD logs for UDE driver, please.

comsyspro, 2024-01-29:
here is the result of windbg.exe. can you do something with this?

[windbg memorydump.txt](https://github.com/vadimgrn/usbip-win2/files/14084352/windbg.memorydump.txt)

vadimgrn, 2024-01-29:
There no names in the backtrace. Try to execute this before !analyze -v
.sympath +C:\Program Files\USBip
!wmitrace.searchpath +%TEMP%\usbip

comsyspro, 2024-01-29:
.sympath +C:\Program Files\USBip gives an error:
```
5: kd> .sympath +C:\Program Files\USBip
Symbol search path is: +C:\Program Files\USBip
Expanded Symbol search path is: +c:\program files\usbip
Error: Execute .sympath(+) command attempts to access '+C:\Program Files\USBip' failed: 0x7b - The filename, directory name, or volume label syntax is incorrect.

************* Path validation summary **************
Response                         Time (ms)     Location
Error                                          +C:\Program Files\USBip
```

here is the new dump, but i think it's the same as before, i followed the steps described here https://github.com/vadimgrn/usbip-win2#debugging-bsod

[windbg memorydump.txt](https://github.com/vadimgrn/usbip-win2/files/14086232/windbg.memorydump.txt)

vadimgrn, 2024-01-29:
It has symbols, I can work with it.  I will look at it on weekends.

vadimgrn, 2024-01-29:
What release are you using? This bug was fixed in the latest release.

comsyspro, 2024-01-29:
i used the git master branch about 2 or 3 weeks ago and compiled it on win10 and visual studio 2022 and with that i got 0.9.5.8 version. at that point in time you hadn't released v.0.9.5.8 yet. i haven't tried it with your latest release from https://github.com/vadimgrn/usbip-win2/releases/tag/v.0.9.5.8 - so i will do it now.

comsyspro, 2024-01-29:
i have good news and am happy! with the newest release ude works with this device now and also the display is showing up now! thank you a lot!

vadimgrn, 2024-01-29:
Thank you for your feedback! It was small fix. This bug should not happen because WDF ensures that the driver receives WDFREQUEST objects with preallocated context. But the reality is that for some devices the received WDFREQUEST doesn't have a context. This is weird, I don't know why it happens, but the workaround was implemented after several bug reports.

vadimgrn, 2024-01-29:
https://github.com/vadimgrn/usbip-win2/commit/32250e8b6161e303010ca4b59e716010572c979e

comsyspro, 2024-01-29:
yes you can. and also this issue is resolved now with graphic tablet: https://github.com/vadimgrn/usbip-win2/issues/54
i have the impression that now more and more usb devices are working properly. before i have often used a mix of wdm and ude drivers and fixed some devices with special code snippets. it seems that your project gets close to the production ready state.

vadimgrn, 2024-01-29:
I have very limited set if devices for testing. Because of that I have to rely on bug reports that have windbg output.

comsyspro, 2024-01-29:
today i learned how to create debug with windbg and i can inform you when i got other usb devices to test. but as i said most of the devices are working now.