vadimgrn/usbip-win2 #61: Intermittent freezing when using Thrustmaster TMX Steering Wheel

ActualTomato, 2024-02-03:
Hi, I've been using this recently to play racing games on my remote computer using my Thrustmaster TMX Racing Wheel, and it's been working great when it doesn't freeze. However, at random times while im playing, probably every minute or two, the controller gets stuck in whatever position and stops responding on the client computer for anywhere from 10 seconds to a minute. I know this isn't a network problem, as I'm streaming video from the client computer at the same time, and that doesn't freeze at any point. Any ideas?

vadimgrn, 2024-02-03:
Hi, th driver's log is required.

ActualTomato, 2024-02-03:
[usbip-flt.txt](https://github.com/vadimgrn/usbip-win2/files/14152684/usbip-flt.txt)
[usbip-ude.txt](https://github.com/vadimgrn/usbip-win2/files/14152685/usbip-ude.txt)

Here, I started the log right before I connected the device, and ended it a few seconds after a big freeze during a game. There were about 4 big freezes throughout this log I think.

vadimgrn, 2024-02-04:
Thank you, I'll take a look at it. But I don't have enough spare time to do this quickly.

ActualTomato, 2024-02-04:
No worries. If you need any other details, let me know.