vadimgrn/usbip-win2 #62: This device cannot start. (code 10)

brandonros, 2024-02-04:
You and I spoke a bit about this in a previous issue that I opened with an unrelated topic.

I am not 100% sure whether the issue is in:

the implementation in this code (client)

the implementation in the code of the driver (that I sort of kind of don't really have/don't have logs into)

the implementation in the code of the host (https://github.com/dorssel/usbipd-win)

Would you be willing to help me understand how to get as many trace level logs as possible to understand what is going wrong? I'm trying to forward what is basically a simple USB CDC serial device (I tested that if I replace the driver with the usbser driver and forward traffic around as COM7 port, it works great)

I could just use your help/guidance in understanding where to look in driver level trace logs to see what the issue is/understanding how this driver (usbip-win2) interfaces with the driver on the device it is virtualizing. Best case scenario: we'd find some logs that reveal in issue in this codebase that can be fixed.

I'll start posting stuff below.

brandonros, 2024-02-04:
One complex thing about this device is it has multiple interfaces. aka, you plug one device in physically and you get

1. https://en.wikipedia.org/wiki/Network_Driver_Interface_Specification (which I don't care about/am not trying to forward)
2. what is really just usbser/CDC device (doesn't work well with WinUSB at all for some reason; unable to claim interface or read/write from the endpoints) that has its own driver

I would say this is atypical and could potentially be root of the issue.

vadimgrn, 2024-02-04:
https://github.com/vadimgrn/usbip-win2?tab=readme-ov-file#obtaining-usbip-logs-on-windows
You will get the most verbose log.

brandonros, 2024-02-04:
Host:

[image]

[image]

[image]

Client:

[image]

brandonros, 2024-02-04:
[usbip-flt.txt](https://github.com/vadimgrn/usbip-win2/files/14157911/usbip-flt.txt)
[usbip-ude.txt](https://github.com/vadimgrn/usbip-win2/files/14157912/usbip-ude.txt)

Do you need the .etl files too?

vadimgrn, 2024-02-04:
**04x! !FUNC!**:vhci aa944698, fobj aa68cda8
TRACE_FORMAT_PREFIX was set not correctly.

brandonros, 2024-02-04:
Updated. It was the `%%` .cmd problem.