vadimgrn/usbip-win2 #69: Hangs when attaching and then does not respond after cancelling

ramiabughazaleh, 2024-04-21:
Hi.

Thank you for usbip-win2.

The desktop application, `wusbip.exe`, hangs when trying to attach a device and then stops responding after trying to cancel.

Steps to reproduce on Windows Server 2022 Standard:
1. Run `bcdedit.exe /set testsigning on` and restart
2. Install USBIP server: https://github.com/dorssel/usbipd-win
3. Run `usbipd list`
4. Run `usbipd bind --busid=3-1`
5. Install usbip-win2 UDE 0.9.6.1 and restart
6. Run USBip shortcut from the desktop
7. Enter the local machine's name in the `Server` field and click `Add devices`
8. Select the device that was shared in step 4 and click `Attach`
9. USBip hung when attaching the device, and my only option is to cancel
10. Click the `Cancel` button
11. USBip stops responding

[image]

I was able to perform the same steps using usbip-win2 UDE 0.9.6.0 without issues.

vadimgrn, 2024-04-21:
Hi, please check all log options and gather driver log too.

ramiabughazaleh, 2024-04-21:
Hi.

Trying to attach the device on the command line also hangs:
```
C:\Program Files\USBip>usbip list -r WIN-RDT9KT7VD3O
Exportable USB devices
======================
    4-1    : Aladdin Knowledge Systems : Token JC (0529:0620)
           : USB\VID_0529&PID_0620\7&25AA2865&0&1
           : (Defined at Interface level) (00/00/00)
           :  0 - (Defined at Interface level) (00/00/00)

C:\Program Files\USBip>usbip attach -r WIN-RDT9KT7VD3O --bus-id 4-1
```

`usbip list -s` does not produce any output.

The screenshot of the Log window is a result of selecting `Verbose` and `Library` in the `Log` menu.

On a side note, check-marking `Library` and then clicking on `Add devices` causes wusbip.exe to crash:
```
Faulting application name: wusbip.exe, version: 0.9.6.1, time stamp: 0x66237e96
Faulting module name: ucrtbase.dll, version: 10.0.20348.1, time stamp: 0xdacb234c
Exception code: 0xc0000409
Fault offset: 0x000000000007b3a8
Faulting process id: 0x68c
Faulting application start time: 0x01da93c8425722b4
Faulting application path: C:\Program Files\USBip\wusbip.exe
Faulting module path: C:\Windows\System32\ucrtbase.dll
Report Id: cf197c36-656c-4365-8dcd-6035a8aba08f
Faulting package full name: 
Faulting package-relative application ID: 
```

I started logging, reproduced the hang issue, and here are the plain text USB/IP logs:
[usbip-flt.txt](https://github.com/vadimgrn/usbip-win2/files/15051226/usbip-flt.txt)
[usbip-ude.txt](https://github.com/vadimgrn/usbip-win2/files/15051227/usbip-ude.txt)

vadimgrn, 2024-04-21:
I suspect that issue is in the driver, it doesn't complete IRP under some conditions.

ramiabughazaleh, 2024-04-23:
It's an English operating system.

After an uninstall, reboot, and install again, I still ran into the same issue.

vadimgrn, 2024-04-23:
I can't replicate that. You can use WinDbg to get backtrace of the app.

vadimgrn, 2024-04-23:
Please try again, the release has been updated.

closed 2024-05-02 by vadimgrn

mrtumnus, 2024-09-20:
Ah, I think I figured it out.  I wasn't running usbip as an administrator.

I think it still would be nice to either force the program to run as an admin, or at least quit if permissions aren't there.  As it is, the program locks up entirely and the system has to be rebooted.

vadimgrn, 2024-09-21:
Administrator permissions are not required to use usbip.exe or GUI. The driver's log is required to find out what is going on.