vadimgrn/usbip-win2 #71: Disabling "Windows Test Signing Mode" once the program is installed will cause every USB device to stop working

rogermiranda1000, 2024-05-10:
I had to access with remote controll because I couldn't move my mouse/keyboard (directly connected to the computer)...

### Steps to reproduce

- Follow the install setup
- Run `bcdedit.exe /set testsigning off`
- Restart the PC

rogermiranda1000, 2024-05-10:
Solved after uninstalling the program; I assume re-enabling testsigning will also work.

vadimgrn, 2024-05-10:
Obviously it cannot work if test signing is disabled. You can disable it for some reasons without uninstalling usbip. It will work if disable test signing again.

rogermiranda1000, 2024-05-10:
> it cannot work if test signing is disabled
Special emphasis on the "directly connected to the computer". By disabling that, it almost caused my computer to brick. None of the (directed) connected USBs were recognised, so I couldn't login.

I think this part is worth mentioning, at least...

vadimgrn, 2024-05-11:
Disable Windows Test Signing Mode without removing USB/IP

usbip2_filter driver must be disabled, otherwise all USB controller

What is wrong? Everything is described.

vadimgrn, 2024-05-11:
I updated README to emphasize do not set testsigning off when USBip is installed.

closed 2024-05-11 by vadimgrn