vadimgrn/usbip-win2 #72: Device (T16000M) disconnects after inactivity

KnedlikMCPE, 2024-05-10:
So I have a remote setup for connecting to my flight sim rig - a T.16000M and TCA A320 throttle from thrustmaster. The issue? Well, the T.16000M seems to disconnect every so often. Seems related to activity in using the device, but I'm not sure. How can I troubleshoot? The server is a Linux Fedora installation, while the client is your usual Win10.

vadimgrn, 2024-05-11:
https://github.com/vadimgrn/usbip-win2/tree/develop#obtaining-usbip-logs-on-windows

KnedlikMCPE, 2024-05-11:
After further testing, it seems that the problem is not in the USB/IP client, but in the joystick being disconnected due to inactivity. Is there a way to make the client keep showing the device as connected to windows, kinda as a dummy USB device?

vadimgrn, 2024-05-11:
I think it is not depend on the usbip driver. Please look into joystick driver or software settings.

vadimgrn, 2024-05-11:
https://www.reddit.com/r/hotas/comments/5ag24m/dual_t1600m_stops_working_when_idle/

KnedlikMCPE, 2024-05-11:
I tried disabling autosuspend both on host (Linux) and on Windows, but no success. I'm kinda worried it's baked into the firmware.