vadimgrn/usbip-win2 #74: USB device disconnect and shutdown

jrogala, 2024-05-31:
I try to make a [VRS Wheel Base](https://vrs.racing/wheel-base) work via usbip. I have my host (ubuntu, fresh install) and client (windows, VRS driver installed). I can declare and bind my usb wheel base but whenever I attach to it, I lose the connection to it and I have to do a hard reboot of the wheel base (pluging/unpluging does not work).

Of course if I plug directly I have none of this and everything works fine.

I believe it may have to do with the firmware of the controller itself, but I don't have any experience on this low level stuff so I cannot pinpoint where is the problem. Isn't it supposed to be passthrough? 

Here are the log: 
- [usbip-ude.txt](https://github.com/user-attachments/files/15516583/usbip-ude.txt)
- [usbip-flt.txt](https://github.com/user-attachments/files/15516649/usbip-flt.txt)

vadimgrn, 2024-06-01:
I see that the couple of IRPs were not completed during the detaching, that causes this issue. I'll give you a build to test if I make a fix.

vadimgrn, 2024-06-01:
I've found a deadlock in the detach code, will give the test build tomorrow.

vadimgrn, 2024-06-02:
Please try this build https://drive.google.com/file/d/1CLo_iCieqlxLeUukgpsXGiEdGEULO8YJ/view?usp=sharing

jrogala, 2024-06-02:
Hello, thanks for your quick answer and build, it still does it with the newest build. 

Here are the new log: 
[usbip-ude.txt](https://github.com/user-attachments/files/15525539/usbip-ude.txt)
[usbip-flt.txt](https://github.com/user-attachments/files/15525540/usbip-flt.txt)

Thanks for your time!

vadimgrn, 2024-06-03:
I did not expect that this will fix the issue, I just fixed the deadlock during the detach. I think usbip server is closing the connection for some reason. To make sure you could set `-flag 0x1F` instead of 0xF in command 
`tracelog.exe -start %NAME%-ude -guid #ed18c9c5-8322-48ae-bf78-d01d898a1562 -f %NAME%-ude.etl -flag 0xF -level 5`