vadimgrn/usbip-win2 #75: USB device suddenly works. But why?

comsyspro, 2024-06-04:
Here is a USB device that has about 7 USB devices in it. This USB device was causing a few problems (with version v.0.9.6.3) and I would like to ask if there have been any changes to the code recently. I compiled usbip manually with the current code (v.0.9.6.4) and the device now works. I also compiled the current code on the server side and it could also be due to the server code that this USB device suddenly works. Can you say whether you have made any changes to the code recently to better support such USB devices?

[image]

vadimgrn, 2024-06-05:
Use git to find the difference between  releases. I can only guess what exactly affected this.

comsyspro, 2024-06-05:
In this case I also have to guess. I can tell you that with version v.0.9.6.3 attaching this USB device is firstly detected and displayed in the device manager. However after detaching and unbinding it was still there in the device manager and instead of removing it from the device manager another unkown device showed up with an error mark. Also in the registry the keys in the USB enum devices persist. When you attach a second time then you will get displayed 2 of these devices in the device manager and after detaching you also get 2 unkown devices showed up with an error mark - and so on ... Because of that I wrote a powershell script which manually deleted the affected registry keys of these USB enum devices to "clear" the status as it should be after detaching. But this was only a fix and didn't go down to the main reason of this issue. So it seems that you changed something in your code since v.0.9.6.3 which also affects this issue.

comsyspro, 2024-06-05:
Thank you. Perhaps this issue has something to do with this here https://github.com/vadimgrn/usbip-win2/commit/d1155fffc7bb386a5139eadeccd93ec09d4e88b3. Also the title "Fix detach deadlock" could suggest this.

vadimgrn, 2024-06-06:
There are no such loops in the driver

comsyspro, 2024-06-06:
Now I have noticed by further investigation that the device stays connecte as long as you use the device (moving sliders, rotating controls, pressing buttons). Then when you do nothing on the device it takes about 1 second and the device diconnects and reconnects. So it seems that if you don't give input signals to the device it disconnects and reconnects. Do you think that probably here in this fix https://github.com/vadimgrn/usbip-win2/commit/d1155fffc7bb386a5139eadeccd93ec09d4e88b3 is another little bug?

vadimgrn, 2024-06-06:
https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/selective-suspend-in-usb-drivers-wdf

vadimgrn, 2024-06-06:
https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/power-management-for-i-o-queues

comsyspro, 2024-06-06:
Thank you. Can you give a hint where these functions should be included. At the moment I have no overview and don't know how to start here. I just tested this device with another USBIP software and there the device is working normal.

Here is also a video. The device reboots when you stop giving input signals. You can see it here if you stop rotating the controller the device reboots. As long as you rotate or do smething the device stays on. The device should stay stable without rebooting.

https://github.com/vadimgrn/usbip-win2/assets/13809377/5f5a90ed-2ed6-4dfd-969d-6302dd590c1d

vadimgrn, 2024-06-06:
I don't know, I have no such devices. You don't need to include something, you could try to modify existing code. For example, make all WDFQUEUEs not power managed

comsyspro, 2024-06-07:
I tried now to set all entries from `cfg.PowerManaged = WdfFalse;` to `cfg.PowerManaged = WdfTrue;` and recompiled  but this is not the solution. So what could I do to find the reason. Can I create logs which you could look into?

vadimgrn, 2024-06-07:
Set all to cfg.PowerManaged = WdfFalse

comsyspro, 2024-06-07:
The `cfg.PowerManaged = WdfFalse;` is your normal code and I went back to this. Here are also the logs. First the device is attached. Then I rotated the controller. Then I stopped rotationg and the device reboots. Then again I rotetd the controller aned pressed another button. Then I stopped doing anything and the device reboots.

Here are the logs and zipped etl files:
[usbip-flt.txt](https://github.com/user-attachments/files/15722438/usbip-flt.txt)
[usbip-ude.txt](https://github.com/user-attachments/files/15722439/usbip-ude.txt)
[usbip-flt.zip](https://github.com/user-attachments/files/15722450/usbip-flt.zip)
[usbip-ude.zip](https://github.com/user-attachments/files/15722451/usbip-ude.zip)

vadimgrn, 2024-06-09:
I am do not get what is working and what isn't. If Linux is a server and usbip2 is a client, that's enough. Windows usbipd server is not a canonical implementation, I do not care about it. I am going to implement server part for usbip2 this year (at least start development).

comsyspro, 2024-06-09:
This would be a good combitnation to also have your usbipd server for windows. Perhaps such issues will then be history. At the moment we try to get it done with usbipd-win.

closed 2024-06-09 by comsyspro