vadimgrn/usbip-win2 #80: USB3.x Support?

NullYing, 2024-07-26:
[image]

I use usbip-win2 0.9.6.5 for my SanDisk on Raspberry Pi 5, but it doesn't work.

NullYing, 2024-07-26:
[LogSession_20240726_221901.etl.zip](https://github.com/user-attachments/files/16393001/LogSession_20240726_221901.etl.zip)

It's ude log

NullYing, 2024-07-26:
[image]

vadimgrn, 2024-07-27:
hwmon: "Undervoltage detected!"

NullYing, 2024-07-27:
@vadimgrn I use VirtualHere, and it works fine. I think it's not power problem.

[image]

NullYing, 2024-07-27:
I changed the power supply, and the 'Undervoltage detected!' message is gone, but usbip-win2 still has the same problem.

vadimgrn, 2024-07-27:
https://github.com/vadimgrn/usbip-win2/tree/develop?tab=readme-ov-file#obtaining-usbip-logs-on-windows

NullYing, 2024-07-27:
> [LogSession_20240726_221901.etl.zip](https://github.com/user-attachments/files/16393001/LogSession_20240726_221901.etl.zip)
> 
> It's ude log

@vadimgrn I have uploaded the UDE log, thank you.I believe the UDE driver is failing to retrieve the device descriptor. If you need any other detailed logs, please let me know.

307259005, 2024-09-05:
I also encountered the same problem, the SanDisk 64G model is CZ73 usb3, and the same error is reported

vadimgrn, 2024-09-05:
I need the driver's log