vadimgrn/usbip-win2 #81: VHCI Device not found system USB post disabled

sprinteroz, 2024-07-29:
I encountered an issue with this driver upon installation. It disabled all USB ports on my Windows 11 system. Attempting to resolve the problem in safe mode proved unsuccessful, as USB functionality remained unavailable. To restore USB port operation, I had to revert to the last known good configuration.
Subsequently, when trying to run the software, I received a "VHCI Driver not found" error. This may be a consequence of using the last known configuration to reset the drivers and regain USB functionality.
For reference, my system specifications can be found at:
https://www.passmark.com/baselines/V11/display.php?id=215047417287

OpenSourceDevelop, 2024-09-22:
> I encountered an issue with this driver upon installation. It disabled all USB ports on my Windows 11 system. Attempting to resolve the problem in safe mode proved unsuccessful, as USB functionality remained unavailable. To restore USB port operation, I had to revert to the last known good configuration. Subsequently, when trying to run the software, I received a "VHCI Driver not found" error. This may be a consequence of using the last known configuration to reset the drivers and regain USB functionality. Error Image: ![USBIP fail](https://private-user-images.githubusercontent.com/3536573/353196545-2d5704e2-4b33-4c09-b0f6-da936d604bf8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwMjcxMzUsIm5iZiI6MTcyNzAyNjgzNSwicGF0aCI6Ii8zNTM2NTczLzM1MzE5NjU0NS0yZDU3MDRlMi00YjMzLTRjMDktYjBmNi1kYTkzNmQ2MDRiZjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjJUMTc0MDM1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDYzZjMzOTk1YmYyYzQxMDY3ZWRiNTZmODQ2OGMwZmJiMTJhODcyMGIzNGUxMDE4NzNlYThlYWJhYTczNjI2NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ._SeDzPxSFR_FP_-uBn3b5YRJzCPLnwYLo1-Qbw1ofMs) For reference, my system specifications can be found at: https://www.passmark.com/baselines/V11/display.php?id=215047417287

had the same problem with win 10! 😫

vadimgrn, 2024-09-23:
As I explained before, the driver disables nothing. usbip_filter driver is installed, but failed to load for unknown reasons. That's the reason why all usb devices do not work. But you should find out the root of cause yourself, I tested the build on Windows 10 x64 Version 1809, it works. It works for most users too. I don't know what is wrong with your system.

MateuszSychXpl, 2024-10-11:
Exactly the same situation, After installing the ports stopped working. MSI motherboard

Windows 11 Pro version
Version 23H2
OS build 22631.4169
Windows Feature Experience Pack 1000.22700.1034.0

vadimgrn, 2024-10-11:
I cannot do something with that because I haven't had this issue on my laptops/PC or virtual machines. I suppose this is about some missing dependencies, but I can't guess what is could be.

MateuszSychXpl, 2024-10-11:
Maybe you can tell me where to look for the problem? In the system logs I only see:
The driver \Driver\usbip2_filter failed to load for the device PCI\VEN_8086&DEV_7A60&SUBSYS_7E061462&REV_11\3&11583659&0&A0.
The driver \Driver\usbip2_ude failed to load for the device ROOT\USB\0000.

vadimgrn, 2024-10-11:
I don't know how to debug drivers that fail to load. I need to learn that myself.

Hiuoy, 2024-10-17:
I had the same issue, all ports got disabled I have windows 10.
after I have installed it, I did the attach command but it says that:
error: VHCI device not found, driver not loaded?

so I reboot than that happend, I couldn't sign to windows every port is disabled

Hiuoy, 2024-10-17:
> I had the same issue, all ports got disabled I have windows 10. after I have installed it, I did the attach command but it says that: error: VHCI device not found, driver not loaded?
> 
> so I reboot than that happend, I couldn't sign to windows every port is disabled

when I rebooted the system I didn't go to testing mode maybe that was the problem?

vadimgrn, 2024-10-18:
Sure, usbip_ude and usbip_filter drivers cannot be loaded if test signing mode is not enabled. If USB upper filter driver fails to load, all USB devices will not be available.

closed 2025-04-13 by vadimgrn