vadimgrn/usbip-win2 #82: I hope to add support for wider area networks, and I’m really looking forward to it.

ethan0371, 2024-08-24:


vadimgrn, 2024-08-24:
I don't get it, what does it mean? You cannot use something else except TCP/IP for usbip protocol.

ethan0371, 2024-08-24:
I am very happy to read your reply. I want to build a usbip for server turnover, but I have no direction. I wonder if you can tell me which direction I should start with.

vadimgrn, 2024-08-24:
I'm going to add a server for this project myself, but currently I don't have enough time to work on it.

ethan0371, 2024-08-24:
I plan to split the connection between the original USBIP and USBIPD, and then interact with the program through a server on both ends of the split
[image]

ethan0371, 2024-08-24:
Looking forward to your answer. Do you think this approach is acceptable?

vadimgrn, 2024-08-24:
I don't understand what you want. Everything will be implemented in the driver.

ethan0371, 2024-08-24:
Thank you very much for your reply. If you encounter any problems in the future, I hope to communicate with you again.

ethan0371, 2024-08-25:
Do you have an account on freelancer?

closed 2024-08-28 by vadimgrn