vadimgrn/usbip-win2 #84: could you compile a arm64 release version package? please

lxwAsm, 2024-08-29:


vadimgrn, 2024-08-30:
I can try, but I don't have arm64 system for the testing

vadimgrn, 2024-08-30:
Try [this](https://drive.google.com/file/d/1k9apsFKaYeWnNMJjnyJd7RQ7i94euO7O/view?usp=sharing), pls

vadimgrn, 2024-09-08:
What is the point in asking about the arm64 build and then not giving any feedback?

lxwAsm, 2024-09-09:
thanks,I have build arm64 version in my pc,it works normal

lxwAsm, 2024-09-09:
Thank you, author, you are very enthusiastic

vadimgrn, 2024-09-09:
I want you to try my build, please. I'm going to publish it for all if it works.

closed 2024-09-14 by vadimgrn

vadimgrn, 2024-09-23:
If you turn test signing on, it should work

lxwAsm, 2024-09-24:
i turned test signing on when testing.now my pc crashes and can't test....

vadimgrn, 2024-09-24:
You said that you built it and it works, now you are saying the opposite.

lxwAsm, 2024-09-25:
yes,but i didn't use the installer to install.I builded its drivers and usbip.exe arm version,then i installed them with company certificate by my code that is the same as your installer. it works fine.I guess wheather the certificate Influences it.

vadimgrn, 2024-09-25:
OK, thank you. Anyway, I don't have ARM64 Windows PC to find out that.