vadimgrn/usbip-win2 #85: Both ude and wdm  can not work with this device

hbsgz008, 2024-09-09:
I have a usb composite device , when use ude driver,  it shows that String descriptors are not available. 
[image]

  
and when I use usbip-win2-0.9.3.4 wdm driver,  there is 3 interfaces  can not  install driver,  and when I use  the usb-win wdm driver, all the interface's drivers are  installed, but the winusb WCID is null. 
[image]

vadimgrn, 2024-09-10:
https://github.com/vadimgrn/usbip-win2/tree/master?tab=readme-ov-file#obtaining-usbip-logs-on-windows

hbsgz008, 2024-09-19:
this is the ude log:
[logude.txt](https://github.com/user-attachments/files/17057601/logude.txt)

vadimgrn, 2024-09-20:
This is usb full speed device, its endpoint descriptors should be patched, but it does not work. I can do nothing, I don't have the device to try something.

hbsgz008, 2024-09-23:
I changed to use wdm driver,  there is no the ude's  problem. but it got BSOD after running for a while.  In the dump file the receive:wsk irp numbers is more than  the  send:wsk irp numbers. 
[bsoddump.txt](https://github.com/user-attachments/files/17094637/bsoddump.txt)

vadimgrn, 2024-09-23:
WDM driver is no longer supported

closed 2024-10-10 by hbsgz008