vadimgrn/usbip-win2 #88: About auto-reconnecting feature.

arda-bildik, 2024-09-29:
Hi I was searching for a software like this for a long time and I cant describe how amazed I'm right now. Thank you so much for developing and supporting a software like this. I wondered around the settings of the client and realized the Toggle Auto. I'm guessing it is intended for auto reconnect but it did not work for me. Could you please help me about the reconnection. I'm working on a prototype open-source T&M instrument and while firmware development a certain MCU requires almost instant reconnection for uploading the code. I'll try to fix it with some python scripts until I hear back from you though.

vadimgrn, 2024-09-29:
Thank you for the feedback! 

Auto connection is about connecting during the driver initialization, it happens once. The driver will attempt to connect until it succeeds or when an error is received after which there is no point in trying to continue. It will not try to reconnect if the connection is interrupted/closed.

See https://github.com/vadimgrn/usbip-win2/blob/develop/drivers/ude/persistent.cpp

vadimgrn, 2024-09-29:
This feature is quite dangerous. If a BSOD occurs during that, Windows will not be able to boot. Creation of restore point is a good idea before installing/updating USBip.

arda-bildik, 2024-09-29:
Oh I see, thanks for pointing that out. A BSOD  with no return wouldn't be very nice in the end. I'll try to add the feature with a few scripts for now I guess. I would like to give some recommendation about what would be nice if implemented is something like auto-use of a certain selected port when any device gets plugged in. I'll create a repo for my script and add a comment under here if anyone wants to do anything similar to this.

vadimgrn, 2024-09-29:
You can use Auto without doubt if the device works with USBip. Some devices do not work and may even cause BSOD. Just make sure it works before set Auto.

closed 2024-12-28 by vadimgrn