vadimgrn/usbip-win2 #93: Intel AX201 Bluetooth not working

johanbijker, 2024-10-28:
Hi

I can confirm iPhone7 is working, you can add that to your list of devices supported.

However my bluetooth device seems not to work. It's a 8087:0026  Intel(R) Wireless Bluetooth(R) and I can see it from USBIP and attach to it, but in Windows Device Manager I get
Unknown USB Device (Invalid Configuration Descriptor)

Under Information I get:
Device USB\VID_0000&PID_0006\2&3b7c36a2&0&1 was configured.

Driver Name: usb.inf
Driver Package ID: usb.inf_amd64_e434a1287cf2a597
Class GUID: {36fc9e60-c465-11cf-8056-444553540000}
Driver Date: 06/21/2006
Driver Version: 10.0.26100.1301
Driver Provider: Microsoft
Driver Section: BADDEVICE.Dev.NT
Driver Rank: 0xFF0000
Matching Device ID: USB\CONFIGURATION_DESCRIPTOR_VALIDATION_FAILURE
Outranked Drivers: usb.inf:USB\CONFIGURATION_DESCRIPTOR_VALIDATION_FAILURE:00FF2000
Device Updated: false
Parent Device: USB\ROOT_HUB30\1&2b53a856&0&0

johanbijker, 2024-10-28:
When I attach the same device in linux it seems to work

[   47.466436] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[   47.466441] vhci_hcd vhci_hcd.0: devid(65546) speed(2) speed_str(full-speed)
[   47.466485] vhci_hcd vhci_hcd.0: Device attached
[   47.745147] vhci_hcd: vhci_device speed not set
[   47.815112] usb 1-1: new full-speed USB device number 2 using vhci_hcd
[   47.895061] vhci_hcd: vhci_device speed not set
[   47.964932] usb 1-1: SetAddress Request (2) to port 0
[   48.001628] usb 1-1: New USB device found, idVendor=8087, idProduct=0026, bcdDevice= 0.02
[   48.001633] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0

$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

vadimgrn, 2024-10-31:
Thank you for the report. Could you please gather log as described [here](https://github.com/vadimgrn/usbip-win2/tree/master?tab=readme-ov-file#obtaining-usbip-logs-on-windows) ?

vadimgrn, 2024-11-03:
I've managed to replicate this issue on AX200, will try to fix that