vadimgrn/usbip-win2 #95: No USB support after installation and reboot in  Windows 11

heikotroetsch, 2024-11-30:
I wish I could give more information on this bug but I can not get past the login screen without keyboard and mouse. 

What I did: 

1. Installed latest release on windows 11 (not current updates installed)  [v.0.9.6.7](https://github.com/vadimgrn/usbip-win2/releases/tag/v.0.9.6.7)
2. Rebooted system 
3. USB device (mainly mouse and keyboard) do not work after restarting.
4. Restoring windows was nessesary.

Fresh install of windows:
1. Installed latest release on windows 11 (not current updates installed)  [v.0.9.6.7](https://github.com/vadimgrn/usbip-win2/releases/tag/v.0.9.6.7)
2. Rebooted system 
3. USB device (mainly mouse and keyboard) do not work after restarting.
4. Restoring windows was nessesary again. 

Last Downloaded and installed on fresh windows 11 install [v.0.9.6.6](https://github.com/vadimgrn/usbip-win2/releases/tag/v.0.9.6.6)
Also did not work.

vadimgrn, 2024-12-01:
Did you set testsigning on before the installation?

jonnytest1, 2024-12-02:
Same issue :/

heikotroetsch, 2024-12-02:
@vadimgrn i did try Test mode but it resulted in the same behavior.

vadimgrn, 2024-12-03:
I'm tired of explaining this issue. I cannot fix it because I haven't had this issue on my two laptops and PC. If install Windows 10 Version 1809, it will work to. I don't know what is missing on yours PC, you should figure out that themselves.

jonnytest1, 2024-12-03:
<EventData>
  <Data Name="DriverNameLength">66</Data> 
  <Data Name="DriverName">PCI\VEN_1022&DEV_43F7&SUBSYS_11421B21&REV_01\6&2af91c3c&0&00600011</Data> 
  <Data Name="Status">3221226536</Data> 
  <Data Name="FailureNameLength">21</Data> 
  <Data Name="FailureName">\Driver\usbip2_filter</Data> 
  <Data Name="Version">0</Data> 
  </EventData>

vadimgrn, 2024-12-03:
The question is what the reason. Maybe network is not available or something like that.

jonnytest1, 2024-12-03:
that seems unlikely i have an event at 
02.12.2024 21:32:02
Marvell AQtion 10Gbit Network Adapter : Network link has been established at 10Gbit/s Full Duplex.

vadimgrn, 2024-12-03:
If you can build the sources, please build and try (develop)[https://github.com/vadimgrn/usbip-win2/commits/develop/] branch.

jonnytest1, 2024-12-03:
im gonna try to reproduce it on my laptop first , cause i would really not like ot have to reset my main PC every time 🙈

vadimgrn, 2024-12-03:
If usb devices work after installing the develop build, you should use trace log to find out what is failed, see bin/dbg.bat

jonnytest1, 2024-12-03:
ok usb devies also dont work on the laptoip (with the release version) - but i can use touchpad nad onboard keyboard

jonnytest1, 2024-12-03:
this one has absolute path btw  https://github.com/vadimgrn/usbip-win2/blob/master/bin/di.bat 🤔

vadimgrn, 2024-12-03:
Just edit these files

Nativu5, 2024-12-13:
I have the same problem. 

If your USB not working anymore and unluckily u don't have PS/2 keyboard at hand, you can reboot Windows and interrupt the booting by pressing the power button. Then the next boot will show the advanced booting menu, and u can select Disable Driver Signature Enforcement. Reboot, USB will work.

Nativu5, 2024-12-13:
> I wonder why people do not follow the installation instructions and then complain. testsigning must be off, period.

I was able to use usbip2. But another day I booted my PC, testsigning was gone and USB was unresponsive. 

I don't know why testsigning is automatically turned off in 24H2. Might be Windows update?

vadimgrn, 2024-12-13:
I've never had this issue, three PC-s were upgraded to 24H2 recently.

closed 2025-04-13 by vadimgrn