vadimgrn/usbip-win2 #96: Sometimes the usbip attach command may become blocked

xiongfeng-xiao, 2024-12-02:
Hi, 
    I found that when using the usbip attach command, the user-space program gets blocked. Debugging revealed that the code is stuck at WdfRequestComplete(request, st) in the complete function.
```c
_Function_class_(EVT_WDF_WORKITEM)
_IRQL_requires_same_
_IRQL_requires_max_(PASSIVE_LEVEL)
PAGED void NTAPI complete(_In_ WDFWORKITEM wi)
{
        PAGED_CODE();
       ..........

        if (st != STATUS_PENDING) {
                TraceDbg("req %04x, %!STATUS!", ptr04x(request), st);
                WdfRequestComplete(request, st);    // ===> This line  will be blocked.
        }
}
```
Tracing the code, I found that the possible reason might be that the cleanupcallback function set in the create_workitem function is not being called by the WDF framework. Is my assumption correct?
```c
PAGED auto create_workitem(_Out_ WDFWORKITEM &wi, _In_ WDFOBJECT parent)
{
        PAGED_CODE();

       .........
        attr.EvtCleanupCallback = workitem_cleanup;    // ====> This func was not called.
        attr.ParentObject = parent;

        return WdfWorkItemCreate(&cfg, &attr, &wi);
}
```
Please help confirm this issue. Thank you.

vadimgrn, 2024-12-03:
I resolved similar issue in the other driver recently. I'll refactor this code.

xiongfeng-xiao, 2024-12-05:
okay,I tried to fix this issue but failed. Could you please help me understand what might be causing it? Thank you!

vadimgrn, 2024-12-05:
The same WDFWORKITEM is used for WskConnect calls. The possible workaround is to move this WDFWORKITEM to device_ctx_ext.

xiongfeng-xiao, 2024-12-06:
Oh, i see. Adding a network proxy layer between the vhci driver and the usbipd link can increase the probability of reproducing the issue, making it easier for you to debug the code for this problem, as shown in the diagram below.

[image]

vadimgrn, 2024-12-08:
Could you please build and test the develop branch?

xiongfeng-xiao, 2024-12-08:
Testing with your latest submitted code, the issue starts to occur after performing attach and detach a total of 32 times. The observed behavior is that after the device successfully attaches to the local system, it immediately disappears. When I re-plug the device and attempt to attach again, Windows crashes with a blue screen immediately.

Conclusion: The issue of attach being blocked no longer occurs, but as the number of attach and detach operations increases, functional anomalies appear, accompanied by blue screen crashes. I will upload detailed logs and other information after capturing them.

vadimgrn, 2024-12-08:
I've caught the BSOD, thank you

vadimgrn, 2024-12-08:
The BSOS has been fixed. The top commit is amended.

xiongfeng-xiao, 2024-12-09:
The Windows BSOD has been fixed, but after performing multiple **attach and detach** operations, the wusbip app reports an error code: 1450 - "Insufficient system resources exist to complete the requested service."

vadimgrn, 2024-12-09:
I am using verifier and it doesn't report any resources leaks. How many attaches you are doing?

xiongfeng-xiao, 2024-12-09:
The same device performs approximately 20 attach operations, and the issue generally occurs after a longer interval between two operations. If the interval between two attach operations is very short, the problem doesn't seem to occur (to be verified).

PS C:\Users\xxf> verifier.exe /query

Timestamp: 12/09/2024 14:06:41.327

Verifier Flags: 0x0311efbf

  Standard Flags:

    [X] 0x00000001 Special Pool.
    [X] 0x00000002 Force IRQL Checking.
    [X] 0x00000008 Pool Tracking.
    [X] 0x00000010 I/O Verification.
    [X] 0x00000020 Deadlock Detection.
    [X] 0x00000080 DMA Checking.
    [X] 0x00000100 Security Checks.
    [X] 0x00000800 Miscellaneous Checks.
    [ ] 0x00020000 DDI Compliance Checking.

  Additional Flags:

    [X] 0x00000004 Randomized Low Resources Simulation.
    [X] 0x00000200 Force Pending I/O Requests.
    [X] 0x00000400 IRP Logging.
    [X] 0x00002000 Fixed MDL Check for Stack.
    [X] 0x00004000 Fixed MDL Check for Driver.
    [X] 0x00008000 Power Framework Delay Fuzzing.
    [X] 0x00010000 Port/Miniport Interface Checking.
    [ ] 0x00040000 Systematic Low Resources Simulation.
    [ ] 0x00080000 DDI Compliance Checking (Additional).
    [ ] 0x00200000 NDIS/WLAN Verification.
    [ ] 0x00800000 Kernel Synchronization Delay Fuzzing.
    [X] 0x01000000 VM Switch Verification.
    [X] 0x02000000 Code Integrity Checks.

  Internal Flags:

    [X] 0x00100000 Extended Verifier Flags (Internal).

    [X] Indicates that flags are enabled.

  Verifier Statistics Summary:

    Raise IRQL:                              0
    Acquire Spinlock:                        0
    Synchronization Executions:             0
    Adjustments:                             0

    Pool Allocations Attempted:          5440
    Pool Allocations Succeeded:          5440
    Special Pool Allocations Succeeded:  5440
    Unmarked Pool Allocations:              0
    Untracked Pool Allocations:           610
    Pool Allocations Failed:                0
    Pool Allocations Intentionally Failed: 11        <=========  The error count is increasing

  Driver Verification List:

    Module: usbip2_ude.sys (Loads: 1 / Unloads: 0)

      Pool Allocation Statistics: ( NonPaged / Paged )

        Current Pool Allocations:   (        5 /        3 )
        Current Pool Bytes:         (     1520 /       40 )
        Peak Pool Allocations:      (       17 /        6 )
        Peak Pool Bytes:            (     4280 /       80 )
        Contiguous Memory Bytes:    0
        Peak Contiguous Memory Bytes: 0

    Module: usbip2_filter.sys (Loads: 6 / Unloads: 5)

      Pool Allocation Statistics: ( NonPaged / Paged )

        Current Pool Allocations:   (        0 /        0 )
        Current Pool Bytes:         (        0 /        0 )
        Peak Pool Allocations:      (        3 /        1 )
        Peak Pool Bytes:            (     1642 /     1024 )
        Contiguous Memory Bytes:    0
        Peak Contiguous Memory Bytes: 0
PS C:\Users\xxf>

vadimgrn, 2024-12-09:
I'm using usbip-win2\bin\attach_loop.py program for the testing. usb flash drive is used on Linux server.

python3.exe attach_loop.py -r pc -b1-10
1600 attaches were made without issues. A resource leak is possible, of course. My laptop has 32GB of RAM, I don't know how long it should run to exhaust it.

Could you please enable driver logging and repro the issue? Maybe the log will show what and where is failed. I used to use scripts from usbip-win2\bin, dbg.bat installs the drivers and enables logging, vz.bat makes text logs.

xiongfeng-xiao, 2024-12-09:
The test logs have been uploaded as an attachment. The possible reproduction steps are as follows: Perform 20 attach and detach operations, wait for 10 minutes, and then perform another attach operation. The issue should occur at this point.
[usbip-ude.txt](https://github.com/user-attachments/files/18059280/usbip-ude.txt)
[usbip-flt.txt](https://github.com/user-attachments/files/18059284/usbip-flt.txt)

vadimgrn, 2024-12-09:
It is OK if you enable this verifier switch
[X] 0x00000004 Randomized Low Resources Simulation.

xiongfeng-xiao, 2024-12-09:
https://github.com/vadimgrn/usbip-win2/issues/96#issuecomment-2527258813

[X] 0x00000004 Randomized Low Resources Simulation.   ---> It seems that I have already enabled this flag. Please refer to the quoted comment. My laptop also has 32GB of memory. I will recheck my environment and perform multiple verifications.

xiongfeng-xiao, 2024-12-09:
Oh, I see. After I disabled it, the issue didn't occur again. Thank you.

closed 2024-12-10 by xiongfeng-xiao

hifihedgehog, 2026-09-09:
@vadimgrn, this follows the callback-reuse mechanism you described in [your earlier comment](https://github.com/vadimgrn/usbip-win2/issues/96#issuecomment-2519419106). I reproduced a post-attach work-item self-wait that blocks unload in 0.9.8.0 and prepared a correction in https://github.com/vadimgrn/usbip-win2/pull/188: 27 insertions and 7 deletions in one source file. The PR contains the full patch, reproduction commands, measured kernel results, and a deterministic regression model.

The current reproduction uses the unmodified 0.9.8.0 installer/client and an independent VIIPER 0.7.0 server on loopback. A cold host-controller restart passes. One neutral Xbox 360 attach/detach succeeds, imports are empty, and VIIPER exits. The next host-controller restart then exceeds the 300-second test limit.

The dump has a callback blocked under `WdfObjectDelete -> FxWorkItem::Dispose -> WaitForSignal`, while driver unload waits behind it. The work item has one running callback but a null thread marker. The earlier thunk can clear that marker after the later callback has started, which makes the later callback wait on itself during deletion.

The correction gives each connection attempt a fresh work item. After socket creation succeeds, its ownership handoff is:

```cpp
WDFWORKITEM wi{};
st = create_workitem(wi, ctx.vhci);
if (NT_ERROR(st)) {
        Trace(TRACE_LEVEL_ERROR, "WdfWorkItemCreate %!STATUS!", st);
        return st;
}

auto &next = *get_workitem_ctx(wi);
next = ctx;
ctx.ctx_ext = WDF_NO_HANDLE;
ctx.addrinfo = nullptr;
ctx.request = WDF_NO_HANDLE;

set_args(next.args, request, __func__, &ai);
auto irp = WdfRequestWdmGetIrp(request);
IoSetCompletionRoutine(irp, irp_complete, wi, true, true, true);
```

After starting WSK, `connect()` still returns `STATUS_PENDING` unconditionally. The predecessor's `complete()` now deletes its own work item before returning:

```cpp
if (st == STATUS_PENDING) {
        WdfObjectDelete(wi);
        return;
}
```

The existing stack `ObjectRef` remains alive until that callback returns. Failures before ownership transfer retain the old cleanup path, including socket closure/free. Retries use the same fresh-item path. The complete diff also includes the helper declaration and signature/call-site changes.

The comparison used pristine and corrected source with the same compiler, WDK, and test certificate in separate clean Windows 11 VMs. The pristine build exceeded 300 seconds. Its matching PDB resolves the blocked callback to `complete+0x311`, an offset specific to that build. The corrected build passed the same operation in 138 ms and passed 22 further targeted kernel checks covering both receive modes, address retry, failed imports, client termination during an import, and parent removal during a pending import. Driver Verifier flags in the failing control were zero.

AMD64 and ARM64 builds pass, with runtime coverage on AMD64 only. The 129-case model using the actual source functions also passes. These results concern the attach lifetime correction. Production signing and full HIDMaestro integration testing remain pending.

If the change is suitable, a Microsoft-signed AMD64/ARM64 candidate would allow the remaining consumer checks to run against the intended package. The PR preserves the tested release-tag baseline and also applies cleanly to current `develop`.