vadimgrn/usbip-win2 #97: cant find signed driver config in vc studio from instruction

Fjox, 2024-12-08:


vadimgrn, 2024-12-08:
Elaborate, pls

Fjox, 2024-12-08:
> Elaborate, pls

baed on instruction i cant find where the setting is for the test certificate

vadimgrn, 2024-12-08:
[image]

closed 2024-12-28 by vadimgrn

Fjox, 2024-12-28:
fyi its not there

eluna7, 2025-01-25:
also having this issue but i believe its because im on windows 10. the sdk and wdk do specifically say for windows 11. i believe like me, fjox may be on windows 10 and thats causing this issue. ~~un~~fortunately not gonna be installing win11, so i wont be able to compile this myself

saw another issue where the driver could potentially be signed but another third party so hoping that happens soonish:tm: !! ty for the work youve done on this vadmingrn <3

Fjox, 2025-02-11:
im on windows 10 and im never gonan touch 11 again. that shit was ass